espressif/esp-idf #5867: Modbus Serial Master resource release failure.

bhavikbhansali, 2020-09-14:
## Environment

- Development Kit:      ESP32 Nodemcu (Clone) (Self-made but used the same schematics)
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    v4.1
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

I have a self-made board which is a copy to esp32 nodemcu but with some additional hardware. I tried the modbus_master code for trial and it compiles and runs but there bit problem with its strange working. 

With Modbus Slave Software, it works well without any issue and it prints the data well with the same resister address and data format. (Same code in example "esp-idf/examples/protocols/modbus/serial/mb_master/") 
With the actual device which is Secure Elite 440 MFM Meter, it not shows any data. (Same code as above)

### Expected Behavior
Output of Modbus Slave Windows Software

> I (559) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 244.100006 (0x4374199a) read successful.
> I (3609) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 244.100006 (0x4374199a) read successful.
> I (6659) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 244.100006 (0x4374199a) read successful.
> I (9719) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 244.100006 (0x4374199a) read successful.
> I (12769) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 244.100006 (0x4374199a) read successful.
> I (15829) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 244.100006 (0x4374199a) read successful.

### Actual Behavior
Output of Secure Elite 440 MFM Meter

> E (649) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x107) (ESP_ERR_TIMEOUT).
> E (649) MASTER_TEST: Characteristic #0 (Voltage RN) read fail, err = 263 (ESP_ERR_TIMEOUT).
> E (659) MB_PORT_COMMON: vMBMasterRunResRelease(172): vMBMasterRunResRelease: resource release failure.
> I (3659) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 0.000000 (0x0) read successful.
> E (3839) MB_PORT_COMMON: vMBMasterRunResRelease(172): vMBMasterRunResRelease: resource release failure.
> I (6659) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 0.000000 (0x0) read successful.
> E (6939) MB_PORT_COMMON: vMBMasterRunResRelease(172): vMBMasterRunResRelease: resource release failure.
> I (9659) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 0.000000 (0x0) read successful.
> E (9949) MB_PORT_COMMON: vMBMasterRunResRelease(172): vMBMasterRunResRelease: resource release failure.
> I (12659) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 0.000000 (0x0) read successful.
> E (12839) MB_PORT_COMMON: vMBMasterRunResRelease(172): vMBMasterRunResRelease: resource release failure.
> I (15659) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 0.000000 (0x0) read successful.
> E (15819) MB_PORT_COMMON: vMBMasterRunResRelease(172): vMBMasterRunResRelease: resource release failure.

### Steps to reproduce
[image]

With my device rs485 port, I connect a 2 device as per picture. On the right side meter from which I want to read data. On a top, I put an RS485 to TTL converter which is connected to ttl to USB converter so that I can monitor the packet send by esp to meter and meter response.

Also with USB to RS485 converter I can read the same resister data from Modbus Slave software too with the same communication setting along with resister address and data type.

###Code File
[master.txt](https://github.com/espressif/esp-idf/files/5220675/master.txt)

Alvin1Zhang, 2020-09-15:
Thanks for reporting, we will look into.

alisitsyn, 2020-09-15:
Hello @bhavikbhansali,

Thanks for this issue. Could you provide more information needed to reproduce the issue:

1. I need the documentation for the device you are going to read. Is this correct link to the manual?
https://www.umsmeters.co.uk/app/download/23259059/Elite+440+Technical+Reference+Manual+R11.pdf
2. Could you set logging verbosity to debug and send me Modbus log from your master application (See below)?
[image]
3. Could you send the captured log from your RS485-USB converter?

Thank you.

bhavikbhansali, 2020-09-15:
> Hello @bhavikbhansali,
> 
> Thanks for this issue. Could you provide more information needed to reproduce the issue:
> 
> 1. I need the documentation for the device you are going to read. Is this correct link to the manual?
>    https://www.umsmeters.co.uk/app/download/23259059/Elite+440+Technical+Reference+Manual+R11.pdf
> 2. Could you set logging verbosity to debug and send me Modbus log from your master application (See below)?
>    [image]
> 3. Could you increase the response time from a slave in Modbus configuration?
>    [image]
> 4. Could you send the captured log from your RS485-USB converter?
> 
> Thank you.

@alisitsyn Thanks for the replay.

Changing the things and after changing the value as mentioned on point it works for me. Below is the output from the monitor.

> I (1804) MASTER_TEST: Start modbus test...
> D (1804) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_FRAME_TRANSMIT
> D (1814) MB_PORT_COMMON: eMBMasterRTUSend: Port enter critical.
> D (1814) MB_PORT_COMMON: eMBMasterRTUSend: Port exit critical
> D (1824) MB_PORT_COMMON: vMBMasterPortTimersRespondTimeoutEnable Respond enable timeout.
> D (1834) MB_MASTER_SERIAL: MB_TX_buffer sent: (9) bytes.
> D (1834) MB_MASTER_SERIAL: MB_uart[2] event:
> D (1834) MB_MASTER_SERIAL: uart rx break.
> D (1834) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_FRAME_SENT
> D (1984) MB_MASTER_SERIAL: MB_uart[2] event:
> D (1984) MB_MASTER_SERIAL: Receive data, len: 9.
> D (1984) MB_MASTER_SERIAL: Received data: 9(bytes in buffer)
> 
> D (1984) MB_PORT_COMMON: eMBMasterRTUReceive: Port enter critical.
> D (1994) MB_PORT_COMMON: eMBMasterRTUReceive: Port exit critical
> D (1994) POLL RCV buffer:: 03 04 15 39 43 68
> D (2004) MB_PORT_COMMON: eMBMasterPoll: Packet data received successfully (0).
> D (2014) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_EXECUTE
> D (2014) MB_PORT_COMMON: vMBMasterCBRequestSuccess: Callback request success.
> D (2024) MB_PORT_COMMON: eMBMasterWaitRequestFinish: returned event = 0x40
> D (2024) MB_CONTROLLER_MASTER: mbc_serial_master_get_parameter: Good response for get cid(0) = ESP_OK
> I (2034) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 232.082901 (0x43681539) read successful.
> D (4044) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_FRAME_TRANSMIT
> D (4044) MB_PORT_COMMON: eMBMasterRTUSend: Port enter critical.
> D (4044) MB_PORT_COMMON: eMBMasterRTUSend: Port exit critical
> D (4044) MB_PORT_COMMON: vMBMasterPortTimersRespondTimeoutEnable Respond enable timeout.
> D (4054) MB_MASTER_SERIAL: MB_TX_buffer sent: (9) bytes.
> D (4064) MB_MASTER_SERIAL: MB_uart[2] event:
> D (4064) MB_MASTER_SERIAL: uart rx break.
> D (4064) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_FRAME_SENT
> D (4364) MB_MASTER_SERIAL: MB_uart[2] event:
> D (4364) MB_MASTER_SERIAL: Receive data, len: 9.
> D (4364) MB_MASTER_SERIAL: Received data: 9(bytes in buffer)
> 
> D (4374) MB_PORT_COMMON: eMBMasterRTUReceive: Port enter critical.
> D (4374) MB_PORT_COMMON: eMBMasterRTUReceive: Port exit critical
> D (4384) POLL RCV buffer:: 03 04 3b 25 43 68
> D (4384) MB_PORT_COMMON: eMBMasterPoll: Packet data received successfully (0).
> D (4394) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_EXECUTE
> D (4394) MB_PORT_COMMON: vMBMasterCBRequestSuccess: Callback request success.
> D (4404) MB_PORT_COMMON: eMBMasterWaitRequestFinish: returned event = 0x40
> D (4414) MB_CONTROLLER_MASTER: mbc_serial_master_get_parameter: Good response for get cid(0) = ESP_OK
> I (4424) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 232.231033 (0x43683b25) read successful.
> D (6434) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_FRAME_TRANSMIT
> D (6434) MB_PORT_COMMON: eMBMasterRTUSend: Port enter critical.
> D (6434) MB_PORT_COMMON: eMBMasterRTUSend: Port exit critical
> D (6434) MB_PORT_COMMON: vMBMasterPortTimersRespondTimeoutEnable Respond enable timeout.
> D (6444) MB_MASTER_SERIAL: MB_TX_buffer sent: (9) bytes.
> D (6454) MB_MASTER_SERIAL: MB_uart[2] event:
> D (6454) MB_MASTER_SERIAL: uart rx break.
> D (6454) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_FRAME_SENT
> D (6594) MB_MASTER_SERIAL: MB_uart[2] event:
[... 53 lines cut ...]
> I (11244) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 232.284103 (0x436848bb) read successful.
> D (13254) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_FRAME_TRANSMIT
> D (13254) MB_PORT_COMMON: eMBMasterRTUSend: Port enter critical.
> D (13254) MB_PORT_COMMON: eMBMasterRTUSend: Port exit critical
> D (13254) MB_PORT_COMMON: vMBMasterPortTimersRespondTimeoutEnable Respond enable timeout.
> D (13264) MB_MASTER_SERIAL: MB_TX_buffer sent: (9) bytes.
> D (13274) MB_MASTER_SERIAL: MB_uart[2] event:
> D (13274) MB_MASTER_SERIAL: uart rx break.
> D (13274) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_FRAME_SENT
> D (13414) MB_MASTER_SERIAL: MB_uart[2] event:
> D (13414) MB_MASTER_SERIAL: Receive data, len: 9.
> D (13414) MB_MASTER_SERIAL: Received data: 9(bytes in buffer)
> 
> D (13424) MB_PORT_COMMON: eMBMasterRTUReceive: Port enter critical.
> D (13424) MB_PORT_COMMON: eMBMasterRTUReceive: Port exit critical
> D (13434) POLL RCV buffer:: 03 04 86 ed 43 68
> D (13434) MB_PORT_COMMON: eMBMasterPoll: Packet data received successfully (0).
> D (13444) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_EXECUTE
> D (13454) MB_PORT_COMMON: vMBMasterCBRequestSuccess: Callback request success.
> D (13454) MB_PORT_COMMON: eMBMasterWaitRequestFinish: returned event = 0x40
> D (13464) MB_CONTROLLER_MASTER: mbc_serial_master_get_parameter: Good response for get cid(0) = ESP_OK
> I (13474) MASTER_TEST: Characteristic #0 Voltage RN (Volt) value = 232.527054 (0x436886ed) read successful.
> I (15484) MASTER_TEST: Destroy master...
> D (15484) MB_PORT_COMMON: eMBMasterRTUStop: Port enter critical.
> D (15484) MB_PORT_COMMON: eMBMasterRTUStop: Port exit critical

bhavikbhansali, 2020-09-15:
@alisitsyn I used to do coding the esp32 with Arduino as it works for me but now want to move ahead and want to utilize the power of the controller can you suggest me some good tutorial from where I can understand the ESP-IDF more properly.

Current documentation is a bit difficult for me to understand.

Thank You.

alisitsyn, 2020-09-15:
@bhavikbhansali,

Good to hear it helps. Please try the book mentioned here: https://www.esp32.com/viewtopic.php?t=5217
It is good to start.
In spite of the release/v4.1 works for you now there are some known issues that can come later. Could you apply the patch below to release/v4.1 and test it with your environment?

[patch_drv_modbus_v41.zip](https://github.com/espressif/esp-idf/files/5223873/patch_drv_modbus_v41.zip)

Please check it, let me know about the results of the testing, and send me your logs. This would help you to avoid issues later and contribute to testing.
Thanks.

bhavikbhansali, 2020-09-15:
@alisitsyn Sure will go through it. Please allow me for some time to test with the current setup as well as release/v4.1 too.

Thanks for the quick and amazing support.

bhavikbhansali, 2020-09-19:
Hello, @alisitsyn Thanks for the help. and suggestion for book.

Currently, my issue is to resolve and want to close this issue. In the future when I face any issue and need help from your guys will raise again with the details.

Thanks.
Bhavik

closed 2020-09-19 by bhavikbhansali

alisitsyn, 2020-09-21:
@bhavikbhansali,
Ok. Please let me know, If you have any issues or there are results related to patch. Thanks.

bhavikbhansali, 2021-02-27:
Hello Guys,

I hope all of you are doing well.

Is there any good tutorial available instead of a book to learn how to use ESP-IDF with windows (maybe with VS Code)?

georgik, 2021-03-03:
@bhavikbhansali This might help you: Quick User Guide on Youtube for VS Code: https://youtu.be/Lc6ausiKvQM