espressif/esp-idf #5868: mbedtls version of master branch reverted back to v2.9

fariouche, 2020-09-15:
Hello,

I've just done a git pull of my esp-idf clone and noticed that mbedtls went from version 2.16.7 to 2.9...
Is that normal? Any reason?
Unfortunately I'm using iotivity-lite and I need the v2.16 to apply some patches to add the ANON cypher suite.
2.9 is way too old :-(
Thank you!

mahavirj, 2020-09-15:
@fariouche 

You may find reference to `mbedtls` submodule per below, which is pointing to correct `v2.16.7` version:

https://github.com/espressif/esp-idf/tree/master/components/mbedtls
https://github.com/espressif/mbedtls/commits/mbedtls-2.16.7-idf

Can you please share information for following commands:

```
$ git describe --tags
$ git submodule
```

fariouche, 2020-09-15:
I can see that my mbedtls is pointing to the development branch, not sure why
Doing a "git remote show origin" shows
remote origin
  Fetch URL: https://github.com/espressif/mbedtls.git
  Push  URL: https://github.com/espressif/mbedtls.git
  HEAD branch: development

Here are the request commands 
git describe --tags : v4.3-dev-907-g6c17e3a64
git submodule:
 f31694c9f1746ba189a4bcae2e34db15135ddb22 components/asio/asio (asio-1-12-0-201-gf31694c9)
 d037ec89546fad14b5c4d5456c2e23a71e554966 components/bootloader/subproject/components/micro-ecc/micro-ecc (v1.0)
 93eeca8b535dd609da8b1cdac44a11707e03c333 components/bt/controller/lib (remotes/m/master)
 95bd8644abf4a410dd3fb914468d3a23ac9bbec2 components/bt/host/nimble/nimble (remotes/m/master)
 085ca40781f7c39febe6d14fb7e5cba342e1804b components/cbor/tinycbor (v0.5.3-6-g085ca40)
 98954eb30a2e728e172a6cd29430ae5bc999b585 components/coap/libcoap (v4.2.0-rc4-114-g98954eb)
 3d951fb6c4fce3a2655e63d475b80b09c6acfddf components/esp_wifi/lib (remotes/m/master)
 1b296066bea253f1eb857188c1ac5548696e49e7 components/esptool_py/esptool (v2.8-52-g1b29606)
 968b8cc46dbee47b83318d5f31a8e7907199614b components/expat/expat (R_2_2_5)
 3c8935676a97c7c97bf006db8312875b4f292f6c components/json/cJSON (v1.7.12)
 70170c28c844a4786e75efc626e1aeebc93caebc components/libsodium/libsodium (1.0.12)
 80d6d19a929c6db577fc44a47fdb4acffdd68933 components/lwip/lwip (remotes/m/master)
 90f46c8b17bc1219a82d4ddf81520d40c5ac5ebf components/mbedtls/mbedtls (mbedtls-2.13.1-1599-g90f46c8b1)
 6bc94add892437d0fd50f26bfabe78c646648c13 components/mqtt/esp-mqtt (ESP-MQTT_FOR_IDF_3.1-170-g6bc94ad)
 8f7b008b158e12de0e58247afd170f127dbb6456 components/nghttp/nghttp2 (v1.41.0)
 dac1a65feac4ad72f612aab99f487056fbcf5c1a components/protobuf-c/protobuf-c (v1.3.0)
 f5e26c4e933189593a71c6b82cda381a7b21e41c components/spiffs/spiffs (0.2-221-gf5e26c4)
 a2ba3dccccf94022d31e939fa2ce4dca5f0a34f0 components/tinyusb/tinyusb (legacy-2054-ga2ba3dcc)
 7d2bf62b7e6afaf38153041a9d53c21aeeca9a25 components/unity/unity (v2.4.3-51-g7d2bf62)
 7e8e249990ec491ec15990cf95b6d871a66cf64a examples/build_system/cmake/import_lib/main/lib/tinyxml2 (2.0.2-659-g7e8e249)
 71e69383a0368bb704e1b0d90b7e5697afae7d5d examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib (remotes/m/master)

mahavirj, 2020-09-15:
@fariouche 

`90f46c8b17bc1219a82d4ddf81520d40c5ac5ebf components/mbedtls/mbedtls`

This is correct commit-id. You may cross check version from file `components/mbedtls/mbedtls/ChangeLog` as well. What makes you think that it has been switched back to `v2.9`?

fariouche, 2020-09-15:
One of my patches didn't apply, and when I've checked the checkout branch I've seen the development branch being used, which is based on the v2.9.

I've redone a clean checkout of all esp-idf and now it's better. SOrry. Not sure what Happenend

closed 2020-09-15 by fariouche