espressif/esp-idf #5869: Deep-sleep high consumption

tsarquis88, 2020-09-15:
## Environment

- Development Kit: **ESP32-DevKitC**
- Kit version (for WroverKit/PicoKit/DevKitC): **v4**
- Module or chip used: **ESP32-WROOM-32D**
- IDF version: **v4.0.1-441-g6093407d7**
- Build System: **idf.py**
- Compiler version: **xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0**
- Operating System: **Linux**
- Using an IDE?: **No**
- Power Supply: **external 3.3V**

## Problem Description

The consumption in deep-sleep mode is very high.

### Expected Behavior

A current value of ~150uA

### Actual Behavior

A current value of ~3mA

### Steps to reproduce

Just testing your *system/deep-sleep* example.

## Other items if possible
- My measurement method is placing the multimeter in serial between the 3.3V battery and the 3.3V pin.
- I read [#3418] about modifying the PCB would give me better results... What would be exactly those modifications? Besides of removing the red LED.
- Is there any change in *sdkconfig* that would help me?

igrr, 2020-09-15:
Hi @tsarquis88, ESP32 DevKit-C contains components such as USB-UART adapter which continue to draw current even when the ESP32 is in deep sleep mode. For this reason, with DevKit-C it is very hard to get accurate current measurement for the ESP32 module only.

To measure current consumption of the ESP32, suggest one of the following options:

* use a bare ESP32-WROOM-32 or WROVER-32 module, connecting the power supply, TXD/RXD, EN, GPIO0 pins to an external programmer.
* use [ESP32-MeshKit-Sense board](https://github.com/espressif/esp-iot-solution/blob/master/documents/evaluation_boards/ESP32-MeshKit-Sense_guide_en.md), which has dedicated jumpers on board to facilitate current measurements.
* use the [WROVER-KIT](https://www.espressif.com/en/products/hardware/esp-wrover-kit/overview) board. Right next to the top left corner of the module, there is a zero ohm shunt resistor. You can desolder it or replace it with larger shunt, and do current measurements.

tsarquis88, 2020-09-15:
Hello @igrr. Thank you for the quick answer.

Okay, I won't get that micro-amperage in my DevKit, but I could be sure that the deep-sleep consumption will be the correct one in a ESP32 module.

closed 2020-09-15 by tsarquis88