espressif/esp-idf #5870: Endian.h bug

yareach, 2020-09-15:
Could not compile endian.h with the release 4. Found out line 86's condition is not evaluated properly:
#if BYTE_ORDER == _LITTLE_ENDIAN

The _LITTLE_ENDIAN is defined as LITTLE_ENDIAN at line 40. When removed the underscore char at line 86, it passed.

igrr, 2020-09-16:
Hi @yareach, I couldn't reproduce this issue by including `<endian.h>` in a hello-world example. Macro expansion shows that both `LITTLE_ENDIAN` and `_LITTLE_ENDIAN` are defined to the same value (one in wpa_supplicant's endian.h, the other in <machine/endian.h> from the toolchain).

Could you please attach the code to reproduce the issue, and fill the issue template that was offered to you when opening the issue? Information such as the IDF commit ID, compiler version, build system, is very useful for us to reproduce the issue quickly.

yareach, 2020-09-19:
Hi igrr,
I've alread moved the project to 3.3. Now I tried to compile the version I'd moved to 3.3 on 4.0 to reproduce the error but built without errors. So I don't understand why before it didn't work.
There was just a wrong cast in ff.c (which is not scope of this ticket):
C:/Users/yarea_000/ESP32/esp-idf-v4.0/components/fatfs/src/ff.c:6284:17: warning: passing argument 1 of 'tchar2uni' from incompatible pointer type [-Wincompatible-pointer-types]
  dc = tchar2uni(&tp); /* UTF-8 ==> UTF-16 */
                 ^~~
C:/Users/yarea_000/ESP32/esp-idf-v4.0/components/fatfs/src/ff.c:746:16: note: expected 'const TCHAR **' {aka 'const char **'} but argument is of type 'TCHAR **' {aka 'char **'}
  const TCHAR** str  /* Pointer to pointer to TCHAR string in configured encoding */
  ~~~~~~~~~~~~~~^~~

igrr, 2020-09-20:
@yareach regarding the ff.c issue — same question, please let us know the IDF commit ID (if different from the latest release tag), compiler version, build system you are using. If you can upload the sdkconfig file used to build the project, that would also help.

yareach, 2020-09-20:
Hi igrr,

Version 4.0.1 - installed as zipped by the IDF Onboarding within VSC.
Compiler 8.2.0
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/5252352/sdkconfig.zip)

More details:

Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.17.1.windows.2")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/yarea_000/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/yarea_000/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/yarea_000/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/yarea_000/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/yarea_000/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v4.0.1-dirty
-- Building ESP-IDF components for target esp32
CMake Warning at C:/Users/yarea_000/ESP32/esp-idf-v4.0/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain
  C:/Users/yarea_000/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
  crosstool-ng version esp-2020r2 doesn't match supported version esp-2019r2.
  Check Getting Started documentation or proceed at own risk.

igrr, 2020-09-20:
Do you have any local modifications in FATFS, or are you passing any extra FATFS configuration via CPPFLAGS? The line of the code where the error happens is guarded with `#if FF_USE_STRFUNC`, and FF_USE_STRFUNC is defined as 0 in ffconf.h, so this line shouldn't be compiled.

Could you please paste the compiler command for this file? It should be printed by the build system immediately above the error message.

yareach, 2020-09-20:
I modified two params in ffconf.h. See the file.
[ffconf.zip](https://github.com/espressif/esp-idf/files/5252410/ffconf.zip)

[876/1087] Building C object esp-idf/fatfs/CMakeFiles/__idf_fatfs.dir/src/ff.c.obj
C:/Users/yarea_000/ESP32/esp-idf-v4.0/components/fatfs/src/ff.c: In function 'putc_bfd':
C:/Users/yarea_000/ESP32/esp-idf-v4.0/components/fatfs/src/ff.c:6284:17: warning: passing argument 1 of 'tchar2uni' from incompatible pointer type [-Wincompatible-pointer-types]
  dc = tchar2uni(&tp); /* UTF-8 ==> UTF-16 */
                 ^~~
C:/Users/yarea_000/ESP32/esp-idf-v4.0/components/fatfs/src/ff.c:746:16: note: expected 'const TCHAR **' {aka 'const char **'} but argument is of type 'TCHAR **' {aka 'char **'}
  const TCHAR** str  /* Pointer to pointer to TCHAR string in configured encoding */

igrr, 2020-09-21:
@yareach Thanks for the extra details. Any chance you have modified anything else in the fatfs component? With the ffconf.h file you have provided, compilation in IDF v4.0.1 should fail with the following error:
```c++
../../../../components/fatfs/src/ff.h:32:2: error: #error Wrong configuration file (ffconf.h).
 #error Wrong configuration file (ffconf.h).
```
due to the mismatch between `FFCONF_DEF` value defined in ffconf.h and the one expected in ff.c.

If you are unsure which files you have modified, you can run `git status --untracked-files=no` in ESP-IDF directory to get the list of modified files. You can also run `git diff` to get the full list of modifications as a plaintext diff.

yareach, 2020-09-21:
I'm adding the diff file.
[diff.txt](https://github.com/espressif/esp-idf/files/5257210/diff.txt)

igrr, 2020-09-21:
Thanks @yareach, I was able to confirm this warning. It looks like an issue in the upstream library, reported in http://elm-chan.org/fsw/ff/bd/?show=3457. A patch for this issue has been published at http://elm-chan.org/fsw/ff/patches.html. We'll apply this patch and also backport it to release/v4.0 branch.
Thanks again for reporting.

AxelLin, 2022-03-01:
> Thanks @yareach, I was able to confirm this warning. It looks like an issue in the upstream library, reported in http://elm-chan.org/fsw/ff/bd/?show=3457. A patch for this issue has been published [...]

Fixed in master: bb8587a13998e6d2fa645c071425f095c3a44698
But the fix is not in any stable branches.

Alvin1Zhang, 2022-05-16:
Thanks for reporting, sorry for slow turnaround, fix is available both on release/4.4 https://github.com/espressif/esp-idf/commit/de7d69e5cce41c911e99587e4cf1105abea16477 and release/4.3 https://github.com/espressif/esp-idf/commit/0c7d32bcdbb6bab3a3f6cef79af737935cfb892c, feel free to reopen.