espressif/esp-idf #5871: Freertos-MPU

beoran, 2020-09-15:
**Is your feature request related to a problem? Please describe.**

I find it annoying that my whole ESP32 system crashes when one task uses memory incorrectly.

**Describe the solution you'd like**

The ESP32 has a MMU/MPU and a process handler with with 8 levels of PID, which are similar to 8 rings of protection in other architectures. I propose a port of Freertos-MPU to ESP32, which uses PID 0 for the privileged tasks, and PID 2 for the user tasks. The ESP32 MMU can be used to implement the Freertos memory mapping mechanism. The XTENSA CPU exception architecture can then be used to catch exceptions of user processes, and shut down the user task that misbehaved only.

**Describe alternatives you've considered**

Not use the MMU, which is not satisfactory.

igrr, 2020-09-16:
Hi @beoran, indeed establishing memory separation between privileged and non-privileged tasks would be quite useful.
However ESP32 MPU feature is by itself not sufficient to implement isolation between tasks in FreeRTOS. First of all, MPU granularity is not enough to efficiently partition heap and memory between tasks. On the other hand, static partitioning is only viable for simple applications with a small number of tasks, since each task's memory requirements can vary a lot during the execution, and the total memory size is typically less than the sum of worst-case per-task memory requirements.
An additional complication is that once memory is partitioned, sharing data between tasks becomes more complex. Small structures and pointers can be successfully passed through FreeRTOS queues, but larger buffers (such as Wi-Fi driver packet buffers, TCP/IP stack buffers) can only be passed through shared memory. This again leads to a problem that either the heap allocations potentially need to be rounded up to the next MPU page boundary (which would waste a lot of RAM), or that some fixed partitioning of heap would be required (which would not satisfy worst-case memory usage for typical apps).
For the reasons mentioned, we are not planning to pursue FreeRTOS MPU support in the context of ESP32 chip.

(There are other possible applications of ESP32 MPU that are much more viable, such as isolating a smaller app/library within the context of the larger framework/app. So we aren't dismissing this idea completely. However to set the correct expectations, this is not something that will be supported in near term.)

beoran, 2020-09-16:
OK, thanks for explaining in detail why the MPU is insufficient for implementing FREERTOS-MPU. 

This brings me to the next point: I am willing to implement an API for using the esp32 PID/MMU/MPU myself, however, as I understand, currently, esp-idf Freertos (ab)uses some of the MPU registers for other purposes. I would like to ask that this be changed somehow so I can program the esp32 PID/MMU/MPU myself without risking causing problems for my Freedos tasks and queues. 

If even that is too hard, then I would like it to be documented more clearly which MPU registers are used so I don't accidentally overwrite them, or can take precautions to save and reload them whenever needed.

igrr, 2020-09-16:
FreeRTOS doesn't ab(use) MPU registers... I think you might be referring to the fact that the FPU coprocessor state is stored inside xMPU_SETTINGS structure:

https://github.com/espressif/esp-idf/blob/8bc19ba893e5544d571a753d82b44a84799b94b1/components/freertos/xtensa/include/freertos/portmacro.h#L366-L368

However this doesn't prevent the rest of xMPU_SETTINGS from being used for its primary purpose.

beoran, 2020-09-16:
Yes, that is what I am referring to. Thanks for explaining that. 

I looked around more in the esp-idf source code to see how  a null pointer crash is handed, but the focus seems to be on printing an error report and then resetting. 

The xtensa cpu and the esp32 seem to have the capability needed to implement a simple kernel mode/user mode, where at least the kernel is protected, but that is likrly not the focus of the esp-idf project. Hence I will close this issue.

Thanks for your patience.

closed 2020-09-16 by beoran