espressif/esp-idf #5872: TRUNCATED ERROR

RickySyr, 2020-09-16:
/esp-idf/components/bt/bluedroid/osi/config.c:361:54: error: '%d' directive output may be truncated writing between 1 and 5 bytes into a region of size 2
/esp-idf/components/bt/bluedroid/osi/config.c:540:54: error: '%d' directive output may be truncated writing between 1 and 5 bytes into a region of size 2 [-Werror=format-truncation=]

BECAUSE the i is uint16_t, max to 65535, %d for 5 bytes max

## Environment
ESP32

projectgus, 2020-10-11:
Hi @RickySyr ,

Thanks for reporting this. Will fix ASAP.

Angus

projectgus, 2020-10-21:
Hi @RickySyr,

Can you please confirm which version of ESP-IDF you are using, and which toolchain version? This looks like it's only a problem in ESP-IDF v3.2.x and earlier:
https://github.com/espressif/esp-idf/blob/release/v3.2/components/bt/bluedroid/osi/config.c#L362

However this warning (`-Wformat-truncation`) is not part of the toolchain for ESP-IDF v3.2, it was added in ESP-IDF v4.0.

Is it possible you are using the newer toolchain from ESP-IDF v4.x (GCC 8) with an older ESP-IDF version? This is unsupported, and probably there will be other problems as well.

Angus

projectgus, 2020-11-12:
Closing as this only appears to be an issue when using the newer toolchain with older ESP-IDF, and is fixed in all supported versions (v3.2 is End of Life from this month).

closed 2020-11-12 by projectgus