espressif/esp-idf #5875: menuconfig crashes with branch 4.2 while was working on branch 4.0

JulienDevillers, 2020-09-17:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
v4.2-dev-2084-g98d5b5dfd
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Linux
- Using an IDE?: no
- Power Supply:         USB

## Problem Description

This project was building smoothly 7 months ago. It was built using branch 4.0. (sorry I don't have the exact commit number).
I swithched to 4.2 and it does not compile anymore. I tried a menuconfig with : 
`$IDF_PATH/add_path.sh
cd boRemote
python $IDF_PATH/tools/idf.py menuconfig
`
And it leads to the same error than compiling:

`Added to PATH: /home/ju/tools/esp/esp-idf/components/esptool_py/esptool:/home/ju/tools/esp/esp-idf/components/espcoredump:/home/ju/tools/esp/esp-idf/components/partition_table/:/home/ju/tools/esp/esp-idf/tools/
Executing action: menuconfig
Running ninja in directory /home/ju/dev/xxx/Afficheur/embedded/afficheur/build
Executing "ninja menuconfig"...
ninja: error: build.ninja:16842: multiple rules generate signature_verification_key.bin.S [-w dupbuild=err]
`
sdkconfig joined.

Thanks
Julien

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/5236751/sdkconfig.zip)

dobairoland, 2020-09-17:
Hi @JulienDevillers. Thank you for your report.

Could you please try to use the `install.sh` and `export.sh` scripts instead of `add_path.sh` which is the [recommended way to setup the tools](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-3-set-up-the-tools).

And please also run `idf.py fullclean` in your project. Do you still experience the issue? If yes then please give us the full build log. Is your project publicly available?

JulienDevillers, 2020-09-18:
Hi @roland Thanks for your answer.

I don't know if this might help, build not work but neither menuconfig.
I did install.sh then export.sh then menuconfig. Full logs of these operations below.

Sorry, no, the project is not publicly available.

```
ju@ju-ubuntu:~/dev/xxx/Afficheur/embedded/afficheur$ . $IDF_PATH/export.sh
Adding ESP-IDF tools to PATH...
Using Python interpreter in /home/ju/.espressif/python_env/idf4.2_py3.8_env/bin/python
Checking if Python packages are up to date...
Python requirements from /home/ju/tools/esp/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /home/ju/tools/esp/esp-idf/components/esptool_py/esptool
  /home/ju/tools/esp/esp-idf/components/espcoredump
  /home/ju/tools/esp/esp-idf/components/partition_table
  /home/ju/tools/esp/esp-idf/components/app_update
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

ju@ju-ubuntu:~/dev/xxx/Afficheur/embedded/afficheur$ idf.py fullclean
Executing action: fullclean
Done
ju@ju-ubuntu:~/dev/xxx/Afficheur/embedded/afficheur$ idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /home/ju/dev/xxx/Afficheur/embedded/afficheur/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/ju/dev/xxx/Afficheur/embedded/afficheur"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Component directory /home/ju/tools/esp/esp-idf/components/aws_iot does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ju/tools/esp/esp-idf/components/micro-ecc does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ju/tools/esp/esp-idf/components/nimble does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ju/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/ju/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/ju/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/ju/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/ju/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Found PythonInterp: /home/ju/.espressif/python_env/idf4.2_py3.8_env/bin/python (found version "3.8.2") 
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- App "afficheur" version: 1
-- Adding linker script /home/ju/tools/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/ju/tools/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/ju/tools/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/ju/tools/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/ju/tools/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/ju/tools/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/ju/dev/xxx/Afficheur/embedded/afficheur/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/ju/tools/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/ju/tools/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio autogeneratedSafetyBoard bootloader bootloader_support bt cbor coap com console crc cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium lmic log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter time tinyusb u8g2 u8g2_esp32 ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/ju/tools/esp/esp-idf/components/app_trace /home/ju/tools/esp/esp-idf/components/app_update /home/ju/tools/esp/esp-idf/components/asio /home/ju/dev/xxx/Afficheur/embedded/afficheur/components/autogeneratedSafetyBoard /home/ju/tools/esp/esp-idf/components/bootloader /home/ju/tools/esp/esp-idf/components/bootloader_support /home/ju/tools/esp/esp-idf/components/bt /home/ju/tools/esp/esp-idf/components/cbor /home/ju/tools/esp/esp-idf/components [... cut ...] ents/wpa_supplicant /home/ju/tools/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ju/dev/xxx/Afficheur/embedded/afficheur/build
Running ninja in directory /home/ju/dev/xxx/Afficheur/embedded/afficheur/build
Executing "ninja menuconfig"...
ninja: error: build.ninja:16842: multiple rules generate signature_verification_key.bin.S [-w dupbuild=err]

ninja failed with exit code 1
ju@ju-ubuntu:~/dev/xxx/Afficheur/embedded/afficheur$ 
```

JulienDevillers, 2020-09-18:
After removing code to get a miminal code, it seems that  this change solved the issue : 

in CMaleLists.txt :
```
idf_component_register(
	SRCS 
	    ${cFiles}  
	INCLUDE_DIRS "." 
	EMBED_FILES "signature_verification_key.bin"
)
```

changed to:
```
idf_component_register(
	SRCS 
	    ${cFiles}  
	INCLUDE_DIRS "." 
)
```

closed 2020-09-18 by JulienDevillers