espressif/esp-idf #5876: Disconnection of AP in MQTT example ssl_mutual_auth

alexanderabc, 2020-09-17:
Hello guys,

Unfortunately I get a disconnection at random time delay of the esp when I am connected to the AP with MQTT. Maybe this is an error of the esp-idf? Or is it my fault?

I tested the ssl_mutual_auth example and I get the output:

...
TOPIC=/topic/qos0
DATA=data
I (6529) MQTTS_EXAMPLE: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
I (140179) wifi:state: run -> init (ce00)
I (140179) wifi:pm stop, total sleep time: 112679992 us / 138324504 us

I (140179) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:6
I (140189) example_connect: Wi-Fi disconnected, trying to reconnect...
E (140199) TRANS_SSL: ssl_poll_read select error 113, errno = Software caused connection abort, fd = 54
E (140199) MQTT_CLIENT: Poll read error: 0, aborting connection
I (140209) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (140319) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:6
I (140319) wifi:state: init -> auth (b0)
I (140319) wifi:state: auth -> assoc (0)
I (140319) wifi:state: assoc -> run (10)
...

Thank you.

Kind Regards,
Alex

david-cermak, 2020-09-17:
@alexanderabc Hi Alex,

Could you please share the version of ESP-IDF you use? Is the same issue reproducible also on some release branch/tab?

Do these disconnect happen also when running only the example itself or do you run some more complex application?

Could you please enable debug/verbose logging? It would be nice to see what's happening just before getting disconnected. Otherwise, looking at your log, it just looks like a normal WiFi disconnection, resulting in MQTT layer disconnection afterwards.

Thanks,
David

chrta, 2020-09-20:
I am seeing the same (or a similar) issue (disconnection of AP `wifi:state: run -> init (ce00)`) with my project that i updated to the current esp-idf master (as of today):

```
I (1938) SipClient: OK :)  <-- Registering to the SIP server succeeded, no more data traffic afterwards.
I (11368) wifi:state: run -> init (ce00)
I (11368) wifi:pm stop, total sleep time: 0 us / 10330504 us

I (11368) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (11468) wifi:state: init -> init (ce00)
I (11498) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (11498) wifi:state: init -> auth (b0)
I (11498) wifi:state: auth -> assoc (0)
I (11508) wifi:state: assoc -> run (10)
I (11518) wifi:connected with MySSID, aid = 4, channel 1, 40U, bssid = 12:34:56:78:90:10
I (11518) wifi:security: WPA2-PSK, phy: bgn, rssi: -71
I (11518) wifi:pm start, type: 0

I (11568) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (12148) esp_netif_handlers: sta ip: 192.168.170.164, mask: 255.255.255.0, gw: 192.168.170.1
I (12148) main: got ip:192.168.170.164
I (24038) wifi:state: run -> init (ce00)
```

My code is https://github.com/chrta/sip_call/tree/3392eac0f54ad9b779994cd62d3f655c75c3e114 (with that revision and without any power-safe-mode enabled).

Before updating my code to the current esp-idf master it ran fine and i did not have these disconnects.

chrta, 2020-09-20:
For my deauth issue i did a git bisect (which was not a nice experience, since a lot of commits did not build, ./install.sh or . ./export.sh did not work):
The offending commit is https://github.com/espressif/esp-idf/commit/7cf59a8d185edecc7e000321d90c6ae76e8dd9e9 The previous commit on master works fine.

alexanderabc, 2020-09-21:
Hello,

IDF-Version: v4.2-dev-2084-g98d5b5dfd

I just adjusted the code with: 
const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = URL,
        .event_handle = mqtt_event_handler,
	.cert_pem = (const char *)aws_root_ca_crt_start,
	.client_id = Client_ID,
        .client_cert_pem = (const char *)client_cert_pem_start,
        .client_key_pem = (const char *)client_key_pem_start,
	.disable_auto_reconnect = true
}
I think there should be no problems...?

If I disable MQTT there is no error, does this make sense :)?

This is the debug output:

V (28103) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (29103) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (30103) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (31103) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
I (31643) wifi:state: run -> init (ce00)
I (31643) wifi:pm stop, total sleep time: 21301691 us / 28926080 us

I (31643) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:6
I (31653) example_connect: Wi-Fi disconnected, trying to reconnect...
E (31653) TRANS_SSL: ssl_poll_read select error 113, errno = Software caused connection abort, fd = 54
E (31663) MQTT_CLIENT: Poll read error: 0, aborting connection
D (31673) MQTT_CLIENT: Reconnect after 10000 ms
I (31683) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (31783) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:6
I (31783) wifi:state: init -> auth (b0)
I (31783) wifi:state: auth -> assoc (0)
I (31783) wifi:state: assoc -> run (10)
I (31813) wifi:connected with WIFI-123456, aid = 97, channel 1, BW20, bssid = 12:34:56:78:90:10
I (31813) wifi:security: WPA2-PSK, phy: bgn, rssi: -50
I (31813) wifi:pm start, type: 1

I (31853) wifi:AP's beacon interval = 104448 us, DTIM period = 1
I (32553) esp_netif_handlers: example_connect: sta ip: xx.xx.xx.xx, mask: 255.255.240.0, gw: 10.100.0.1
I (32553) example_connect: Got IPv4 event: Interface "example_connect: sta" address: xx.xx.xx.xx
I (33553) example_connect: Got IPv6 event: Interface "example_connect: sta" address: xx:xx:xx:xx type: ESP_IP6_ADDR_IS_LINK_LOCAL

alexanderabc, 2020-09-23:
Also this output occurs in another project:

V (00:07:38.898) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (00:07:39.898) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (00:07:40.898) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (00:07:41.898) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
I (463643) wifi:state: run -> init (ce00)
I (463653) wifi:pm stop, total sleep time: 369217957 us / 459715630 us

I (463653) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:6
D (00:07:42.232) event: running post WIFI_EVENT:5 with handler 0x400ef5f0 and context 0x3ffca820 on loop 0x3ffc0d8c
0x400ef5f0: wifi_default_action_sta_disconnected at C:/esp-idf/components/esp_wifi/src/wifi_default.c:103

D (00:07:42.234) esp_netif_handlers: esp_netif action disconnected with netif0x3ffcac34 from event_id=5
D (00:07:42.246) esp_netif_lwip: check: remote, if=0x3ffcac34 fn=0x400e8a1c
0x400e8a1c: esp_netif_down_api at C:/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1135

D (00:07:42.258) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffcac34
D (00:07:42.261) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffcacb4
D (00:07:42.263) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffcac34
D (00:07:42.275) esp_netif_lwip: if0x3ffcac34 start ip lost tmr: interval=120
D (00:07:42.287) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffcac34
D (00:07:42.288) esp_netif_lwip: if0x3ffcac34 start ip lost tmr: already started
D (00:07:42.300) esp_netif_lwip: check: local, if=0x3ffcac34 fn=0x400e9258
0x400e9258: esp_netif_update_default_netif_lwip at C:/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:163

D (00:07:42.301) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffcac34
V (00:07:42.312) esp_netif_objects: esp_netif_next_unsafe 0x0
V (00:07:42.314) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffcac34
V (00:07:42.325) esp_netif_objects: esp_netif_next_unsafe 0x3ffcac34
V (00:07:42.326) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffca580
V (00:07:42.338) esp_netif_objects: esp_netif_next_unsafe 0x3ffca580
D (00:07:42.349) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (00:07:42.350) event: running post WIFI_EVENT:5 with handler 0x400da650 and context 0x3ffd4a58 on loop 0x3ffc0d8c
0x400da650: wifi_st_wifi_event_handler at c:\work\test\build/../components/comm/src/st_event_handler.c:15

E (00:07:42.352) TRANS_SSL: ssl_poll_read select error 113, errno = Software caused connection abort, fd = 54
E (00:07:42.376) MQTT_CLIENT: Poll read error: 0, aborting connection
D (00:07:42.389) MQTT_CLIENT: Reconnect after 10000 ms
V (00:09:42.267) esp_netif_objects: esp_netif_next_unsafe 0x0
D (00:09:42.268) esp_netif_lwip: esp_netif_ip_lost_timer esp_netif:0x3ffcac34
D (00:09:42.269) esp_netif_lwip: if0x3ffcac34 ip lost tmr: raise ip lost event

alexanderabc, 2020-09-23:
And this one occurs while subscribing:

D (00:00:06.995) MQTT_CLIENT: mqtt_enqueue id: 44449, type=8 successful
D (00:00:06.996) OUTBOX: ENQUEUE msgid=44449, msg_type=8, len=79, size=79
D (00:00:07.011) MQTT_CLIENT: Sent subscribe topic=$aws/things/xx/shadow/update/delta, id: 44449, type=8 successful
E (00:00:07.031) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=0
E (00:00:07.033) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
D (00:00:07.048) MQTT_CLIENT: Reconnect after 10000 ms

Thank you ;)

david-cermak, 2020-09-29:
@chrta @alexanderabc Thanks for providing more details and finding the offending commit. Can you please share the brand and the model of your AP?
It has been confirmed that some routers don't comply with WiFi standards (specifically related to the TFS field in beacon packets) so that we are experiencing random disconnects on such devices after the 7cf59a8 been merged/backported. This issue has been reported on ASUS N66U router, so far.

chrta, 2020-09-29:
I use the following setup:
- https://mikrotik.com/product/hex_s firmware version 6.45.8 (as CAPsMAN manager)
- https://mikrotik.com/product/cap_ac firmware version 6.45.8 (managed by CAPsMAN)

There are A LOT of settings possible. Most of them are left in default, but i can change them if you provide some hints.

AxelLin, 2020-09-30:
> @chrta @alexanderabc Thanks for providing more details and finding the offending commit. Can you please share the brand and the model of your AP?
> It has been confirmed that some routers don't comply [...]

You can try to  check disconnected reason once you got WIFI_EVENT_STA_DISCONNECTED:

wifi_event_sta_disconnected_t *disconnected = (wifi_event_sta_disconnected_t *) event_data;
ESP_LOGI(TAG,"Disconnect reason: %d", disconnected->reason);

chrta, 2020-09-30:
I cannot reproduce this problem any more. The current esp-idf master seems to work fine now. The only thing that changed on my side is that i compiled and flashed from another PC. The router and AP did not reboot since the problem appeared. Maybe it was some issue with my development environment/esp-idf clone.

AxelLin, 2020-10-19:
> I (463643) wifi:state: run -> init (ce00)
> I (463653) wifi:pm stop, total sleep time: 369217957 us / 459715630 us

This disconnect reason is WIFI_REASON_AP_TSF_RESET.
This is similar to https://github.com/espressif/esp-idf/issues/5915
What is your router/ap model?

AxelLin, 2020-10-19:
> @chrta @alexanderabc Thanks for providing more details and finding the offending commit. Can you please share the brand and the model of your AP?
> It has been confirmed that some routers don't comply [...]

I might misunderstand, do you mean no such disconnect issue before 7cf59a8 ?
It seems there are several models of AP have such issue (https://github.com/espressif/esp-idf/issues/5915),
so I'm not sure if it's AP's problem or ESP32's problem.

espxiehang, 2021-03-31:
Hi @chrta @alexanderabc , 
Abort this issue `I (31643) wifi:state: run -> init (ce00)`
We have solved this issue. Fix commit is  3c3bff6eb

Alvin1Zhang, 2021-04-14:
Thanks for reporting, feel free to reopen.

closed 2021-04-14 by Alvin1Zhang

rafalino75, 2021-07-10:
> Hi @chrta @alexanderabc ,
> Abort this issue `I (31643) wifi:state: run -> init (ce00)`
> We have solved this issue. Fix commit is [3c3bff6](https://github.com/espressif/esp-idf/commit/3c3bff6ebb67ddc734d078d5549a09066e8df032)

Was this fix merged into release 4.3?

ogghst, 2022-01-23:
i have the same error, the disconnect reason is WIFI_REASON_ASSOC_LEAVE. this happens only when i output a lot of info in the serial for debugging purposes. can the reason be in a task hanging?

AxelLin, 2022-01-24:
> i have the same error, the disconnect reason is WIFI_REASON_ASSOC_LEAVE. this happens only when i output a lot of info in the serial for debugging purposes. can the reason be in a task hanging?

@ogghst 
Please create a new issue with detail log and your esp-idf version (git describe --tags)

nitinGangurde, 2023-09-05:
E (100120) TRANS_TCP: tcp_poll_read select error 113, errno = Software caused connection abort, fd = 54
E (100120) MQTT_CLIENT: Poll read error: 119, aborting connection
I (100130) MQTT_TEST: MQTT_EVENT_DISCONNECTED
I (100130) MQTT_CLIENT: Client force reconnect requested
I (100140) MQTT_TEST: Other event id:7

I am using esp-idf version 4.3.4

nitinGangurde, 2023-09-05:
Please provide me the solution