espressif/esp-idf #5877: Changing certificate bundle type all/common doesn't change app size

JakubRakus, 2020-09-17:
## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version v4.2-dev-2084-g98d5b5dfd
- Build System: idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Linux

## Problem Description

I'd like to change cerificate bundle form all to most common to save some space, but it doesn't change size of overall app. `idf.py size` output when compiled with full bundle:
```
 DRAM .data size:   42628 bytes
 DRAM .bss  size:   73560 bytes
Used static DRAM:  116188 bytes (  64548 available, 64.3% used)
Used static IRAM:  102295 bytes (  28777 available, 78.0% used)
      Flash code:  693667 bytes
    Flash rodata:  188004 bytes
Total image size:~1100154 bytes (.bin may be padded larger)
```
and with most commonly used:
```
DRAM .data size:   42628 bytes
 DRAM .bss  size:   73560 bytes
Used static DRAM:  116188 bytes (  64548 available, 64.3% used)
Used static IRAM:  102295 bytes (  28777 available, 78.0% used)
      Flash code:  693667 bytes
    Flash rodata:  188004 bytes
Total image size:~1100154 bytes (.bin may be padded larger)
```
My partition table: 1100K for factory app, 1100K + 1100K for OTA images, 528K for FatFS

Alvin1Zhang, 2020-09-18:
Thanks for reporting, we will look into.

ESP-Marius, 2020-09-29:
Hi @JakubRakus 

The certificate bundle is only pulled into your build if it's actually used (i.e. configured in the `esp_tls_cfg_t` struct). So if is not used then changing that option wont change the size.

For reference here is the sizes from `https_request_example`:

Full bundle:

```
DRAM .data size:   12796 bytes
DRAM .bss  size:   15904 bytes
Used static DRAM:       0 bytes (      0 available, nan% used)
Used static IRAM:       0 bytes (      0 available, nan% used)
Used stat D/IRAM:  101455 bytes ( 275377 available, 26.9% used)
Flash code:  525599 bytes
Flash rodata:  172688 bytes
Total image size:~ 799742 bytes (.bin may be padded larger)
```

Common:
```
DRAM .data size:   12796 bytes
DRAM .bss  size:   15904 bytes
Used static DRAM:       0 bytes (      0 available, nan% used)
Used static IRAM:       0 bytes (      0 available, nan% used)
Used stat D/IRAM:  101455 bytes ( 275377 available, 26.9% used)
Flash code:  525599 bytes
Flash rodata:  123960 bytes
Total image size:~ 751014 bytes (.bin may be padded larger)
```

AdityaHPatwardhan, 2020-09-29:
@JakubRakus can you please send you sdkconfig file in both the cases.
When I checked, I can see that the app size is changed as it should.
can you also send the exact size of `x509_crt_bundle` for cases, which may be obtained by following command ( after building the project)
```
ls -l build/esp-idf/mbedtls/x509_crt_bundle
```

JakubRakus, 2020-09-30:
OK, thanks for clarifying, I get it. But now it leads to another problem (which is connected with issue  #5504 I reported some time ago). Example code in `https_request_example` uses TLS transport related functions directly, but in my code I'd rather want to use an `http_clent` set of functions and don't touch underlying transport internals.

Struct `esp_http_client_config_t` have flag `use_global_ca_store` - I wrongly assumed that setting it to true should lead to  use certificate bundle automatically, beacuse we have this in `esp_http_client_init()`:

```
if (config->use_global_ca_store == true) {
    esp_transport_ssl_enable_global_ca_store(ssl);
}
```

Diving deeper into the code, this leads to set `use_global_ca_store` flag in transport config (type `esp_tls_cfg_t).` And it looks this not doing anything else - certificate bundle is still not attached, and invoking `esp_http_client_open()` ends with error - no certificate found.

So, now the question is: how to make this:

```
esp_tls_cfg_t cfg = {
            .crt_bundle_attach = esp_crt_bundle_attach,
        };
```
like in `https_request_example` when I only have access to structs `esp_http_client_config_t` and `esp_http_client_handle_t`. There is `transport` pointer in client handle, but I can see no easy way to change it's config - is there any option to do this?

boarchuz, 2020-09-30:
An example showing usage of the certificate bundle with esp_http_client would be a great addition.

vbvchauthmal, 2020-10-05:
> OK, thanks for clarifying, I get it. But now it leads to another problem (which is connected with issue #5504 I reported some time ago). Example code in `https_request_example` uses TLS transport [...]

I also assumed that setting `.use_global_ca_store = true` in `esp_http_client_config_t` structure will help with my project module and ESP HTTPS client can get server 's public certificate for secure connection establishment over SSL. My project's server certificate is issued from letsencrypt.

With `.use_global_ca_store = true` I am getting following error log :
```
I (5229) [HTTP-CLIENT]: Connected to AP, begin http_task
I (5229) [HTTP-CLIENT]: ==> https_request
D (5239) HTTP_CLIENT: Begin connect to: https://my_server_IP:443
D (5249) esp-tls: host:my_server_IP: strlen 17
D (5369) phy_init: wifi mac time delta: 42273
D (5469) phy_init: wifi mac time delta: 96217
D (5569) phy_init: wifi mac time delta: 96273
E (5589) esp-tls: global_cacert is NULL
D (5589) esp-tls: create_ssl_handshake failed
E (5589) esp-tls: Failed to open new connection
E (5589) TRANS_SSL: Failed to open a new connection
E (5599) HTTP_CLIENT: Connection failed, sock < 0
E (5599) [HTTP-CLIENT]: Error perform https request ESP_ERR_HTTP_CONNECT
I (5609) [HTTP-CLIENT]: HTTP_EVENT_DISCONNECTED
I (5609) [HTTP-CLIENT]: http_task completed
```
My environment is ESP-IDF version : `v3.3.1 (commit id : 143d26aa49df524e10fb8e41a71d12e731b9b71d)`

boarchuz, 2020-10-09:
Having had a look at this today, I now understand that the global CA store and the certificate bundle are completely different things.
The global CA store allows you to set a text-formatted PEM certificate list to be used globally by all esp_http_clients; the bundle is the optional binary-formatted list which is currently incompatible with esp_http_client.

For demonstration, there are changes here - https://github.com/boarchuz/esp-idf/tree/http_client_bundle - that will allow you to set use_cert_bundle=true in esp_http_client_config_t.

JakubRakus, 2020-10-15:
@ESP-Marius @Alvin1Zhang  Any thoughts on (or maybe possibilities to implement/merge) changes made by @boarchuz in his fork? It looks promising.

ESP-Marius, 2020-10-16:
@JakubRakus Implementing it this way would cause a big increase in app size even if you didnt set `use_cert_bundle` which would be unfortunate. 

But we could probably implement it the same way it's current done in the other example, by letting the user pass a pointer to attach. Less user-friendly though...

AdityaHPatwardhan, 2021-04-15:
Hi @JakubRakus The crt bundle functionality has been enabled for the `esp_http_client`, can you please check once ?
If it looks good to you then I think you may close down this issue.

mahavirj, 2021-04-20:
Fixed with https://github.com/espressif/esp-idf/commit/f66bdf1233203d6edb7b7208c6fb898321654931

closed 2021-04-20 by mahavirj