espressif/esp-idf #5879: After cpu_start during device startup, ESP32 stops executing function and prints out garbage

jonhunterking66, 2020-09-17:
## Environment

- Development Kit:     none
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
v4.1-beta2-141-g84b51781c
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: VS code
- Power Supply: Battery

## Problem Description

I have several boards running my firmware (about 10). Occasionally (once a week) these boards stop executing their code. I can diagnose this because the board will connect to wifi and push data to our backend. We take a timestamp each time this handshake occurs (every five minutes) so we can get a sense of what our units are doing. When the ESP32 is not polling wifi it sleeps for 5 minutes. 

However, sometimes they just drop out infinitely and stop executing their code. We have set up several monitored units to catch the output when this occurs and this is what we caught. It occurred after a sleep period during startup. 

Has anyone seen something like this before? Is there anyway I can diagnose further to understand what is happening?

## Debug Logs

```   
I (12) boot: ESP-IDF v4.1-dirty 2nd stage bootloader
I (12) boot: compile time 09:05:57
I (12) boot: chip revision: 1
I (14) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (21) boot.esp32: SPI Speed      : 40MHz
I (26) boot.esp32: SPI Mode       : DIO
I (30) boot.esp32: SPI Flash Size : 4MB
I (35) boot: Enabling RNG early entropy source...
I (40) boot: Partition Table:
I (44) boot: ## Label            Usage          Type ST Offset   Length
I (51) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (58) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (66) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (73) boot:  3 ota_0            OTA app          00 10 00020000 00150000
I (81) boot:  4 ota_1            OTA app          00 11 00170000 00150000
I (88) boot: End of partition table
I (92) boot_comm: chip revision: 1, min. application chip revision: 0
I (99) esp_image: segment 0: paddr=0x00170020 vaddr=0x3f400020 size=0x384b0 (230576) map
I (196) esp_image: segment 1: paddr=0x001a84d8 vaddr=0x3ffbdb60 size=0x042e4 ( 17124) load
I (204) esp_image: segment 2: paddr=0x001ac7c4 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/TonyWu/Desktop/esp-idf/components/freertos/xtensa_vectors.S:1778

I (204) esp_image: segment 3: paddr=0x001acbd0 vaddr=0x40080404 size=0x03448 ( 13384) load
I (218) esp_image: segment 4: paddr=0x001b0020 vaddr=0x400d0020 size=0xef52c (980268) map
0x400d0020: _stext at ??:?

I (594) esp_image: segment 5: paddr=0x0029f554 vaddr=0x4008384c size=0x1a878 (108664) load
0x4008384c: r_ea_alarm_set at ??:?

I (641) esp_image: segment 6: paddr=0x002b9dd4 vaddr=0x400c0000 size=0x00064 (   100)
I (658) boot: Loaded app from partition at offset 0x170000
I (658) boot: Disabling RNG early entropy source...
I (658) cpu_start: Pro cpu up.
I (662) cpu_start: Application information:
I (667) cpu_start: Project name:     oto-template
I (672) cpu_start: App version:      f8f79036
I (677) cpu_start: Compile time:     Sep 16 2020 17:07:27
I (683) cpu_start: ELF file SHA256:  6971aecc21d58d35...
I (689) cpu_start: ESP-IDF:          v4.1-beta2-141-g84b51781c
I (696) cpu_start: Starting app cpu, entry point is 0x400814b4
0x400814b4: call_start_cpu1 at C:/Users/TonyWu/Desktop/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (706) heap_init: Initializing. RAM available for dynamic allocation:
I (713) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (719) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (725) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (731) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (737) heap_init: At 3FFCEB18 len 000114E8 (69 KiB): DRAM
I (744) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (750) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (756) heap_init: At 4009E0C4 len 00001F3C (7 KiB): IRAM
I (762) cpu_start: Pro cpu start user code
I (781) spi_flash: detected chip: generic
I (781) spi_flash: flash io: dio
I (782) cpu_start: S�`���~����f` �����~��������� �~������f~`�~���f�������~��~���`�~��x��`��~�f�~�xf��fx`�� �f��f��x������~f���~��~���������x��`�����`���fx�fx`��~��f��~��~���`�~����� �~���f�~�`����f����`���~����f` �����~��������� �~������f~`�~���������`�f������ff�������f��f����`���~����f` �����~��������� 
```

projectgus, 2020-09-18:
Hi @jonhunterking66,

Thanks for a clear description of the problem you're seeing.

This looks like it's most likely a hardware issue rather than a software issue. I noticed that your power source is a battery and you're using a module on a custom device, does the design follow the [ESP32 Hardware Design Guidelines](https://www.espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf) closely? Is it possible the main power supply is unstable, or has insufficient capacitance to maintain a stable voltage during high bursts of current draw?

Suggest also checking the brownout detector configuration. The default level is the minimum (level 0, ~2.43V) but you should be able to increase this and you may find that a BOD reset is triggered rather than unexpected behaviour:
https://docs.espressif.com/projects/esp-idf/en/v4.1/api-reference/kconfig.html#config-esp32-brownout-det

(The long term fix for this probably still a hardware fix, though.)

Finally, if you can then it's strongly recommended to update the ESP-IDF `v4.1` final release now that it's available. Although I don't expect this will fix the problem, it's possible that the particular release branch commit you're using has some kind of bug.

jonhunterking66, 2020-09-18:
@projectgus thank you for the feedback! 

I will update to 4.1 and review the documents you have supplied. 

Could you explain a little bit more about why unstable power could cause this problem? Since we are not seeing brownout warnings in the debug output I would expect it to be something else.

jonhunterking66, 2020-09-18:
@projectgus I also wanted to clarify that we are using a WROOM-32D on our custom board. This makes me feel like its unlikely a layout issue. Unless you expect it is circuitry unrelated to the module. Most of the ESP32 Hardware design guidelines relate to layouts of standalone SoCs.

jonhunterking66, 2020-09-18:
@projectgus I have updated to v4.1-273-g94cb7e8b8

jonhunterking66, 2020-09-18:
@projectgus any chance this could be a memory issue?

projectgus, 2020-09-21:
Hi @jonhunterking66,

> Could you explain a little bit more about why unstable power could cause this problem? Since we are not seeing brownout warnings in the debug output I would expect it to be something else.

Of course. Some key things include:

* The default brownout detector level is very conservative, so it is possible that a short voltage drop caused by power supply instability will push the voltage below the specified range but not trigger the detector. Increasing the level should help rule this out.
* It is also possible for short power supply faults to occur that don't trigger the brownout detector.
* Electrical noise in general can cause interference in other components inside the module, for example the communication lines between the flash chip and the ESP32.

>  I also wanted to clarify that we are using a WROOM-32D on our custom board. This makes me feel like its unlikely a layout issue. Unless you expect it is circuitry unrelated to the module. Most of the ESP32 Hardware design guidelines relate to layouts of standalone SoCs.

Apologies, I should have linked you to the WROOM datasheet which has some layout & supporting hardware guidelines.

It is still possible for supporting hardware to cause issues - particularly if power supply ever drops below 3.0V, or if there's not enough power supply capacitance (and/or layout issues with placement of capacitors), and/or electrical noise and EMI.

> I have updated to v4.1-273-g94cb7e8b8

Were you able to reproduce the problem with this version?

FWIW, this is a still a "release branch" - specifically it has 273 commits made after v4.1 which will be included in v4.1.1. To get the exact stable release you would run `git checkout v4.1`. I know the versioning is a bit complex to understand, and some of the names are conterintuitive - there is a [longer explanation here](https://docs.espressif.com/projects/esp-idf/en/stable/versions.html).

> any chance this could be a memory issue?

It's not impossible, however I'd expect more consistent failures in this case.

It looks like the UART baud rate goes bad just about the time that the `app_main()` routine in the firmware is scheduled to run. (The UART has a hardware FIFO buffer and a software buffer so execution is usually a short time ahead of where the UART log is.)

* What does your firmware do in the app_main() routine? Is there anything there which might cause an intermittent issue if it does corrupt memory or similar?
* Which peripheral(s) and/or peripheral drivers does your firmware enable/use?
* Do you have any power management features enabled, apart from using deep sleep? Are you able to share the project sdkconfig file, please?

Thanks.

jonhunterking66, 2020-09-21:
@projectgus 

We ran several boards with our firmware over the weekend all based on 4.1. Several boards repeated the same issue. I also ran the code on a dev board to see if I could recreate the problem. It did occur on the dev board so I feel like we can rule out a hardware issue. 

To answer your follow up questions: 

> What does your firmware do in the app_main() routine? Is there anything there which might cause an intermittent issue if it does corrupt memory or similar?

```
void app_main(void)
{
  xTaskCreate(&read_task, "read", 1024*6, NULL, 1, &read_handle);
  vTaskSuspend(read_handle);
  xTaskCreate(&main_task, "MAIN", 1024 *40, NULL, 2, &main_handle);
}
```

The read task reads an adc value and performs two i2c reads. The main task is huge. It runs a state machine that performs several functions that relate to BLE, wifi and motor control. There are several places that I could imagine memory issues. The Wifi and BLE stack are both fairly complex and its possible there is memory issues there. 

>Which peripheral(s) and/or peripheral drivers does your firmware enable/use?

ADC, BLE, Flash, GPIOs, I2C, PWM, OTA, sleep, and WiFi

>Do you have any power management features enabled, apart from using deep sleep? Are you able to share the project sdkconfig file, please?

Nothing that I can note. I will add the sdkconfig to this comment.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5255643/sdkconfig.txt)

jonhunterking66, 2020-09-22:
@projectgus Is there any other information I can give to help with this?

jonhunterking66, 2020-09-23:
I have caught more instances of this occurring. These took place after sleep started: 
[image]

jonhunterking66, 2020-09-23:
@projectgus we have video of the error happening, it does not sleep for the 5 minutes as I expected but starts printing out garbage ~10 seconds after entering sleep.

projectgus, 2020-09-24:
Hi @jonhunterking66,

Thanks for the extra info and for being patient while I got back to you.

I agree that if you can reproduce on a dev board then it's probably not hardware related.

Do I understand correctly that there are two slightly different failure patterns that you are seeing:

* Device goes to deep sleep, wakes up (after an unknown amount of time?), normal boot log is printed until some time during startup but before booting finishes the UART output becomes corrupt. (This is the failure shown in the first post.)
* Device appears to go into 5 minute deep sleep but spontaneously prints corrupted UART output 10 seconds later. (This is the failure shown two posts above.)

Do you have access to a logic analyzer that you can use to capture the UART output? The corrupted output may be output at `baudrate/2` (as the APB clock is halved when going into deep sleep), or it may be at some other random baud rate. Capturing it and seeing if there is anything decipherable there may give additional clues as to where it's getting stuck.

Otherwise, the best possible next step would be if we can reproduce this ourselves. Do you think it would it be possible for you to remove parts of your firmware iteratively by trial and error to try and come up with a minimal test case that you could send us? Or are the failures too infrequent for this?

jonhunterking66, 2020-09-24:
@projectgus no worries. I really appreciate you getting back to me. 

Yes, there are two different instances where we have seen this: 

1. The device sleeps for 5 minutes, normal boot log is printed until cpu_start in which case it becomes corrupted
2. The device goes into the 5 minute deep sleep, but starts printing out garbage 10 seconds after entering deep sleep. 

The most confusing part for me is that if it is just UART corruption then why does the code stop executing properly? 

I will capture the baud rate to see if we can get any clues from there. In addition I will simplify the code to try and see if I can recreate this. Maybe I can start by taking out the WiFi related stuff? Not sure if you have any suggestions about simplifications.

I might be able to get you a code copy as well but I would have to speak to my manager about that as this code is being used on a product.

jonhunterking66, 2020-09-24:
@projectgus I can also confirm that the baud rate during the garbage outputting state is half. 57000 during garbage output, 115200 during normal operation

projectgus, 2020-09-25:
Hi @jonhunterking66 ,

> I can also confirm that the baud rate during the garbage outputting state is half. 57000 during garbage output, 115200 during normal operation

OK, this is very useful as it confirms the APB clock rate is somehow being set back to the XTAL value during the boot (it should be 80MHz but it's being set to 40MHz). Does the same also during the "false sleep" state?

If you look at this half-rate UART output do you see anything useful like a crash log, or does it look like normal execution apart from the bad baud rate?

>      The most confusing part for me is that if it is just UART corruption then why does the code stop executing properly?

If the APB clock is reverting to 40MHZ instead of 80MHz then a lot of things may be trying to work but failing (for example Wi-Fi will fail to initialize the RF subsystem properly and you'll probably get a Wi-Fi error, other peripherals may also be working but at half the configured speed.) The UART is probably just one symptom.

> Maybe I can start by taking out the WiFi related stuff? Not sure if you have any suggestions about simplifications.

This is a bit of a guessing game to be honest, as we've not seen this behaviour before and we don't know how to reproduce it except that you can reproduce it on your firmware. My guess is that some combination of things that are running when going to deep sleep (maybe drivers, especially drivers which are processing data in ISRs, plus possibly also Wi-Fi) are not fully resetting like they should and this is causing the bug somehow.

If you're confident enough about reproducing on your current firmware that you can disable Wi-Fi entirely and see if the problem goes away or not, this would be a very useful piece of information. And/or one by one not initialising certain drivers and seeing if the problem goes away?

jonhunterking66, 2020-09-25:
@projectgus 

>OK, this is very useful as it confirms the APB clock rate is somehow being set back to the XTAL value during the boot (it should be 80MHz but it's being set to 40MHz)

So just to be clear the 40Mhz clock speed is not expected during sleep? 

>If you look at this half-rate UART output do you see anything useful like a crash log, or does it look like normal execution apart from the bad baud rate

Do you mean if I was able to read the baud rate at 40Mhz? I have not tried to convert the garbage output to logs. Any suggestion on how I could do that? 

I can definitely try remove part of the code. Wifi is a big one that is used a lot. The code would pretty drastically change however. 

Would there be value for you seeing the code? I would need you to sign an NDA. Not sure if this ever happens but if that would be helpful let me know. 

I will try removing the Wifi stack.

projectgus, 2020-09-28:
> So just to be clear the 40Mhz clock speed is not expected during sleep?

When the chip is actually going into deep sleep the APB clock is set to 40MHz. This is half the usual value during normal operation. So this part is expected.

The unexpected part is that the chip shouldn't be doing anything externally visible at this time. If you're seeing UART output at half the configured  baud rate then the most likely direct cause is that the UART is outputting while the APB clock is set to 40MHz not the normal 80MHz (the UART baud rate is divided from the APB clock.)

>      Do you mean if I was able to read the baud rate at 40Mhz? I have not tried to convert the garbage output to logs. Any suggestion on how I could do that?

If the "garbage" output is output at 57600bps (ie half the configured 115200) then you should be able to use a logic analyzer capture to do this, or run a serial program at 57600bps while the program is running. The already-captured serial logs probably can't be decoded,

> I can definitely try remove part of the code. Wifi is a big one that is used a lot. The code would pretty drastically change however.
>
> Would there be value for you seeing the code? I would need you [...]

Normally we try to find a more minimal test case to reproduce, and work from this. So if you are able to try removing parts of the firmware to narrow it down then this is ideal, thanks.

jonhunterking66, 2020-09-30:
To clarify further, a APB clock speed of 40Mhz corresponds to a 57600bps on UART? 

We determined that the UART was outputting at 57600bps using an oscilloscope.

I will try removing wifi from the firmware to test this further.

igrr, 2020-09-30:
@jonhunterking66 Can you check what exactly UART is outputting? You can do that by running e.g. `miniterm.py PORT_NAME 57600` — the output printed at 115200 will be shown as garbage, but the output printed at 57600 will be visible.

jonhunterking66, 2020-09-30:
@igrr Could I get a little more guidance on this? 

Are you suggesting that once I see the garbage output at 57600, stop the idf.py monitoring and run this miniterm.py script? 

Thanks for getting involved and helping out!

jonhunterking66, 2020-09-30:
@igrr just started messing around with miniterm.py. Its awesome. The idf.py monitor call resets the unit but this just picks up where the unit is running. 

I feel like this is going to be very useful. I will let you guys know what I find.

projectgus, 2020-09-30:
> To clarify further, a APB clock speed of 40Mhz corresponds to a 57600bps on UART?

I'll explain this connection in detail, as there are a few different variables here:

* The UART baudrate is calculated as `(clock source / divider)` where both the clock source and the exact divider value are configured from registers (more details in ESP32 Technical Reference Manual).
* When power management is disabled (as it is in your sdkconfig), the clock source is the APB clock
* The APB clock runs at 80MHz during normal app execution
* So the divider value will be approx 694.4 (`80MHz/115200`)
* If the baud rate spontaneously changes from 115200 to 57600 then there are two plausible root causes: the clock source has halved in frequency, or the UART divider register has doubled in value.

We know that when we head into deep sleep the APB clock goes from 80MHz to 40MHz as we disable the internal PLL and revert the APB to the bare crystal frequency (40MHz). When the system resets out of deep sleep it should reconfigure the PLL and the APB clock goes back to 80MHz (side note: this also happens on first boot, the bootloader starts off running at the crystal frequency and then configures the clocks during startup.)

So although we can't say for sure that this is why the UART baud rate is spontaneously half the normal value, or what is triggering the bug, it's the most likely internal behaviour that's responsible for what you're seeing.

Hopefully capturing the "garbage" output with miniterm gives some clues.

jonhunterking66, 2020-10-01:
@igrr thank you for the clarification. I will try and capture a failure again while using miniterm.py

jonhunterking66, 2020-10-05:
@projectgus @igrr I have not been able to recreate the issue. I recently restructured my WiFi library to deal with some heap losses. I did not think this would fix the problem as I deep sleep between every WiFi poll, and it seems the heap was back to full size at the start of every cycle. 

Not convinced yet that it is entirely fixed until I have run it for about a week. Let me know if any of this information helps lead to a conclusion.

projectgus, 2020-12-01:
@jonhunterking66 How did you go running your updated firmware?

jonhunterking66, 2020-12-01:
Its been going great. I have not seen it since I restructured my WiFi library to deal with heap losses. I have no idea if they were related but I have not seen it since.

projectgus, 2020-12-01:
Thanks for the update. In that case I'm going to close this as unreproducible, but if you come across this issue again then please let us know.

closed 2020-12-01 by projectgus