espressif/esp-idf #5880: libsodium api

i0x71, 2020-09-17:
Hi, are there any docs on libsodium support or usage ?
Thanks

projectgus, 2020-09-18:
Hi @i0x71,

The libsodium is upstream libsodium v1.0.12 more or less as-is, so you should be able to use anything in the [libsodium API documentation](https://doc.libsodium.org/) and compile any C code examples that use libsodium.

I note that our version has gotten out of date, will fix this ASAP. However the [ChangeLog](https://github.com/jedisct1/libsodium/blob/master/ChangeLog) notes no security issues and only minor API changes to low-level APIs.

Where possible, it's generally better to use mbedTLS or possibly [wolfSSL](https://github.com/espressif/esp-wolfssl) as these have a lower resource footprint and can make better use of hardware acceleration features.

However there are some algorithms which are not supported in these libraries, which is why we currently bundle libsodium as well.

i0x71, 2020-09-18:
Thanks alot for a detailed and quick response!
My goal was to use these libraries to sign messages with private key, i will give wolfssl and mbed a try.

Thanks

projectgus, 2020-09-18:
Good to hear.

For future reference, a good place to ask questions about ESP-IDF is the https://esp32.com forums.

closed 2020-09-18 by projectgus