espressif/esp-idf #5882: Support OpenXC and Enter Automoble Industry

keelung-yang, 2020-09-18:
Is possible to support OpenXC?
Maybe it's impossible since espressif is escaping from CAN and there is no accurate timestamp in CAN message #4527 
But ESP32 have great coast advantage and all enough components to support OpenXC.

And what's about entering automobile industry officially?
And what's about releasing a new ESP32 chip for using it as an ECU in a car?
There are more and more ECUs in a modern car, more and more cars on roads. 
Think about it: ESP32 chips are running in more and more cars.

About OpenXC
> OpenXC™ is an open source, data-focused API for your car. By installing a small hardware module, the vehicle data becomes accessible to Android or other desktop applications using the OpenXC library.

http://openxcplatform.com/
http://openxcplatform.com/vehicle-interface/hardware.html

igrr, 2020-09-18:
Hi @keelung-yang, regarding OpenXC support, we will evaluate the request and get back to you.

Regarding the 2nd batch of chip and business related questions, I'm afraid they are out of scope of ESP-IDF project's issue tracker.

keelung-yang, 2020-09-18:
@igrr I've moved the 2nd question to ESP32 forum. Thanks!