espressif/esp-idf #5883: Add CAN standby mode and wake-up support

keelung-yang, 2020-09-18:
ESP32 need a CAN transceiver to commicate with other units on bus.
And most of transcervers have a standby pin, like NXP's TJA1XXX, TI's TCAN1XXX and so on.
So why not add  standby pin to config,  standby state to status and wake-up from standby automatically function and so on?

kdschlosser, 2021-09-28:
I agree with this.  The TWAI interface needs to be designed like the MCP25625 IC where even in a low power state if a CAN frame is received it has the ability to wake the MCU from sleeping. TWAI = Two Wire "Automotive" Interface. It is a given that to be used in an automotive application the MCU needs to be able to go into a non parasitic power state when the bus has no traffic and be able to wake from that sleep once traffic is seen. without the ability to have the MCU sleep because it cannot be woken removes the ability for it to be used in a vehicle other then a diagnostic scan tool. Kinda limiting.

kdschlosser, 2021-09-28:
That would work but.. in order to make it work a pull-up would need to be used in order to supply voltage to the  transceivers standby pin. A pin on the ESP32 would need to be used to wake the transceiver by turning the pin on and off as needed. When no frames have been received for a specified amount of time the interface in the ESP32 deactivated  then the pin connected to the transceivers standby would need to be turned on and the pin connected to the transceivers receive pin would need to be used to wake the ESP32 on that pin falling. When the ESP32 enters sleep the pin connected to the standby would float and the pull-up in the transceiver would keep the pin pulled high. Once the ESP32 wakes the TWAI interface needs to be activated and then the pin to the transceiver standby would need to be enabled and by default it will take the transceiver out of standby mode.

It has a reference pin which is good.

Only bad thing I can see is there is no through hole version, that means no breadboard prototyping. and for those DIY'ers not savvy enough to solder SMDs  we are still SOL.

iosix, 2021-09-28:
It is extremely standard in automotive circles to use a sleep pin with the can transceiver. You are really overcomplicating this. No one uses through hole ICs anymore, certainly not for automotive. Just roll a quick pcb and it’ll be on your doorstep in a week for ~$20 (sparkfun is even cheaper but I use JLCpcb). 

Robert Vogt IV
CEO
IOSiX, LLC
1300 Tefft Ct #1
Saline, MI  48176
P: +1-855-OBD-1939
C: +1-734-730-9690
***@***.***

kdschlosser, 2021-09-28:
I like to breadboard this way I am able to test and make sure I have everything right. The MCP2562 has a through hole version and it also supports the wakeup feature.