espressif/esp-idf #5884: ESP32 4.0 BLE GATT SERVER 一个service uuid，多个写特征符及通知特征符，怎么区分是哪个特征符来的数据及ESP32怎么主动推送通知到客户端

iot1688, 2020-09-18:
你好
ESP32，做BLE GATT SERVER用。
现在建了一个服务，然后建了2个写的特征符，2个通知特征符，2个读特征符。
APP客户端以上特征符都有绑定操作，现在的话有2个问题疑惑：
1，ESP32端收到APP客户端写数据的事件触发，这里怎么区分中是那个特征符的写事件触发（读类同）
2，ESP32 串口对接的单片机发来数据，想通过通知主推给APP客户端，怎么选择哪个通知特征符主推？函数又是哪个呢？
感谢！

iot1688, 2020-09-24:
乐鑫的朋友，能否帮忙解答下，感谢！

xiewenxiang, 2020-09-24:
@iot1688 

1. 读写操作，在event 参数区，有相应的 handle，用于区分针对哪个 characteristic 操作
2. 选择用哪个特征值来通知，取决于你自己，前提是这个属性值可以 notify 或者 indicate，函数是

```c
esp_err_t esp_ble_gatts_send_indicate(esp_gatt_if_t gatts_if, uint16_t conn_id, uint16_t attr_handle,
 uint16_t value_len, uint8_t *value, bool need_confirm);
```

iot1688, 2020-09-25:
@xiewenxiang 
你好，
1，每次event触发时，是有handle，但好像只知道每个charactersitic的 UUID，莫非要没创建成功时再建一个handle与char uuid的映射表？,
2，esp_ble_gatts_send_indicate这个函数知道，也就是跟（1）一样，事件触发时要记录一个映射表，然后ESP32主动发消息时选择其一char uuid通道?

iot1688, 2020-09-26:
@xiewenxiang 
了解，谢谢

closed 2020-09-26 by iot1688