espressif/esp-idf #5885: can't compile SoftAP example code

kalata23, 2020-09-18:
I am trying to build softAP example with ESP-IDF v4.3-dev-903-g754782039-dirty, cloned from master branch of this repo. But no mather if I set-target to esp32 or esp32s2 i got this error during building:

`[107/107] Generating binary image from built executable
esptool.py v3.0-dev
Generated /home/kalata23/github/esp-idf/examples/wifi/getting_started/softAP/build/bootloader/bootloader.bin
[949/950] Linking CXX executable wifi_softAP.elf
FAILED: wifi_softAP.elf 
: && /home/kalata23/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++  -mlongcalls   CMakeFiles/wifi_softAP.elf.dir/project_elf_src.c.obj  -o wifi_softAP.elf  esp-idf/xtensa/libxtensa.a esp-idf/mbedtls/libmbedtls.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/esp_ipc/libesp_ipc.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/pthread/libpthread.a esp-idf/esp_system/libesp_system.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/vfs/libvfs.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_event/libesp_event.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/lwip/liblwip.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32s2/libesp32s2.a esp-idf/esp_common/libesp_common.a esp-idf/esp_timer/libesp_timer.a esp-idf/freertos/libfreertos.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/cbor/libcbor.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/esp_hid/libesp_hid.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/sdmmc/libsdmmc.a esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/fatfs/libfatfs.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/perfmon/libperfmon.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/main/libmain.a -Wl,--cref -Wl,--Map=/home/kalata23/github/esp-idf/examples/wifi/getting_started/softAP/build/wifi_softAP.map -fno-rtti -fno-lto esp-idf/xtensa/libxtensa.a esp-idf/mbedtls/libmbedtls.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/esp_ipc/libesp_ipc.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/pthread/libpthread.a esp-idf/esp_system/libesp_system.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/vfs/libvfs.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_event/libesp_event.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/lwip/liblwip.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32s2/libesp32s2.a esp-idf/esp_common/libesp_common.a esp-idf/esp_timer/libesp_timer.a esp-idf/freertos/libfreertos.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/cbor/libcbor.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/esp_hid/libesp_hid.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/sdmmc/libsdmmc.a esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/fatfs/libfatfs.a esp-idf/sdmmc/libsdmmc.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/perfmon/libperfmon.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/protocomm/libprotocomm.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/mdns/libmdns.a esp-idf/console/libconsole.a esp-idf/json/libjson.a esp-idf/cxx/libcxx.a esp-idf/newlib/libnewlib.a esp-idf/freertos/libfreertos.a esp-idf/heap/libheap.a esp-idf/log/liblog.a esp-idf/lwip/liblwip.a esp-idf/soc/libsoc.a esp-idf/esp_rom/libesp_rom.a esp-idf/esp_common/libesp_common.a esp-idf/esp_system/libesp_system.a esp-idf/esp32s2/libesp32s2.a esp-idf/xtensa/libxtensa.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/spi_flash/libspi_flash.a esp-idf/mbedtls/libmbedtls.a esp-idf/efuse/libefuse.a esp-idf/app_update/libapp_update.a esp-idf/esp_ipc/libesp_ipc.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/pthread/libpthread.a esp-idf/app_trace/libapp_trace.a esp-idf/soc/soc/esp32s2/libsoc_esp32s2.a esp-idf/esp_event/libesp_event.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_timer/libesp_timer.a esp-idf/wpa_supplicant/libwpa_supplicant.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/libcoexist.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/libcore.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/libespnow.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/libmesh.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/libnet80211.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/libpp.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/librtc.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/libsmartconfig.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/libphy.a esp-idf/vfs/libvfs.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/espcoredump/libespcoredump.a esp-idf/ulp/libulp.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/cxx/libcxx.a esp-idf/newlib/libnewlib.a esp-idf/freertos/libfreertos.a esp-idf/heap/libheap.a esp-idf/log/liblog.a esp-idf/lwip/liblwip.a esp-idf/soc/libsoc.a esp-idf/esp_rom/libesp_rom.a esp-idf/esp_common/libesp_common.a esp-idf/esp_system/libesp_system.a esp-idf/esp32s2/libesp32s2.a esp-idf/xtensa/libxtensa.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/spi_flash/libspi_flash.a esp-idf/mbedtls/libmbedtls.a esp-idf/efuse/libefuse.a esp-idf/app_update/libapp_update.a esp-idf/esp_ipc/libesp_ipc.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/pthread/libpthread.a esp-idf/app_trace/libapp_trace.a esp-idf/soc/soc/esp32s2/libsoc_esp32s2.a esp-idf/esp_event/libesp_event.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_timer/libesp_timer.a esp-idf/wpa_supplicant/libwpa_supplicant.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/libcoexist.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/libcore.a /home/kalata23/github/esp-idf/components/esp_wifi/lib/esp32s2/libespnow.a /home/kalata [... cut ...] _cb'
/home/kalata23/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(wifi_default.c.obj): in function `wifi_start':
/home/kalata23/github/esp-idf/components/esp_wifi/src/wifi_default.c:56: undefined reference to `esp_wifi_internal_reg_netstack_buf_cb'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
`

felmue, 2020-09-18:
Hello @kalata23 

I've just tried to build softAP on my Mac and it compiled without issues so I am not sure why your build failed.

I presume you followed these steps for your platform?

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/

Thanks
Felix

kalata23, 2020-09-18:
> Hello @kalata23
> 
> I've just tried to build `softAP` on my Mac and it compiled without issues so I am not sure why your build failed.
> 
> I presume you followed these steps for your platform?
> 
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/
> 
> Thanks
> Felix

Yes, I have used esp-idf v4.3 cloned from master branch of this repository.  I entered 
`./install.sh `
then 
`. .export.sh`
 then i went into example dir and entered:
`idf.py set-target esp32s2`
then:
`idf.py flash monitor`
  which also builds the entire project.
I am using Linux Mint 19.3 (x64)

I put the result in my previous comment.

felmue, 2020-09-18:
Hi @kalata23 

Just to make sure, did you also follow these preparation steps to setup the Linux tool chain?

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/linux-setup.html

I've also made a copy of the project instead of building it inside the examples directory as outlined here (not sure if that's relevant though):

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/#step-5-start-a-project

Thanks
Felix

kalata23, 2020-09-18:
Ok, just to be sure,  I did again the instructions from the first link. Then i ran again `install.sh` script. Also again sourced with `. ./export.sh` and again did `idf.py set-target esp32s2` and `idf.py flash monitor`. Thre result is the same. And no, it doesn't matter if I compile the project withing examples directory or make a copy of it in another directory.
Can you tell me the version that you used? Type: `idf.py --version`

felmue, 2020-09-18:
Hi @kalata23 

Hmm, that's odd. Unfortunately I don't have a Linux installation to try.

Anyhow here is the version I am using on my Mac: `ESP-IDF v4.3-dev-1197-g8bc19ba89-dirty`

I am sorry for not being more of a help.

Maybe somebody with a Linux installation has some additional ideas?

Felix

mahavirj, 2020-09-18:
@kalata23 Maybe some issue with recursive clone. Following commands may help you:

```
cd /path/to/idf/
git submodule init .
git submodule update --recursive
```

If it does not help, please post output of command `git submodule`.

Interim, symbol you are referring is part of `libnet80211.a`, can be cross checked using:

```
$ xtensa-esp32-elf-nm components/esp_wifi/lib/esp32/libnet80211.a | grep esp_wifi_internal_reg_netstack_buf_cb
00000008 T esp_wifi_internal_reg_netstack_buf_cb
00000004 T esp_wifi_internal_reg_netstack_buf_cb_local
```
Thanks.

kalata23, 2020-09-21:
I've deleted my local copy and cloned the repo again. I still think it can be related to a specific version of idf.py. Before I delete my local copy, I found that idf's version wasn't the latest version.

closed 2020-09-21 by mahavirj

adrien-n, 2020-09-27:
I had the same issue and it was indeed git submodules not being up-to-date.