espressif/esp-idf #5888: HTTP Digest Auth fails to connect

atlasoft, 2020-09-18:
Hi, 
I'm trying to connect to a lighttpd server in order to make a GET request with this chunk of code:
```c
        esp_http_client_config_t config = {
            .url = "https://192.168.1.50/00AA11442233.json",
            .cert_pem = root_cert_pem_start,
            .auth_type = HTTP_AUTH_TYPE_DIGEST,
            .username = "espUpdateUser",
            .password = "123456",
            .skip_cert_common_name_check = 1,
            .transport_type = HTTP_TRANSPORT_OVER_SSL,
            .method = HTTP_METHOD_GET,
        };
        esp_http_client_handle_t client = esp_http_client_init(&config);
        esp_err_t err = esp_http_client_perform(client);
```
and I'm getting this GuruMeditation:
```
D (23:47:57.567) HTTP_AUTH: Digest espUpdateUser Authorized users only 123456

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400014fd  PS      : 0x00060730  A0      : 0x80146018  A1      : 0x3ffd20a0  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffda758  
A10     : 0x3ffda738  A11     : 0x3ffdb538  A12     : 0x00000020  A13     : 0x00000001  
A14     : 0x3ffda720  A15     : 0x3ffda650  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace:0x400014fa:0x3ffd20a0 0x40146015:0x3ffd20b0 0x4014339d:0x3ffd23c0 0x400f1216:0x3ffd2480 0x400f0007:0x3ffd24e0 0x400f0d9e:0x3ffd2500 0x400d544d:0x3ffd2520
0x40146015: _svfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1528

0x4014339d: asprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/asprintf.c:74

0x400f1216: http_auth_digest at /home/tokyo/esp-idf/v4.2/components/esp_http_client/lib/http_auth.c:132

0x400f0007: esp_http_client_prepare at /home/tokyo/esp-idf/v4.2/components/esp_http_client/esp_http_client.c:503

0x400f0d9e: esp_http_client_perform at /home/tokyo/esp-idf/v4.2/components/esp_http_client/esp_http_client.c:930

0x400d544d: app at /home/tokyo/Documentos/Proyectos/BASE/build/../main/app_main.c:151
```
After some research I figured out that the server does not set the "opaque" header file during the authentication, so the line 
https://github.com/espressif/esp-idf/blob/8bc19ba893e5544d571a753d82b44a84799b94b1/components/esp_http_client/esp_http_client.c#L1359
returns a NULL pointer which cannot be referenced in further steps.

I'm not totally sure if it's mandatory for the server to set the opaque field (I have not foud how to set this in lighttpd nor apache)?
For the moment I modified the line quoted above in order to return an empty null-terminated string and works, but I'm sure there is another way to do. 
Any suggestion?

KR

atlasoft, 2020-09-18:
BTW, I modified 
https://github.com/espressif/esp-idf/blob/8bc19ba893e5544d571a753d82b44a84799b94b1/components/esp_http_client/lib/http_utils.c#L93

to look like 

```
char *http_utils_get_string_between(const char *str, const char *begin, const char *end) {
    char *found = strstr(str, begin);
    char *ret = NULL;
    if (found) {
        found += strlen(begin);
        char *found_end = strstr(found, end);
        if (found_end) {
            ret = calloc(1, found_end - found + 1);
            mem_check(ret);
            memcpy(ret, found, found_end - found);
            return ret;
        }
    } else {
        ret = calloc(1, 1);
        mem_check(ret);
        ret[0] = '\0';
        return ret;
    }
    return NULL;
}
```
But I think that is risky because if another function could expect a null pointer...

shubhamkulkarni97, 2020-09-30:
Hi @atlasoft,

[RFC](https://tools.ietf.org/html/rfc7616) states the following about opaque field:
```
opaque

      A string of data, specified by the server, that SHOULD be returned
      by the client unchanged in the Authorization header field of
      subsequent requests with URIs in the same protection space.  It is
      RECOMMENDED that this string be Base64 or hexadecimal data.

```

This does not mention opaque as optional field and indicates an issue related to server.

Thanks,
Shubham

negativekelvin, 2020-09-30:
@shubhamkulkarni97 anything that is not an absolute requirement should be considered optional and esp_http_client should handle this case

> SHOULD   This word, or the adjective "RECOMMENDED", mean that there
>    may exist valid reasons in particular circumstances to ignore a
>    particular item, but the full implications must be understood and
>    carefully weighed before choosing a different course.

> MUST   This word, or the terms "REQUIRED" or "SHALL", mean that the
>    definition is an absolute requirement of the specification.

https://tools.ietf.org/html/rfc2119

shubhamkulkarni97, 2020-10-05:
@atlasoft @negativekelvin,

Please try applying patch attcahed below and check if it fixes the issue.
[0001-http_auth.c-Fix-crash-when-opaque-field-is-not-prese.patch.zip](https://github.com/espressif/esp-idf/files/5326421/0001-http_auth.c-Fix-crash-when-opaque-field-is-not-prese.patch.zip)

Let me know if it works.

Thanks,
Shubham

KaeLL, 2020-10-05:
> @atlasoft @negativekelvin,
>
> Please try applying patch attcahed below and check if it fixes the issue. [...]

It didn't to me.

<details>
<summary>Click to expand http-client example log of the digest auth req.</summary>

```
I (13627) HTTP_CLIENT: HTTP Basic Auth redirect Status = 200, content_length = 47
I (13627) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
D (13637) HTTP_CLIENT: Begin connect to: http://httpbin.org:80
D (13637) TRANS_TCP: [sock=54],connecting to server IP:34.194.129.11,Port:80...
D (13837) HTTP_CLIENT: HTTP_EVENT_ON_CONNECTED
D (13837) HTTP_CLIENT: Write header[3]: GET /digest-auth/auth/user/passwd/MD5/never HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: httpbin.org
Content-Length: 0

D (13877) HTTP_CLIENT: HTTP_EVENT_HEADER_SENT
D (14137) HTTP_CLIENT: on_message_begin
D (14137) HTTP_CLIENT: HEADER=Date:Mon, 05 Oct 2020 15:50:20 GMT
D (14137) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Date, value=Mon, 05 Oct 2020 15:50:20 GMT
D (14137) HTTP_CLIENT: HEADER=Content-Type:text/html; charset=utf-8
D (14137) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Content-Type, value=text/html; charset=utf-8
D (14137) HTTP_CLIENT: HEADER=Content-Length:0
D (14137) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Content-Length, value=0
D (14137) HTTP_CLIENT: HEADER=Connection:keep-alive
D (14137) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
D (14137) HTTP_CLIENT: HEADER=Server:gunicorn/19.9.0
D (14147) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Server, value=gunicorn/19.9.0
D (14147) HTTP_CLIENT: HEADER=WWW-Authenticate:Digest realm="me@kennethreitz.com", nonce="ce9b90a59f7c509564d4be9e8337e9c5", qop="auth", opaque="ab15c94f91979d8913c4bf4e228c5a54", algorithm=MD5, stale=FALSE
D (14147) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=WWW-Authenticate, value=Digest realm="me@kennethreitz.com", nonce="ce9b90a59f7c509564d4be9e8337e9c5", qop="auth", opaque="ab15c94f91979d8913c4bf4e228c5a54", algorithm=MD5, stale=FALSE
D (14147) HTTP_CLIENT: HEADER=Set-Cookie:stale_after=never; Path=/
D (14147) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Set-Cookie, value=stale_after=never; Path=/
D (14147) HTTP_CLIENT: HEADER=Set-Cookie:fake=fake_value; Path=/
D (14147) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Set-Cookie, value=fake=fake_value; Path=/
D (14147) HTTP_CLIENT: HEADER=Access-Control-Allow-Origin:*
D (14157) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Origin, value=*
D (14157) HTTP_CLIENT: HEADER=Access-Control-Allow-Credentials:true
D (14157) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Credentials, value=true
D (14157) HTTP_CLIENT: http_on_headers_complete, status=401, offset=501, nread=501
D (14157) HTTP_CLIENT: http_on_message_complete, parser=3ffbcd3c
D (14157) HTTP_CLIENT: content_length = 0
D (14157) HTTP_CLIENT: UNAUTHORIZED: Digest realm="me@kennethreitz.com", nonce="ce9b90a59f7c509564d4be9e8337e9c5", qop="auth", opaque="ab15c94f91979d8913c4bf4e228c5a54", algorithm=MD5, stale=FALSE
D (14157) HTTP_CLIENT: type = Digest
D (14157) HTTP_CLIENT: HTTP_EVENT_ON_FINISH
D (14157) HTTP_AUTH: Digest user me@kennethreitz.com passwd

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c250  PS      : 0x00060930  A0      : 0x800576e9  A1      : 0x3ffbc4d0  
A2      : 0x0000002b  A3      : 0x00000000  A4      : 0x0000002a  A5      : 0x00000001  
A6      : 0x3ffbda43  A7      : 0x0000000f  A8      : 0x00000002  A9      : 0x3ffbda44  
A10     : 0x0000003a  A11     : 0x3ffbc562  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x3ffbda42  A15     : 0x00000060  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002b  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 617993a725118531

Backtrace: 0x4000c24d:0x3ffbc4d0 0x400576e6:0x3ffbc4f0 0x4005730f:0x3ffbc520 0x40056ea5:0x3ffbc5b0 0x40056ece:0x3ffbc650 0x400f9b8f:0x3ffbc680 0x400f9d8a:0x3ffbc750 0x400f882e:0x3ffbc7b0 0x400f999e:0x3ffbc7d0 0x400d6de3:0x3ffbc7f0 0x400d734b:0x3ffbc880
0x400f9b8f: md5_printf at /esp-idf/components/esp_http_client/lib/http_auth.c:52

0x400f9d8a: http_auth_digest at /esp-idf/components/esp_http_client/lib/http_auth.c:116

0x400f882e: esp_http_client_prepare at /esp-idf/components/esp_http_client/esp_http_client.c:478

0x400f999e: esp_http_client_perform at /esp-idf/components/esp_http_client/esp_http_client.c:889

0x400d6de3: http_auth_digest at /ESP32-W5100/build/../components/ex_esp_http_client/esp_http_client_example.c:354

0x400d734b: http_test_task at /ESP32-W5100/build/../components/ex_esp_http_client/esp_http_client_example.c:621

Rebooting...
```

</details>

shubhamkulkarni97, 2020-10-09:
@KaeLL which server did you use for the test? From the logs, I can see `opaque` field is present in the response.

KaeLL, 2020-10-09:
Yeah, scratch that test, as I might have tested with v4.1 without the patch, I'm not sure. The server is httpbin.org, as seen [here](https://github.com/KaeLL/ESP32-W5100/blob/master/components/ex_esp_http_client/esp_http_client_example.c#L347-L368)
I tested it again today, but correctly this time, on v4.1-beta1, and I had no issues, both with and without your patch.

gstrauss, 2020-10-15:
@shubhamkulkarni97 you quoted part of RFC 7616 and concluded
> This does not mention opaque as optional field and indicates an issue related to server.

I disagree.

Both RFC 7616 "HTTP Digest Access Authentication" and its predecessor RFC 2617 "HTTP Authentication: Basic and Digest Access Authentication" name the token `digest-challenge`.  Due to an apparent oversight, the ABNF for `digest-challenge` does not appear in RFC 7616, but is present in RFC 2617:

>       challenge        =  "Digest" digest-challenge
> 
>       digest-challenge  = 1#( realm | [ domain ] | nonce |
>                           [ opaque ] |[ stale ] | [ algorithm ] |
>                           [ qop-options ] | [auth-param] )

The items in square brackets are **optional** for the server generating the `digest-challenge`.  **IFF** `opaque` is provided by the server **THEN** the client is required to return it, which is noted elsewhere in both of those RFCs.

KaeLL, 2020-10-16:
@shubhamkulkarni97 

I managed to repro the issue.

Here's the describe and diff

```diff
$ git describe && git diff components/esp_http_client/lib/http_auth.c
v4.1
diff --git a/components/esp_http_client/lib/http_auth.c b/components/esp_http_client/lib/http_auth.c
index 1fcc2f115..648958ef3 100644
--- a/components/esp_http_client/lib/http_auth.c
+++ b/components/esp_http_client/lib/http_auth.c
@@ -71,7 +71,7 @@ char *http_auth_digest(const char *username, const char *password, esp_http_auth
 {
     char *ha1, *ha2 = NULL;
     char *digest = NULL;
-    char *auth_str = NULL;
+    char *auth_str = NULL, *temp_auth_str = NULL;
 
     if (username == NULL ||
         password == NULL ||
@@ -123,8 +123,13 @@ char *http_auth_digest(const char *username, const char *password, esp_http_auth
         }
     }
     asprintf(&auth_str, "Digest username=\"%s\", realm=\"%s\", nonce=\"%s\", uri=\"%s\", algorithm=\"MD5\", "
-             "response=\"%s\", opaque=\"%s\", qop=%s, nc=%08x, cnonce=\"%016llx\"",
-             username, auth_data->realm, auth_data->nonce, auth_data->uri, digest, auth_data->opaque, auth_data->qop, auth_data->nc, auth_data->cnonce);
+             "response=\"%s\", qop=%s, nc=%08x, cnonce=\"%016llx\"",
+             username, auth_data->realm, auth_data->nonce, auth_data->uri, digest, auth_data->qop, auth_data->nc, auth_data->cnonce);
+    if (auth_data->opaque) {
+        asprintf(&temp_auth_str, "%s, opaque=\"%s\"", auth_str, auth_data->opaque);
+        free(auth_str);
+        auth_str = temp_auth_str;
+    }
 _digest_exit:
     free(ha1);
     free(ha2);
```

and here's the log
<details>
<summary>Click to expand</summary>

```
I (12149) HTTP_CLIENT: HTTP Basic Auth redirect Status = 200, content_length = 47
I (12151) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
D (12151) w5100_eth_mac: buf = 0x3ffc86de       length = 54
D (12151) emac_w5100_transmit: 0x3ffc86de   d4 ca 6d e2 f5 a1 80 7d  3a f5 02 4b 08 00 45 00  |..m....}:..K..E.|
D (12153) emac_w5100_transmit: 0x3ffc86ee   00 28 00 3c 00 00 ff 06  39 5e c0 a8 fb 50 23 ac  |.(.<....9^...P#.|
D (12153) emac_w5100_transmit: 0x3ffc86fe   a2 90 d1 e1 00 50 00 00  1a a0 4d 96 09 a2 50 11  |.....P....M...P.|
D (12155) emac_w5100_transmit: 0x3ffc870e   14 62 d5 31 00 00                                 |.b.1..|
D (12167) HTTP_CLIENT: Begin connect to: http://httpbin.org:80
D (12167) TRANS_TCP: [sock=54],connecting to server IP:35.172.162.144,Port:80...
D (12167) w5100_eth_mac: buf = 0x3ffc881a       length = 58
D (12167) emac_w5100_transmit: 0x3ffc881a   d4 ca 6d e2 f5 a1 80 7d  3a f5 02 4b 08 00 45 00  |..m....}:..K..E.|
D (12169) emac_w5100_transmit: 0x3ffc882a   00 2c 00 3d 00 00 ff 06  39 59 c0 a8 fb 50 23 ac  |.,.=....9Y...P#.|
D (12169) emac_w5100_transmit: 0x3ffc883a   a2 90 d1 e2 00 50 00 00  19 a7 00 00 00 00 60 02  |.....P........`.|
D (12171) emac_w5100_transmit: 0x3ffc884a   16 70 13 bb 00 00 02 04  05 a0                    |.p........|
V (12217) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
D (12259) w5100_eth_mac: buf = 0x3ffbe6bc       length = 60
D (12259) emac_w5100_receive: 0x3ffbe6bc   ff ff ff ff ff ff 58 10  8c 3e 22 08 08 06 00 01  |......X..>".....|
D (12261) emac_w5100_receive: 0x3ffbe6cc   08 00 06 04 00 01 58 10  8c 3e 22 08 c0 a8 19 54  |......X..>"....T|
D (12261) emac_w5100_receive: 0x3ffbe6dc   00 00 00 00 00 00 c0 a8  19 01 00 00 00 00 00 00  |................|
D (12263) emac_w5100_receive: 0x3ffbe6ec   00 00 00 00 00 00 00 00  00 00 00 00              |............|
D (12311) w5100_eth_mac: buf = 0x3ffbe6bc       length = 60
D (12311) emac_w5100_receive: 0x3ffbe6bc   80 7d 3a f5 02 4b d4 ca  6d e2 f5 a1 08 00 45 00  |.}:..K..m.....E.|
D (12313) emac_w5100_receive: 0x3ffbe6cc   00 28 aa d8 40 00 f0 06  5d c1 23 ac a2 90 c0 a8  |.(..@...].#.....|
D (12313) emac_w5100_receive: 0x3ffbe6dc   fb 50 00 50 d1 e1 4d 96  09 a2 00 00 1a a1 50 11  |.P.P..M.......P.|
D (12315) emac_w5100_receive: 0x3ffbe6ec   6c e0 7c b2 00 00 00 00  00 00 00 00              |l.|.........|
[... 113 lines cut ...]
A2      : 0x0000002b  A3      : 0x00000000  A4      : 0x0000002a  A5      : 0x00000001  
A6      : 0x3ffc893b  A7      : 0x0000000f  A8      : 0x00000002  A9      : 0x3ffc893c  
A10     : 0x0000003a  A11     : 0x3ffbd962  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x3ffc893a  A15     : 0x00000060  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002b  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 06bd90668003b862

Backtrace: 0x4000c24d:0x3ffbd8d0 0x400576e6:0x3ffbd8f0 0x4005730f:0x3ffbd920 0x40056ea5:0x3ffbd9b0 0x40056ece:0x3ffbda50 0x40100cc3:0x3ffbda80 0x40100ec1:0x3ffbdb50 0x400ff962:0x3ffbdbb0 0x40100ad2:0x3ffbdbd0 0x400d6ffb:0x3ffbdbf0 0x400d7563:0x3ffbdc80
0x40100cc3: md5_printf at /esp-idf/components/esp_http_client/lib/http_auth.c:52

0x40100ec1: http_auth_digest at /esp-idf/components/esp_http_client/lib/http_auth.c:116

0x400ff962: esp_http_client_prepare at /esp-idf/components/esp_http_client/esp_http_client.c:478

0x40100ad2: esp_http_client_perform at /esp-idf/components/esp_http_client/esp_http_client.c:889

0x400d6ffb: http_auth_digest at /ESP32-W5100/build/../components/ex_esp_http_client/esp_http_client_example.c:354

0x400d7563: http_test_task at /ESP32-W5100/build/../components/ex_esp_http_client/esp_http_client_example.c:621

Rebooting...
```

<details>

shubhamkulkarni97, 2020-10-27:
@gstrauss I agree with your opinion and have added a fix to work with lighttpd server. Please check patch attached in [above comment](https://github.com/espressif/esp-idf/issues/5888#issuecomment-703511167).

@KaeLL Thanks for reporting the issue, can you please share the server you used for the test and the steps to reproduce the issue? It will help us to add a fix.

KaeLL, 2020-10-27:
@shubhamkulkarni97 It's the esp_http_client example. Just yesterday I managed to repro the issue again using the snapshot on the master branch. For the sake of accuracy, [here](https://github.com/KaeLL/ESP32-W5100/blob/a67e5b13de8f5ef9f298e68ed531924041f1f7ec/components/ex_esp_http_client/esp_http_client_example.c#L347-L368)'s the commit used to produce the log I provided, which was tested against IDF v4.1, and [here](https://github.com/KaeLL/ESP32-W5100/blob/c76cc328dea31ae08ccb1b5c14e94337e465b9bf/components/ex_esp_http_client/esp_http_client_example.c#L409-L430)'s the version tested against the latest master snapshot of IDF so far yesterday.

closed 2020-12-13 by espressif-bot

KaeLL, 2021-03-25:
Still broken as of 1067b28 running its esp_http_client example
<details>
<summary>Log</summary>

```
I (11349) HTTP_CLIENT: HTTP Basic Auth redirect Status = 200, content_length = 47
I (11349) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
D (11363) HTTP_CLIENT: Begin connect to: http://httpbin.org:80
D (11363) esp-tls: host:httpbin.org: strlen 11
D (11363) esp-tls: [sock=54] Resolved IPv4 address: 54.166.163.67
D (11363) esp-tls: [sock=54] Connecting to server. HOST: httpbin.org, Port: 80
D (11551) esp-tls: non-tls connection established
D (11551) HTTP_CLIENT: HTTP_EVENT_ON_CONNECTED
D (11551) HTTP_CLIENT: Write header[3]: GET /digest-auth/auth/user/passwd/MD5/never HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: httpbin.org
Content-Length: 0

D (11587) HTTP_CLIENT: HTTP_EVENT_HEADER_SENT
D (11839) HTTP_CLIENT: on_message_begin
D (11839) HTTP_CLIENT: HEADER=Date:Thu, 25 Mar 2021 01:54:10 GMT
D (11839) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Date, value=Thu, 25 Mar 2021 01:54:10 GMT
D (11839) HTTP_CLIENT: HEADER=Content-Type:text/html; charset=utf-8
D (11839) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Content-Type, value=text/html; charset=utf-8
D (11841) HTTP_CLIENT: HEADER=Content-Length:0
D (11841) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Content-Length, value=0
D (11843) HTTP_CLIENT: HEADER=Connection:keep-alive
D (11843) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
D (11843) HTTP_CLIENT: HEADER=Server:gunicorn/19.9.0
D (11845) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Server, value=gunicorn/19.9.0
D (11845) HTTP_CLIENT: HEADER=WWW-Authenticate:Digest realm="me@kennethreitz.com", nonce="1c759827399852d85cb5113ca4d6e398", qop="auth", opaque="8170d6d842597240b8f24a430b1e56d2", algorithm=MD5, stale=FALSE
D (11847) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=WWW-Authenticate, value=Digest realm="me@kennethreitz.com", nonce="1c759827399852d85cb5113ca4d6e398", qop="auth", opaque="8170d6d842597240b8f24a430b1e56d2", algorithm=MD5, stale=FALSE
D (11851) HTTP_CLIENT: HEADER=Set-Cookie:stale_after=never; Path=/
D (11851) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Set-Cookie, value=stale_after=never; Path=/
D (11853) HTTP_CLIENT: HEADER=Set-Cookie:fake=fake_value; Path=/
D (11853) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Set-Cookie, value=fake=fake_value; Path=/
D (11853) HTTP_CLIENT: HEADER=Access-Control-Allow-Origin:*
D (11855) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Origin, value=*
D (11855) HTTP_CLIENT: HEADER=Access-Control-Allow-Credentials:true
D (11857) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Credentials, value=true
D (11857) HTTP_CLIENT: http_on_headers_complete, status=401, offset=501, nread=501
D (11859) HTTP_CLIENT: http_on_message_complete, parser=3ffbffc4
D (11859) HTTP_CLIENT: content_length = 0
D (11859) HTTP_CLIENT: UNAUTHORIZED: Digest realm="me@kennethreitz.com", nonce="1c759827399852d85cb5113ca4d6e398", qop="auth", opaque="8170d6d842597240b8f24a430b1e56d2", algorithm=MD5, stale=FALSE
D (11861) HTTP_CLIENT: type = Digest
D (11863) HTTP_CLIENT: HTTP_EVENT_ON_FINISH
D (11863) HTTP_AUTH: Digest user me@kennethreitz.com passwd

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000c250  PS      : 0x00060d30  A0      : 0x800576e9  A1      : 0x3ffbf8e0  
A2      : 0x0000002b  A3      : 0x00000000  A4      : 0x0000002a  A5      : 0x00000001  
A6      : 0x3ffc0ceb  A7      : 0x0000000f  A8      : 0x00000002  A9      : 0x3ffc0cec  
A10     : 0x0000003a  A11     : 0x3ffbf972  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x3ffc0cea  A15     : 0x00000060  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002b  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x4000c24d:0x3ffbf8e0 0x400576e6:0x3ffbf900 0x4005730f:0x3ffbf930 0x40056ea5:0x3ffbf9c0 0x40056ece:0x3ffbfa60 0x400ff8e6:0x3ffbfa90 0x400ffadd:0x3ffbfb60 0x400fe3e6:0x3ffbfbc0 0x400ff6c6:0x3ffbfbe0 0x400e3230:0x3ffbfc00 0x400e3a7f:0x3ffbfca0
0x400ff8e6: md5_printf at /esp-idf/components/esp_http_client/lib/http_auth.c:52

0x400ffadd: http_auth_digest at /esp-idf/components/esp_http_client/lib/http_auth.c:117

0x400fe3e6: esp_http_client_prepare at /esp-idf/components/esp_http_client/esp_http_client.c:516

0x400ff6c6: esp_http_client_perform at /esp-idf/components/esp_http_client/esp_http_client.c:997

0x400e3230: http_auth_digest at /ESP32-W5100/components/ex_esp_http_client/esp_http_client_example.c:415

0x400e3a7f: http_test_task at /ESP32-W5100/components/ex_esp_http_client/esp_http_client_example.c:821

ELF file SHA256: eef2ab7ca00ff363

Rebooting...
```

</details

KaeLL, 2021-05-29:
@shubhamkulkarni97 the problem is related to enabling `NEWLIB_NANO_FORMAT`. Doing so causes `http_auth_digest`  to break like I reported in the comment above.

AxelLin, 2021-05-31:
> @shubhamkulkarni97 the problem is related to enabling `NEWLIB_NANO_FORMAT`.

I think you should open a new issue.

zzzh, 2023-10-25:
Why this is closed, seems not fixed yet?

zzzh, 2023-10-25:
@KaeLL , did you open a new issue?

zzzh, 2023-10-25:
> > @shubhamkulkarni97 the problem is related to enabling `NEWLIB_NANO_FORMAT`.
> 
> I think you should open a new issue.

What made you think that?

KaeLL, 2023-10-25:
@zzzh it's working on 5.1.1
[image]

zzzh, 2023-10-26:
> @zzzh it's working on 5.1.1 [image]

Cool, any ideas to make it work on 4.4.2, which I'm using and can't move for some reasons.

zzzh, 2023-10-27:
@shubhamkulkarni97