espressif/esp-idf #5889: ADC Calibration feature in ESP32s2 for v4.1

william-ferguson-au, 2020-09-19:
ADC driver for ESP32s2 is not supported on the 4.1 branch but has been applied to master.
This is a request for that to be ported back to the 4.1 branch.

Original ticket for master https://github.com/espressif/esp-idf/issues/5455

william-ferguson-au, 2020-09-19:
This feature was committed to master as rev#6a0951ecb2c30d461a54297ab2475f399886b950
Would be great to see it in 4.2 or 4.1

Alvin1Zhang, 2021-01-11:
@william-ferguson-au Please help check https://github.com/espressif/esp-idf/commit/1b0691f29aab52065a4ed2c5fd2e7ed52c03fb82 on release/4.2, and also help note IDF v4.1 supports ESP32-S2 beta https://github.com/espressif/esp-idf/releases/tag/v4.1. Feel free to reopen, thanks.

closed 2021-01-11 by Alvin1Zhang