espressif/esp-idf #5891: IDF Onboarding - Verify ESP-IDF Tools reports "Python requirements are not satisfied"

stefa69, 2020-09-20:
I use ESP-IDF with VSCode on Windows for some time. When I installed latest esp-idf, in VSCode (Espressif IDF extension) there is a strange issue. In "IDF Onboarding", step "Verify ESP-IDF Tools", all tools are found, seems to be ok but there is message: 

"Command failed: "C:\Users\korisnik\.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe" "C:\Users\korisnik\Desktop\esp-idf\tools\check_python_dependencies.py" -r "c:\Users\korisnik\.vscode\extensions\espressif.esp-idf-extension-0.4.0\esp_debug_adapter\requirements.txt"" 

and following report in OUTPUT window:

Checking ESP-IDF Python requirements using C:\Users\korisnik\Python\Python38\python.exe
Python requirements from C:\Users\korisnik\Desktop\esp-idf\requirements.txt are satisfied.

The following Python requirements are not satisfied:
backports.functools-lru-cache>=1.5
gcovr
graphviz
psutil>=5.5.1
pygdbmi>=0.9.0.0
pylint>=1.9.4
requests>=2.21.0
typing>=3.6.6
pywin32; platform_system == "Windows"
xmlrunner>=1.7.7
To install the missing packages, please run "C:\Users\korisnik\Desktop\esp-idf\install.bat"
Command failed: "C:\Users\korisnik\.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe" "C:\Users\korisnik\Desktop\esp-idf\tools\check_python_dependencies.py" -r "c:\Users\korisnik\.vscode\extensions\espressif.esp-idf-extension-0.4.0\esp_debug_adapter\requirements.txt"

Tried to run install.bat, export.bat, python -m pip install -r %IDF_PATH%/requirements.txt, etc. but nothing seems to work.

In ESP-IDF Command Prompt everything works fine. I can compile, flash...etc. 

Is this BUG or I am doing something wrong?

Thank you!

brianignacio5, 2020-09-21:
Can you try using `python -m pip install -r c:\Users\korisnik.vscode\extensions\espressif.esp-idf-extension-0.4.0\esp_debug_adapter\requirements.txt` since the extension also has some python packages requirements.

stefa69, 2020-09-21:
It worked! Thank you!!!!

Alvin1Zhang, 2020-09-22:
Glad to hear it worked, feel free to reopen, thanks.

closed 2020-09-22 by Alvin1Zhang

avipreshel, 2020-10-08:
> Can you try using `python -m pip install -r c:\Users\korisnik.vscode\extensions\espressif.esp-idf-extension-0.4.0\esp_debug_adapter\requirements.txt` since the extension also has some python packages requirements.
> 
> is %IDF_PATH% the same as the ESP-IDF Path in your extension settings ?

I had similar issue and I ran similar command (given my user folder path):
C:\Users\myfolder>python -m pip install -r C:\Users\myfolder\esp-idf\requirements.txt

And it returned following error

> Processing c:\users\myfolder\esp-idf\tools\kconfig_new\esp-windows-curses
> Requirement already satisfied: setuptools>=21 in c:\users\myfolder\appdata\local\programs\python\python39\lib\site-packages (from -r C:\Users\myfolder\esp-idf\requirements.txt (line 4)) (49.2.1)
> Collecting click>=5.0
>   Using cached click-7.1.2-py2.py3-none-any.whl (82 kB)
> Collecting pyserial>=3.0
>   Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
> Collecting future>=0.15.2
>   Using cached future-0.18.2.tar.gz (829 kB)
> Collecting cryptography>=2.1.4
>   Using cached cryptography-3.1.1-cp36-abi3-win_amd64.whl (1.5 MB)
> Collecting pyparsing<2.4.0,>=2.0.3
>   Using cached pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
> Collecting pyelftools>=0.22
>   Using cached pyelftools-0.26-py2.py3-none-any.whl (136 kB)
> **ERROR: Could not find a version that satisfies the requirement windows-curses (from esp-windows-curses==0.1->-r C:\Users\myfolder\esp-idf\requirements.txt (line 19)) (from versions: none)
> ERROR: No matching distribution found for windows-curses (from esp-windows-curses==0.1->-r C:\Users\myfolder\esp-idf\requirements.txt (line 19))**

The requirements.txt file contains:

> click>=5.0
> pyserial>=3.0
> future>=0.15.2
> cryptography>=2.1.4
> pyparsing>=2.0.3,<2.4.0
> pyelftools>=0.22

Thanks