espressif/esp-idf #5892: Tried SoftAP example on ESP32S2, but ESP reboots after connection.

kalata23, 2020-09-21:
I am using latest esp-idf version. The problem is that, when I build and upload  SoftAP example, whenever I connect to the board, the ESP32S2 reboots. 
Here is the log:

PS: I intentionaly didn't change ssid or password. 
```
kalata23@kalata23-ESPRIMO-P556:~/github/esp-idf/examples/wifi/getting_started/softAP$ idf.py  monitor
Executing action: monitor
Choosing default port b'/dev/ttyUSB0' (use '-p PORT' option to set a specific serial port)
Running idf_monitor in directory /home/kalata23/github/esp-idf/examples/wifi/getting_started/softAP
Executing "/home/kalata23/.espressif/python_env/idf4.3_py3.6_env/bin/python /home/kalata23/github/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- /home/kalata23/github/esp-idf/examples/wifi/getting_started/softAP/build/wifi_softAP.elf -m '/home/kalata23/.espressif/python_env/idf4.3_py3.6_env/bin/python' '/home/kalata23/github/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x178c
load:0x4004c000,len:0x944
load:0x40050000,len:0x2cec
entry 0x4004c1bc
I (46) boot: ESP-IDF v4.3-dev-1197-g8bc19ba89 2nd stage bootloader
I (46) boot: compile time 09:23:34
I (46) boot: chip revision: 0
I (49) boot.esp32s2: SPI Speed      : 80MHz
I (54) boot.esp32s2: SPI Mode       : DIO
I (59) boot.esp32s2: SPI Flash Size : 2MB
I (64) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 00100000
I (102) boot: End of partition table
I (107) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x160d8 ( 90328) map
I (134) esp_image: segment 1: paddr=0x00026100 vaddr=0x3ffc6080 size=0x0331c ( 13084) load
I (138) esp_image: segment 2: paddr=0x00029424 vaddr=0x40022000 size=0x00404 (  1028) load
0x40022000: _WindowOverflow4 at /home/kalata23/github/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (141) esp_image: segment 3: paddr=0x00029830 vaddr=0x40022404 size=0x067e8 ( 26600) load
0x40022404: _coredump_iram_start at ??:?

I (157) esp_image: segment 4: paddr=0x00030020 vaddr=0x40080020 size=0x6a720 (436000) map
0x40080020: _stext at ??:?

I (250) esp_image: segment 5: paddr=0x0009a748 vaddr=0x40028bec size=0x0d488 ( 54408) load
0x40028bec: _xt_context_save at /home/kalata23/github/esp-idf/components/freertos/xtensa/xtensa_context.S:212

I (275) boot: Loaded app from partition at offset 0x10000
I (276) boot: Disabling RNG early entropy source...
I (287) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (287) cpu_start: cpu freq: 160
I (290) cpu_start: Pro cpu up.
I (348) cpu_start: Pro cpu start user code
I (348) cpu_start: Application information:
I (348) cpu_start: Project name:     wifi_softAP
I (351) cpu_start: App version:      v4.3-dev-1197-g8bc19ba89
I (358) cpu_start: Compile time:     Sep 21 2020 09:23:27
I (364) cpu_start: ELF file SHA256:  3742f134c1f6d510...
I (370) cpu_start: ESP-IDF:          v4.3-dev-1197-g8bc19ba89
I (376) heap_init: Initializing. RAM available for dynamic allocation:
I (383) heap_init: At 3FFCD7C8 len 0002E838 (186 KiB): DRAM
I (390) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (396) spi_flash: detected chip: generic
I (401) spi_flash: flash io: dio
W (404) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (421) cpu_start: Starting scheduler on PRO CPU.
I (432) wifi softAP: ESP_WIFI_MODE_AP
I (452) wifi:wifi driver task: 3ffd6030, prio:23, stack:6656, core=0
[... 191 lines cut ...]
I (474) wifi:config nano formating: disabled
I (484) wifi:Init data frame dynamic rx buffer num: 32
I (484) wifi:Init management frame dynamic rx buffer num: 32
I (494) wifi:Init management short buffer num: 32
I (494) wifi:Init dynamic tx buffer num: 32
I (504) wifi:Init static rx buffer size: 1600
I (504) wifi:Init static rx buffer num: 10
I (504) wifi:Init dynamic rx buffer num: 32
I (514) wifi_init: rx ba win: 6
I (514) wifi_init: tcpip mbox: 32
I (524) wifi_init: udp mbox: 6
I (524) wifi_init: tcp mbox: 6
I (524) wifi_init: tcp tx win: 5744
I (534) wifi_init: tcp rx win: 5744
I (534) wifi_init: tcp mss: 1440
I (544) wifi_init: WiFi IRAM OP enabled
I (544) wifi_init: WiFi RX IRAM OP enabled
I (684) phy: phy_version: 603, 72dfd77, Jul  7 2020, 19:57:05, 0, 2
I (684) wifi:enable tsf
I (684) wifi:mode : softAP (7c:df:a1:00:87:7f)
I (694) wifi:Total power save buffer number: 16
I (694) wifi:Init max length of beacon: 752/752
I (694) wifi:Init max length of beacon: 752/752
I (704) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword channel:1
```

espxiehang, 2020-09-21:
Hi @kalata23 ,Please provide me with the "wifi_softAP.elf" file corresponding to this log.

felmue, 2020-09-21:
Hi @kalata23 

glad you got softAP to compile (the other issue you've reported).

I did a quick test on my S2 board and softAP runs fine for me and a client (iPad) can connect without the board rebooting.

```I (1492) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword channel:1
I (112092) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (112092) wifi:station: 04:52:f3:79:d1:d5 join, AID=1, bgn, 20
I (112102) wifi softAP: station 04:52:f3:79:d1:d5 join, AID=1
I (113622) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (140452) wifi:station: 04:52:f3:79:d1:d5 leave, AID = 1, bss_flags is 134259, bss:0x3ffcf590
I (140452) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (140452) wifi softAP: station 04:52:f3:79:d1:d5 leave, AID=1
```

- have you tried with a different client? (maybe the client is trying something the code isn't expecting?)
- have you tried with a different USB power supply? (maybe the additional traffic when a client connects makes power go down for a bit?)

Thanks
Felix

kalata23, 2020-09-22:
Hi @felmue

First I plugged in a battery to my board, so if there is any voltage drop, the battery should handle it. Unfortunately this didn't fix my issue. I have tried to measure and detect current consumption with oscilloscope in believe that if there are any current consumption pikes, I would see them. Well there were no current pikes. The current consumption was 50-150 mA. That consumption could be easily provided by USB.  I also tried connecting with 2 devices - one of them is my PC with WiFi dongle, and the other was my smartphone - Huawei Mate 20 Pro. I found that I couldn't connect with my phone, but after severarl reboots, my PC connected to ESP32-S2. 
Then  I thought it could be pcb issue. I replaced my board with a new one. Unfortunately this didn't fix my issue either. 
My board uses WROOM module, so I have tried with another board (same scheme, just esp-module difference), that is WROVER based. And the result is that with WROVER the board doesnt reboot after connection and I get  log like the one you posted in your last comment.

espxiehang, 2020-09-22:
1,This problem should have nothing to do with PSRAM.(Because by default, PSRAM is turned off.)
2,According to log analysis, this phenomenon is like calling the esp_restart_noos_dig() function after sta is connected to the softap, which is strange.
3,Have you made any changes to this example?

kalata23, 2020-09-23:
> 1,This problem should have nothing to do with PSRAM.(Because by default, PSRAM is turned off.)
> 2,According to log analysis, this phenomenon is like calling the esp_restart_noos_dig() function after sta is connected to the softap, which is strange.
> 3,Have you made any changes to this example?

No, I haven't. Anyways, it seems to be PCB issue. On ESP32-Saola-1R I don't have such issues.

kalata23, 2020-09-23:
Please, close this issue. It was hardware issue.

Alvin1Zhang, 2020-09-24:
@kalata23 Thanks for the notes.

closed 2020-09-24 by Alvin1Zhang