espressif/esp-idf #5893: Flash Encryption example project build problem

tayph, 2020-09-21:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v4]
- Module or chip used: 32-WROVER-B
- IDF version (run ``git describe --tags`` to find it):
 v4.1
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: [vs code]
- Power Supply:         [USB]

## Problem Description
i just try the example project securtiy/flash encryption and i followed instructions 

Ensure Bootloader log verbosity is Info
Enable the flash encryption mode Development
Change the partition table offset to 0x9000

but i gave  error something like this

raise InputError("CSV Error: First partition offset 0x%x overlaps end of partition table 0x%x"
gen_esp32part.InputError: CSV Error: First partition offset 0x9000 overlaps end of partition table 0xa000

Alvin1Zhang, 2020-09-22:
Thanks for reporting, we have a fix under internal reviewing, will update once the fix is available on GitHub. Thanks.

KonstantinKondrashov, 2020-09-22:
Hi @tayph !
This issue resolved on the master 5e8795eebed3f7a93e5c906f0ffc074a3b158193 but not published yet. This will be backported to v4.1 as well.

You can fix this issue. The partition table.csv file should be without offsets in it:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,        data, nvs,      ,  0x6000,
# Extra partition to demonstrate reading/writing of encrypted flash
storage,    data, 0xff,     ,  0x1000, encrypted
factory,    app,  factory,  , 1M,
```
Thanks.

tayph, 2020-09-22:
thanks @KonstantinKondrashov it's work it.
By the way normal build command (idf.py -p PORT encrypted-flash monitor) did  encrypted flash.
i mean i didn't need use idf.py -p PORT encrypted-flash monitor.

closed 2020-09-22 by tayph