espressif/esp-idf #5894: The Socket's function of send() or recv() takes too  much time sometimes

helittr, 2020-09-21:
# Recently, I am completing a project about Low power consumption. And I've got a sticky problem.

My needs is that when the **ESP32** wakes up from deep sleep mode, send some data to server by WIFI.

But, sometimes, the function of `send()` or `recv()` takes time that is up to about 4 seconds. And sometimes, it  takes less than one seconds. All tests were performed when the WIFI signal was greater than **-60dbm**.

Is there any way to optimize this problem?

Thankyou.

Alvin1Zhang, 2020-09-21:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

david-cermak, 2020-10-23:
Hi @helittr 

Could you please explain a bit your environment? Are you sending the data to a local server, which is *always* available or to an internet, how much data you're sending, get these data fragmented, are they sent in multiple packets?
As you know the TCP is not real time so might be tricky with the predictable throughput based on WiFi signal.
There are ways of course to adjust the TCP parameters, but need to understand what's happening on the network. In general we could:
* Disable naggling  (`TCP_NODELAY`)
* Adjusting buffer sizes
* Adjusting low level TCP params, like time to retransmit, re-trasmit period, etc..

closed 2021-03-12 by helittr