espressif/esp-idf #5896: ESP32-S2 reboot after waking up from light sleep

r-escudero, 2020-09-22:
## Environment
- Development Kit:   ESP32-S2-Saola-1
- Kit version (for WroverKit/PicoKit/DevKitC): v1.2
- Module or chip used:  ESP32-S2-WROVER
- IDF version:  v4.3-dev-907-g6c17e3a64
- Build System:      idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-eld-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:    Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?:    No
- Power Supply:    USB

## Problem Description
Running the light sleep example, the microcontroller is rebooting after waking up.
### Expected Behavior
Continue execution in the line after the "esp_light_sleep_start();"
### Actual Behavior
Rebooting

### Steps to reproduce
1. cd esp-idf\examples\system\light_sleep
2. idf.py set-target esp32s2
3. idf.py build
4. idf.py flash monitor

### Code to reproduce this issue
https://github.com/espressif/esp-idf/blob/8bc19ba893e5544d571a753d82b44a84799b94b1/examples/system/light_sleep/main/light_sleep_example_main.c

## Debug Logs
Entering light sleep
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x17bc
load:0x4004c000,len:0x14c4
load:0x40050000,len:0x2158
SHA-256 comparison failed:
Calculated: 090db62871e9bdaa852f0d7f81ec10e50f9bce38f95f14690a7d2f79ca1f7438
Expected: 838bbbdb43d02a4035ccfc711aa3b667a9202647133edb77ba82ae5212bce1d5
Attempting to boot anyway...
entry 0x4004c2d8
I (65) boot: ESP-IDF v4.3-dev-907-g6c17e3a64-dirty 2nd stage bootloader
I (65) boot: compile time 12:52:12
I (65) boot: chip revision: 0
I (69) boot.esp32s2: SPI Speed      : 80MHz
I (73) boot.esp32s2: SPI Mode       : DIO
I (78) boot.esp32s2: SPI Flash Size : 4MB
I (83) boot: Enabling RNG early entropy source...
I (88) boot: Partition Table:
I (92) boot: ## Label            Usage          Type ST Offset   Length
I (99) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (107) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (114) boot:  2 factory          factory app      00 00 00010000 00100000
I (122) boot: End of partition table
I (126) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x05afc ( 23292) map
I (140) esp_image: segment 1: paddr=0x00015b24 vaddr=0x3ffbc090 size=0x01f94 (  8084) load
I (146) esp_image: segment 2: paddr=0x00017ac0 vaddr=0x40022000 size=0x00404 (  1028) load
0x40022000: _WindowOverflow4 at C:/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (153) esp_image: segment 3: paddr=0x00017ecc vaddr=0x40022404 size=0x0814c ( 33100) load
0x40022404: _coredump_iram_end at ??:?

I (170) esp_image: segment 4: paddr=0x00020020 vaddr=0x40080020 size=0x148b0 ( 84144) map
0x40080020: _stext at ??:?

I (188) esp_image: segment 5: paddr=0x000348d8 vaddr=0x4002a550 size=0x01b3c (  6972) load
0x4002a550: gpspi_flash_ll_get_addr_bitlen at C:/esp/esp-idf/components/soc/src/esp32s2/include/hal/gpspi_flash_ll.h:301
 (inlined by) spi_flash_hal_gpspi_read at C:/esp/esp-idf/components/soc/src/hal/spi_flash_hal_common.inc:116

I (196) boot: Loaded app from partition at offset 0x10000
I (196) boot: Disabling RNG early entropy source...
I (209) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (209) cpu_start: cpu freq: 160
I (209) cpu_start: Pro cpu up.
I (264) cpu_start: Pro cpu start user code
I (265) cpu_start: Application information:
I (265) cpu_start: Project name:     light_sleep_example
I (269) cpu_start: App version:      1
I (273) cpu_start: Compile time:     Sep 22 2020 12:52:17
I (279) cpu_start: ELF file SHA256:  0560d172f133a888...
I (285) cpu_start: ESP-IDF:          v4.3-dev-907-g6c17e3a64-dirty
I (292) heap_init: Initializing. RAM available for dynamic allocation:
I (299) heap_init: At 3FFBE7E0 len 0003D820 (246 KiB): DRAM
I (306) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (312) spi_flash: detected chip: generic
I (317) spi_flash: flash io: dio
I (321) cpu_start: Starting scheduler on PRO CPU.
I (326) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Entering light sleep

## Other items if possible

Alvin1Zhang, 2020-09-22:
Thanks for reporting, we will look into.

felmue, 2020-09-22:
Hi guys

I ran a quick test on the esp32-s2-kaluga-1 board and it doesn't crash for me. However light sleep immediately returns without waiting for 2 seconds or for `GPIO0` to go low with reason `other`.
```
Entering light sleep
Returned from light sleep, reason: other, t=955 ms, slept for 1 ms
Entering light sleep
Returned from light sleep, reason: other, t=965 ms, slept for 0 ms
```
After modifying the code to use `GPIO19` instead of `GPIO0` (and pulling it up) the light sleep sample code started to work as expected for me and wakes up every 2 seconds w/o crash.
```
#define BUTTON_GPIO_NUM_DEFAULT     19
```
```
Entering light sleep
Returned from light sleep, reason: timer, t=2931 ms, slept for 2000 ms
Entering light sleep
Returned from light sleep, reason: timer, t=4940 ms, slept for 2000 ms
```
@r-escudero : I've noticed you're not using the latest code from github which is `ESP-IDF v4.3-dev-1197-g8bc19ba89-dirty`, maybe that is part of the issue you're seeing as well?

Thanks
Felix

igrr, 2020-09-22:
Linking https://github.com/espressif/esp-idf/issues/5692 which looks related, and was resolved by b5e9047. @r-escudero, could you please try that?

@felmue when facing the issue with GPIO0 on Kaluga board, could you confirm with a multimeter or an oscilloscope that GPIO0 is not actually low when the wakeup happens?

r-escudero, 2020-09-23:
Solved.
I didn't realize of not having installed the last version. After updating it is running smooth
@felmue thanks for the heads up.

closed 2020-09-23 by r-escudero

felmue, 2020-09-23:
Hi guys

@igrr : I've double checked with a multimeter and GPIO0 is about 2.1 volts on its own (unconnected). But the issue happens even when I externally pull it up (using a 10k resistor) to 3.3 volts.
Also notice that the wake-up reason isn't `pin` but `other` when using GPIO0.
```
I (325) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Entering light sleep
Returned from light sleep, reason: other, t=921 ms, slept for 1 ms
Entering light sleep
Returned from light sleep, reason: other, t=931 ms, slept for 0 ms
```
@r-escudero : You are very welcome. I am glad to hear you've sorted it out.

Thanks
Felix