espressif/esp-idf #5899: I2C scanner example is not detecting device

shivam5594, 2020-09-23:
### Environment
Development Kit: ESP32-cam (ESP32-Wrover-Kit)
Module or chip used: ESP32-D0WDQ6 (revision 1)
IDF version v4.3-dev-1197-g8bc19ba89
Build System: CMake
Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
Operating System: macOS
Power Supply: external 5V

### Problem Description
I am trying to run i2c-tool app as it is after changing SCL and SDA pin number as my connection. i2cdetect is not detecting my MPU9250. Its device ID is 0x68 and I ran arduino example code for same pins and it is working fine. I have most updated code from esp-idf and this same hardware config is working fine in Arduino IDE.

### Expected Behavior
i2c-tools> i2cdetect
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
i2c-tools>

### Actual Behavior
i2c-tools> i2cdetect
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
i2c-tools>

### Steps to repropduce
Download the i2c-tool code
Run i2cdetect (SCL 15 SDA 14)

MPU9250       ESP32-CAM
VCC <==> 3.3v
GND <==> GND
SCL  <==> 15
SDA  <==> 14

### Code to reproduce this issue
```c++
static const char *TAG = "cmd_i2ctools";

static gpio_num_t i2c_gpio_sda = 14;
static gpio_num_t i2c_gpio_scl = 15;
static uint32_t i2c_frequency = 100000;
static i2c_port_t i2c_port = I2C_NUM_0;
```

felmue, 2020-09-23:
Hi @shivam5594 

I am running `i2c-tools` on an ESP32 LyraT board (ESP32 WROVER kit) and `i2cdetect` works with both sets of pins just fine. I've changed the pins using `i2cconfig` (w/o recompiling).
```
i2c-tools> i2cconfig --sda 18 --scl 19
i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- 3e -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
i2c-tools> i2cconfig --sda 14 --scl 15
i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- 3e -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
i2c-tools>
```
Are you sure that on your board pins 14 and 15 aren't used for anything else already? If I found the correct ESP32-Cam schematics, pins 14 and 15 are used for the SD card. Any chance you have an SD card inserted? Maybe you've swapped SDA and SCL by accident. Maybe try with a different i2c slave?

Thanks
Felix

P.S. I've also tried your method and modified pins in code, rebuilt and reflashed the board. The i2c slave was still detected.

shivam5594, 2020-09-24:
Hi @felmue 

Thank you for pointing out. I checked all my connections and Arduino Twowire(I2C) code works on same pin even though those pins are connected with SD card slot. 

I realized one thing, even though I set the pins for I2C in code initially,
```
static gpio_num_t i2c_gpio_sda = 14;
static gpio_num_t i2c_gpio_scl = 15;
static uint32_t i2c_frequency = 100000;
```
 it **doesn't config** pins for I2C.

After explicitly running i2cconfig --sda=14 --scl=15 it detects,

i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
**i2c-tools> i2cconfig --scl=2 --sda=14**
**i2c-tools> i2cdetect**
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

isn't that weird? I feel that is a confusing code structure.

shivam5594, 2020-09-24:
I checked the code, all i2cconfig command does is parse the parameters and save it in a structure without actually configuring those pins. 

`static struct {
    struct arg_int *port;
    struct arg_int *freq;
    struct arg_int *sda;
    struct arg_int *scl;
    struct arg_end *end;
} i2cconfig_args;

static int do_i2cconfig_cmd(int argc, char **argv)
{
    int nerrors = arg_parse(argc, argv, (void **)&i2cconfig_args);
    if (nerrors != 0) {
        arg_print_errors(stderr, i2cconfig_args.end, argv[0]);
        return 0;
    }

    /* Check "--port" option */
    if (i2cconfig_args.port->count) {
        if (i2c_get_port(i2cconfig_args.port->ival[0], &i2c_port) != ESP_OK) {
            return 1;
        }
    }
    /* Check "--freq" option */
    if (i2cconfig_args.freq->count) {
        i2c_frequency = i2cconfig_args.freq->ival[0];
    }
    /* Check "--sda" option */
    i2c_gpio_sda = i2cconfig_args.sda->ival[0];
    /* Check "--scl" option */
    i2c_gpio_scl = i2cconfig_args.scl->ival[0];
    return 0;
}`

Then why is it not able to detect I2C device without running i2cconfig first?
and i2cdetect command is not even using that structure, and it ultimately goes and uses the local variables which I set (showed in the problem statement)

`
static gpio_num_t i2c_gpio_sda = 14;
static gpio_num_t i2c_gpio_scl = 15;
static uint32_t i2c_frequency = 100000;
static i2c_port_t i2c_port = I2C_NUM_0;

static esp_err_t i2c_master_driver_initialize(void)
{
    i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = i2c_gpio_sda,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_io_num = i2c_gpio_scl,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = i2c_frequency
    };
    return i2c_param_config(i2c_port, &conf);
}

static int do_i2cdetect_cmd(int argc, char **argv)
{
    i2c_driver_install(i2c_port, I2C_MODE_MASTER, I2C_MASTER_RX_BUF_DISABLE, I2C_MASTER_TX_BUF_DISABLE, 0);
    i2c_master_driver_initialize();
    uint8_t address;
    printf("     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f\r\n");
    for (int i = 0; i < 128; i += 16) {
        printf("%02x: ", i);
        for (int j = 0; j < 16; j++) {
            fflush(stdout);
            address = i + j;
            i2c_cmd_handle_t cmd = i2c_cmd_link_create();
            i2c_master_start(cmd);
            i2c_master_write_byte(cmd, (address << 1) | WRITE_BIT, ACK_CHECK_EN);
            i2c_master_stop(cmd);
            esp_err_t ret = i2c_master_cmd_begin(i2c_port, cmd, 50 / portTICK_RATE_MS);
            i2c_cmd_link_delete(cmd);
            if (ret == ESP_OK) {
                printf("%02x ", address);
            } else if (ret == ESP_ERR_TIMEOUT) {
                printf("UU ");
            } else {
                printf("-- ");
            }
        }
        printf("\r\n");
    }

    i2c_driver_delete(i2c_port);
    return 0;
}
`

shivam5594, 2020-09-24:
This is my code. I used the entire i2cdetect function and made it to look for particular address 0x68. Still not able to detect. Am I missing any step?
```
void app_main(void)
{
    esp_err_t ret = ESP_FAIL;
    i2c_port_t I2Cport = I2C_NUM_0;

    i2c_config_t conf;
    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = 14;
    conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
    conf.scl_io_num = 15;
    conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
    conf.master.clk_speed = 100000;

    ESP_ERROR_CHECK(i2c_param_config(I2Cport, &conf));
    ESP_ERROR_CHECK(i2c_driver_install(I2Cport, I2C_MODE_MASTER, I2C_MASTER_RX_BUF_DISABLE, I2C_MASTER_TX_BUF_DISABLE, 0));

    uint8_t address;
    address = 0x68; // IMU I2C address
    while(1)
    {
	i2c_cmd_handle_t cmd = i2c_cmd_link_create();
	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (address << 1) | I2C_MASTER_WRITE, 0x1);
	i2c_master_stop(cmd);
	esp_err_t ret1 = i2c_master_cmd_begin(I2Cport, cmd, 50 / portTICK_RATE_MS);
	i2c_cmd_link_delete(cmd);
	if (ret == ESP_OK) {
		printf("Detected device at %02x ", address);
	} else if (ret1 == ESP_ERR_TIMEOUT) {
		printf("UU ");
	} else {
		printf("-- %d", ret1);
	}
	fflush(stdout);
    	vTaskDelay(1000);
    }
    ESP_ERROR_CHECK(ret);
}
```

felmue, 2020-09-24:
Hey @shivam5594 

I think there is only a small typo - mixup between `ret` and `ret1`.
This line: `if (ret == ESP_OK) {`
should be: `if (ret1 == ESP_OK) {`

Cheers
Felix

shivam5594, 2020-09-24:
@felmue thanks for that. But even after changing that to ret1, same result.

Here is my connection.

 [image]

shivam5594, 2020-09-24:
My bad, pins were swapped. Thank you Felix for answering my stupid questions 😊

closed 2020-09-24 by shivam5594

felmue, 2020-09-24:
Hi Shivam

no worries, you are very welcome. I am glad to hear you've got it sorted out and working.

Cheers
Felix