espressif/esp-idf #5901: WiFi isn't working with external 3.5V battery

tsarquis88, 2020-09-23:
## Environment

- Development Kit: **ESP32-DevKitC**
- Kit version (for WroverKit/PicoKit/DevKitC): **v4**
- Module or chip used: **ESP32-WROOM-32D**
- IDF version: **v4.0.1-441-g6093407d7**
- Build System: **idf.py**
- Compiler version: **xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0**
- Operating System: **Linux**
- Using an IDE?: **No**
- Power Supply: **external 3.3V "ncr18650b"**

## Problem Description

As the title says, I cannot use WiFi module while using an external 3.5V battery. The battery is connected directly to the 3V3 pin.

I can do other things, like blinking a LED.

On one hand, multimeter gives me the 3.5V expected value for the input supply, but on the other, the DevKit's red LED is quite blinking and not so shining like when I use the microUSB.

I've measured the input current and it oscillates from 40mA to 62mA in active mode, while I was waiting a current value higher than 160mA.

I don't know very much about batteries... I should consider trying another one? Maybe is discharged, despite the fact I've measured the 3.5V?

Thanks for any help

nx518, 2020-09-23:
Datasheet recommends at least 500mA supply for the module.
Try more batteries in parallel.

negativekelvin, 2020-09-24:
It's a 4 amp battery so either it is bad or there is a wiring problem

Francois-Belanger, 2020-09-28:
Hi, it's not always the battery, but the wiring.  How long are the wires ?  Should be pretty short.  To make sure, add some 100uF close to the module.

Alvin1Zhang, 2020-10-12:
Thanks for reporting, would you please help share if any updates for this issue? Thanks.

tsarquis88, 2020-10-12:
Sorry for the late reply. And thank you all for the answers.
I'm waiting for the arrival of another board where I'll do another try.

@nx518 for the moment I cannot do that, and as @negativekelvin said, the one battery should work.
@Francois-Belanger wires are pretty short. I'm currently working on a protoboard. And I do have that capacitor.

I promise to update this issue when I have news.

tsarquis88, 2020-10-22:
Okay, so I aquired the WROOM module and it worked.
As far as I know, the DevKit needs more current than just the module, so I think my battery is not at it's 100% of capacity for making the DevKit to run properly (or at least, for making the WiFi to turn on in the DevKit).

Thank you all, I'm closing this.

closed 2020-10-22 by tsarquis88