espressif/esp-idf #5902: ESP v3.3, spiffs write error

elieDaan, 2020-09-23:
## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version : v3.3-172-ge7e0ead31
- Build System:         CMake
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:    macOS

- Using an IDE?: Vs Studio Code
- Power Supply:      USB

## Problem Description

Hello, I'm using ESP32 on Amazon FreeRtos with the [loboris library  ](https://github.com/loboris/ESP32_TFT_library) library to use a tft lcd screen. 
In this library, there is a specific spiffs component so in my project I'm using this spiffs functions.  I must use this because with the spiffs esp idf component, it's impossible to display images on tft screen. 

The problem is that when my application is running, randomly a write on a file contained on the spiffs is not working and then every write/read is failed until I reset the board.  

On this example, I'm writing a structure into a file as binary.  It's the same for other functions... 

Thank you very much for your help. 
### Expected Behavior
fwrite/fread/fclose successed 

### Actual Behavior
Randomly Failed with error -10012 
### Steps to reproduce

1. Flash your code 
2. Flash your images with make flashfs 
3. Write a file and reboot until you've got the error 

### Code to reproduce this issue

```cpp
/* Function to write Parameters_bob into a file to save it    */
int writeParametersFile(char *fname, char *mode, Parameters_bob *parameters)
{   
    FILE *fd =NULL;
    #if PRINT_DEBUG
    printf("=======================\n");
    printf("==== Write the parameters settings to file ====\n");
    printf("=======================\n");
    printf("  file: \"%s\"\n", fname);
    #endif
    if (fileExists(fname)==0){
        fd = fopen(fname, "wb+");
        
    }
    else {
	    fd = fopen(fname, "wb+");
    }
    if (fd == NULL) {
        ESP_LOGE("[write]", "fopen failed");
        ESP_LOGE("[write]","errno : %d ",errno );
    	return -1;
    }
    int size = sizeof(Parameters_bob);
    
	int res = fwrite(parameters, size, 1, fd);
	if (res != 1) {
        ESP_LOGE("[write]", "fwrite failed: %d  ", res);
        ESP_LOGE("[write]","errno : %d ",errno );
        res = fclose(fd);
        if (res) {
            ESP_LOGE("[write]", "fclose failed: %d", res);
            ESP_LOGE("[write]","errno : %d ",errno );
            return -2;
        }
        return -3;
    }
	res = fclose(fd);
	if (res) {
        ESP_LOGE("[write]", "fclose failed: %d", res);
        ESP_LOGE("[write]","errno : %d ",errno );
    	return -4;
	}
    return 0;
}
```
=======================
==== Write the parameters settings to file ====
=======================
  file: "/spiffs/config_data.txt"
E (60734) [write]: fclose failed: -1
E (60734) [write]: errno : -10012 
```

## Other items if possible
[sdkconfig](https://gist.github.com/elieDaan/f5e1a5d1c01836e14d7dc928e77e57fd)

igrr, 2020-10-02:
Hi @elieDaan, it looks like the library you've mentioned uses a spiffs wrapper which is not derived from ESP-IDF code: https://github.com/loboris/ESP32_TFT_library/tree/master/components/spiffs. In general we don't offer support for third party code not included into ESP-IDF. If you are facing an issue with reading the images using the ESP-IDF spiffs component, please describe what you are seeing (preferably in a new issue) and we'll try to address this.

Alvin1Zhang, 2020-10-20:
@elieDaan Thanks for reporting. Would you please help share if any updates? Thanks.

elieDaan, 2020-10-27:
Hey, It seems that we have this issue due to output connected on the GPIO12 which is used internally for SPI. Do you know if it's okay to let GPIO12 floating to avoid this issue ?

mahavirj, 2021-01-11:
@elieDaan (In case your problem is not solved) Please see note about GPIO12 at https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#other-pins

mahavirj, 2021-01-11:
Closing the issue, please feel file new issue if there are any updates.

closed 2021-01-11 by mahavirj

elieDaan, 2021-01-11:
Yes, thank you. Finally, I've solved the issue by burning the fuse.