espressif/esp-idf #5904: export.bat didn't work to add environment variables

ctz16, 2020-09-23:
## Environment

- IDF version v4.1
- Operating System: Windows 10 professional 2004
- (Windows only) environment type: cmd (run as administrator)
- Using an IDE?: no

## Problem Description
I installed the esp-idf using vscode extension Onboard and things get done well. Then I tried export.bat to add environment variables but it didn't work. The output seems good as below:
```
Setting IDF_PATH: C:\Users\chang\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Users\chang\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin
    C:\Users\chang\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin
    C:\Users\chang\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\Users\chang\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Users\chang\.espressif\tools\cmake\3.13.4\bin
    C:\Users\chang\.espressif\tools\openocd-esp32\v0.10.0-esp32-20191114\openocd-esp32\bin
    C:\Users\chang\.espressif\tools\ninja\1.9.0\
    C:\Users\chang\.espressif\tools\idf-exe\1.0.1\
    C:\Users\chang\.espressif\tools\ccache\3.7\
    C:\Users\chang\.espressif\python_env\idf4.1_py3.8_env\Scripts
    C:\Users\chang\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from C:\Users\chang\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```
However, when I opened the environment variables, I found IDF_PATH was not set, and nothing added to PATH. 
Then I found `idf.py` worked well in the same console where I just used export.bat, but when I closed the console and opened a new one, it didn't work. It said `'idf.py' is not recognized as an internal or external command`
Well I think it cannot be designed to export everytime I want to use `idf.py`. So there must be something wrong.

dobairoland, 2020-09-23:
Hi @ctz16. Yes, this is intended behaviour. You are supposed to run export.bat in each terminal you want to use idf.py. It will not affect the global settings. It is done this way so you could use different versions of IDFs in each terminal. There is a convenience shortcut installed with IDF tools installer which would start a command line with already setup environment.

In Vscode, you directly modify settings.json to set up the environment:
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md#directly-modify-settingsjson

ctz16, 2020-09-24:
OK, thank you for your support@dobairoland , seems my misunderstanding. I get the vscode extension works well and that's enough to use.

closed 2020-09-24 by ctz16