espressif/esp-idf #5905: secure boot V2: BLK2 and the signature block don't match

sicanins, 2020-09-23:
## Environment

- Module or chip used:   ESP32-WROVER-E
- IDF version: v4.3-dev-911-g740827c41

## Problem Description

OTA update fails. Environment is like this: the module was previously used with a master version of mid-May. Updates were working fine at until I updated the esp-idf. 

Now the ota_end() fails with error:

E (143574) secure_boot_v2: Public key digest in eFuse BLK2 and the signature block don't match.
E (143576) esp_image: Secure boot signature verification failed

In order to track that problem down I am looking at the code that causes this error message.
But seems I am lost understanding what its supposed to do:

```
esp_err_t esp_secure_boot_verify_rsa_signature_block(const ets_secure_boot_signature_t *sig_block, const uint8_t *image_digest, uint8_t *verified_digest)
{
    uint8_t i = 0;
#if CONFIG_SECURE_BOOT_V2_ENABLED /* Verify key against efuse block */
    uint8_t sig_block_key_digest[SECURE_BOOT_NUM_BLOCKS][DIGEST_LEN] = {0};

    /* Note: in IDF verification we don't add any fault injection resistance, as we don't expect this to be called
       during boot-time verification. */
    memset(verified_digest, 0, DIGEST_LEN);

#if CONFIG_IDF_TARGET_ESP32
    uint8_t efuse_trusted_digest[DIGEST_LEN] = {0};
    memcpy(efuse_trusted_digest, (uint8_t *) EFUSE_BLK2_RDATA0_REG, sizeof(efuse_trusted_digest));

    if (memcmp(efuse_trusted_digest, sig_block_key_digest, DIGEST_LEN) != 0) {
        const uint8_t zeroes[DIGEST_LEN] = {0};
        /* Can't continue if secure boot is enabled, OR if a different digest is already written in efuse BLK2

           (If BLK2 is empty and Secure Boot is disabled then we assume that it will be enabled later.)
         */
        if (esp_secure_boot_enabled() || memcmp(efuse_trusted_digest, zeroes, DIGEST_LEN) != 0) {
            ESP_LOGE(TAG, "Public key digest in eFuse BLK2 and the signature block don't match.");
            return ESP_FAIL;
        }
    }
```

In my understanding, that code can only pass if the BLK2 is all zero, which it should not be when a secure key is written, right?
My guess is, that the code is supposed to compare against the key in sig_block_key_digest, but that is zero at this time and only written a few lines later.

Any help here would be highly appreciated, maybe someone familiar with that code can point out what I am thinking wrong.

Thanks,
Simon

Alvin1Zhang, 2020-09-24:
Thanks for the detailed report and letting us know, and sorry for the inconvenience. We will look into.

negativekelvin, 2020-09-24:
Move these lines up to before the `#if CONFIG_IDF_TARGET_ESP32`

https://github.com/espressif/esp-idf/blob/c77c4ccf6c43ab09fd89e7c907bf5cf2a3499e3b/components/bootloader_support/src/idf/secure_boot_signatures.c#L235-L241

sicanins, 2020-09-24:
@negativekelvin @Alvin1Zhang 
Great thanks, that solved it!

Will you guys close the issue once that is committed?

projectgus, 2020-09-25:
HI @sicanins ,

Yes, the issue will be closed on this fix is available on master. Sorry for the inconvenience caused by this regression.

Angus

sicanins, 2020-09-25:
Hi @projectgus,

thanks, I am happy that it got resolved so quickly!

Simon

closed 2020-10-15 by espressif-bot