espressif/esp-idf #5906: Misleading MQTT_ERROR_TYPE_ESP_TLS error_type when publish failed with TLS disabled

AxelLin, 2020-09-24:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v4.3-dev-1197-g8bc19ba893e5
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

I got MQTT_EVENT_ERROR when publish failed, then check the error_type from event->error_handle->error_type.
It shows MQTT_ERROR_TYPE_ESP_TLS even though I don't use TLS at all.

### Expected Behavior
The error_type should make it easier to identify the reason of error.

### Actual Behavior
It shows MQTT_ERROR_TYPE_ESP_TLS even when TLS is not used at all.

### Steps to reproduce

1. connect to a mqtt broker (use TCP only, don't enable TLS) and keep publishing data.
2. stop the broker
3. the mqtt client got MQTT_EVENT_ERROR but the error_type is MQTT_ERROR_TYPE_ESP_TLS which is obviously wrong.

Alvin1Zhang, 2020-09-24:
Thanks for reporting, we will look into.

AxelLin, 2020-11-18:
@david-cermak 
I got below build error when I build with latest esp-idf (v4.3-dev-1720-g494a124d961c)  + master tree of esp-mqtt.
Maybe it's just because some changes in esp-idf is not yet sync to github.
Just FYI, in case if something missed...

error: implicit declaration of function 'esp_transport_get_errno'; did you mean 'esp_transport_set_func'? [-Werror=implicit-function-declaration]
         client->event.error_handle->esp_transport_sock_errno = esp_transport_get_errno(client->transport);
                                                                ^~~~~~~~~~~~~~~~~~~~~~~
                                                                esp_transport_set_func

david-cermak, 2020-11-18:
Thanks for the note. yes, correct, the reason is that the latest mqtt needs the latest idf (to get the related socket's `errno` in case of a failure) which hasn't been yet pushed to GitHub.

closed 2020-12-13 by espressif-bot