espressif/esp-idf #5907: DFS(dynamic frequency scaling) changes pwm frequency

thedayu, 2020-09-24:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version v4
- Module or chip used:ESP32-WROOM-32U|ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it):
v4.0
- Build System:         CMake|idf.py
- Operating System:     Linux
- Using an IDE: VSCODE
- Power Supply:         USB

## Problem Description

PWM frequency changes over time.

//Detailed problem description goes here.

### Expected Behavior

API should do as it says, and PWM freq should not change.
` uint32_t freq_hz;                      /*!< LEDC timer frequency (Hz) */
`

### Actual Behavior

Changes when open when BLE scan/WIFI data, etc.

### Steps to reproduce

Enable DFS, produce pwm output, use a oscilloscope to monitor pin accordingly.

```
ledc_timer_config_t conf = {
			.duty_resolution = LEDC_TIMER_8_BIT,
			.freq_hz = 1000,
			.speed_mode = LEDC_HIGH_SPEED_MODE,
			.timer_num = 0
	};
	ERROR_CHECK(ledc_timer_config(&conf));
	ledc_channel_config_t conf = {
		.channel    = 0,
		.duty       = 128,
		.gpio_num   = pin,
		.speed_mode = LEDC_HIGH_SPEED_MODE,
		.hpoint     = 0,
		.timer_sel  = 0
	};
	ledc_channel_config(&conf);
```
```

Alvin1Zhang, 2020-09-24:
Thanks for reporting, we will look into.

igrr, 2020-09-24:
Hi @thedayu, as mentioned in https://docs.espressif.com/projects/esp-idf/en/v4.0.1/api-reference/system/power_management.html#dynamic-frequency-scaling-and-peripheral-drivers, LEDC can be configured with REF_TICK clock source to prevent the frequency changing when DFS is enabled. You can set LEDC_USE_REF_TICK value for clk_cfg in the ledc_timer_config_t structure.

thedayu, 2020-09-24:
@igrr That's it. Thanks for quick reply.

closed 2020-09-24 by thedayu

InfiniteYuan, 2020-10-28:
@igrr When the LEDC_USE_REF_TICK clock source is configured for LEDC to prevent frequency changes, the frequency of LEDC clock source is reduced to 1MHz. 

The frequency of LEDC clock source is not meet my application needs. 

Do you have a good way?

igrr, 2020-10-28:
Per the docs, "For low speed channels, you can also specify the source clock using LEDC_USE_RTC8M_CLK, in this case, all low speed channel’s source clock must be RTC8M_CLK". This should raise the frequency to ~8 MHz. Higher LEDC clock frequencies are not possible when APB clock frequency of the ESP32 is changing, I'm afraid.