espressif/esp-idf #5908: ./install.sh doesn't work for ubuntu 20.04

ThomasGeor, 2020-09-24:
I am trying to make an esp idf installation in my ubuntu 20.04 system from scratch following the ESP documentation. My problem is when i run ./install.sh script which fails in :
${IDF_PATH}/tools/idf_tools.py install-python-env.
The reason for this is that pip is not supported in Ubuntu 20.04. 
How do i fix this? I tried to change the idf_tools.py 'pip' argument to 'pip3' but it doesn't seem to work.
Is there any solution for that to continue my installation?

The logs are presented below : 
./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Installing Python environment and packages
Creating a new Python environment in /home/thomas/.espressif/python_env/idf4.3_py2.7_env
Installing virtualenv
/usr/bin/python: No module named pip3
Traceback (most recent call last):
  File "/home/thomas/esp/esp-idf/tools/idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "/home/thomas/esp/esp-idf/tools/idf_tools.py", line 1488, in main
    action_func(args)
  File "/home/thomas/esp/esp-idf/tools/idf_tools.py", line 1205, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'pip3', 'install', '--user', 'virtualenv']' returned non-zero exit status 1

igrr, 2020-09-24:
Hi @ThomasGeor, I have just tried to install ESP-IDF on Ubuntu 20.04.

install.sh should be successful once you create a `python` symlink pointing to your python3 interpreter:

```
ln -s `which python3` ~/.local/bin/python
```

(assuming you have a .local/bin directory created and added to the PATH; if not, you can use /usr/local/bin instead).

Could you please revert the change to idf_tools.py, create the symlink, and see if that works?

ThomasGeor, 2020-09-25:
**Since i don't have .local/bin directory added to the PATH, i used  /usr/local/bin to create a symlink instead. I reverted the changes in idf_tools.py and tried to run  ./install.sh again but run into the same error logs:**

Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Installing Python environment and packages
Creating a new Python environment in /home/thomas/.espressif/python_env/idf4.3_py2.7_env
Installing virtualenv
/usr/bin/python: No module named pip
Traceback (most recent call last):
  File "/home/thomas/esp/esp-idf/tools/idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "/home/thomas/esp/esp-idf/tools/idf_tools.py", line 1488, in main
    action_func(args)
  File "/home/thomas/esp/esp-idf/tools/idf_tools.py", line 1205, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'pip', 'install', '--user', 'virtualenv']' returned non-zero exit status 1

igrr, 2020-09-25:
After creating the symlink, could you check whether python3 becomes available as the default "python"?

I.e. `which python` should return the python interpreter pointed to by the symlink, and `python --version` should print "Python 3.x.y", and same should happen when you run `/usr/bin/env python --version`.

The error indicates that /usr/bin/python, which is a python 2.7 interpreter, is still being used. Since there is no `pip` installed for python 2, the installation fails. I think once you make sure that `/usr/bin/env python --version` reports Python 3, the installation should succeed.

ThomasGeor, 2020-09-30:
Sorry fot the late response , i had some other business to take care of. You are correct about the interpreter. After running python --version, i am getting **Python 2.7.18rc1** . How can i change that to python 3 which should be used?

igrr, 2020-09-30:
Probably the easiest option with the out-of-the-box Ubuntu 20.04 is:

1. Create $HOME/bin directory: `mkdir -p $HOME/bin`. You can also use `$HOME/.local/bin`, if you prefer.
2. Create a symlink to a Python 3 interpreter in that directory: `ln -s $(which python3) $HOME/bin/python`
3. If you are in a console session, log out and log back in. If you are in a GUI session, you can close the terminal window and open a new one. The $HOME/.profile script provided in Ubuntu 20.04 will automatically add the `bin` directory to your `PATH`.
4. Run `python --version` again and verify it now reports Python 3. If not, run `which python` — it should print the path of the Python interpreter, and `echo $PATH` — it should print the list of directories added to `PATH`.

ThomasGeor, 2020-09-30:
Problem solved using this command :
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 10

igrr, 2020-09-30:
Thanks for posting the steps, @ThomasGeor.

Let's keep this issue open until we either update the scripts to pick `python3` when that is available, or update the documentation to give detailed instructions for Ububtu 20.04.

chegewara, 2020-09-30:
I am working on Ubuntu 20.04 for quit some time and i am using `update-alternatives` to select python3:
https://linuxconfig.org/ubuntu-20-04-python-version-switch-manager

rweller59, 2020-12-09:
I have a fresh install of ubuntu 20.04. Using vscode with extension. Followed prereqs in doc for cmake, ninja, python. then used extension. Got errors that python couldn't be found and partially solved with 'sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 1'

I am now stuck on error messages 
  The following Python requirements are not satisfied:
  pyparsing>=2.0.3,<2.4.0
  pyelftools>=0.22

Any suggestions?

chegewara, 2020-12-09:
https://esp32.com/viewtopic.php?f=40&t=18468&sid=b46c78d42d9e4ce91bc2494f578dc71f

dobairoland, 2020-12-09:
> Let's keep this issue open until we either update the scripts to pick python3 when that is available, or update the documentation to give detailed instructions for Ububtu 20.04.

This is now included in the documentation:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html#setting-up-python-3-as-default-for-ubuntu-and-debian

Also a nice page. https://linuxconfig.org/ubuntu-20-04-python-version-switch-manager. Thanks @chegewara.

closed 2020-12-09 by dobairoland

dobairoland, 2020-12-09:
@rweller59 Beside the nice tutorial linked by Chegewara, let me explain what is behind your issue.

There are probably two or more Python interpreters in your computer. You are installing packages for one of them but Vscode is linked to use the other one. You can solve your issue if you run `python -m pip install -r $IDF_PATH/requirements.txt` once you are inside your Vscode with activated IDF environment.

rweller59, 2020-12-09:
Thanks for the help, I can now build. 

One thing, the espidf docs default to installing in /esp/esp-idf, the extension defaults to /esp-idf/esp-idf. Simple thing but would be nice if they were consistent.