espressif/esp-idf #5912: Secure Boot can not used

poisonousfish, 2020-09-25:
# Environment
* Development Kit: none
* Module or chip used: ESP32-WROOM-3D
* IDF version : ESP-IDF v4.1
* Build System: idf.py
* Compiler version : (crosstool-NG esp-2020r2) 8.2.0
* Operating System: windows10
* Using an IDE?: no，just idf
* Power Supply: USB

# Problem Description

According to https://docs.espressif.com/projects/esp-idf/en/v4.1/security/secure-boot-v1.html#keys 
1. open “Secure Boot Configuration”
[image]

2. change offset partition table to 0x9000
[image]

3. generate a new signing key by running the following command:
`espsecure.py generate_signing_key secure_boot_signing_key.pem`
like this
[image]

4. falsh bootloader by running following commands
```
idf.py bootloader
idf.py -p com3  bootloader-flash
```
everything like fine
[image]

5. falsh by the running following command

`idf.py -p com3 all flash monitor`
or
`idf.py -p com3 flash monitor`

but it not working
[image]

6. i try check efuse by the running following command

`espefuse.py -p com3 summary`
 
I found that bootLoader didn't seem to be working
[image]

FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
UART_DOWNLOAD_DIS      Disable UART download mode (ESP32 rev3 only)      = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK2                   Secure boot key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

I couldn't find where it was wrong. Can you help me?

Thank you so much.

negativekelvin, 2020-09-25:
If the bootloader is 34784 bytes and starts at 0x1000 it will end at 0x97e0 so you can't put the partition table at 0x9000

ringo156, 2020-12-07:
@poisonousfish @negativekelvin 
I was worried about a similar error.
The solution was to change the offset partition table to 0xA000.
Thank you!

projectgus, 2021-03-12:
Closing as this seems to be resolved by moving the partition table.

More information about this can be found in the Secure Boot docs:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html#bootloader-size

closed 2021-03-12 by projectgus