espressif/esp-idf #5915: WiFi randomly disconnect from AP with disconnect reason WIFI_REASON_ASSOC_LEAVE

AxelLin, 2020-09-26:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v4.3-dev-1197-g8bc19ba893e5
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

This seems happen randomly when I test with join TP-LINK TL-WR1043ND AP.
From below log, the RSSI is prettyr good.

I (4194) wifi:wifi driver task: 3ffc6870, prio:23, stack:6656, core=0
I (4214) wifi:wifi firmware version: d3be909
I (4214) wifi:wifi certification version: v7.0
I (4214) wifi:config NVS flash: enabled
I (4215) wifi:config nano formating: disabled
I (4219) wifi:Init data frame dynamic rx buffer num: 32
I (4224) wifi:Init management frame dynamic rx buffer num: 32
I (4229) wifi:Init management short buffer num: 32
I (4234) wifi:Init dynamic tx buffer num: 32
I (4238) wifi:Init static rx buffer size: 1600
I (4242) wifi:Init static rx buffer num: 10
I (4246) wifi:Init dynamic rx buffer num: 32
I (13875) wifi:set country: cc=US schan=1 nchan=11 policy=0
I (13982) wifi:mode : sta (f0:08:d1:79:8c:28)
I (14711) wifi:new:<6,2>, old:<1,0>, ap:<255,255>, sta:<6,2>, prof:1
I (15351) wifi:state: init -> auth (b0)
I (15357) wifi:state: auth -> assoc (0)
I (15380) wifi:state: assoc -> run (10)
I (15401) wifi:connected with TEST-AP, aid = 3, channel 6, 40D, bssid = 18:a6:f7:ab:ca:66
I (15402) wifi:security: WPA2-PSK, phy: bgn, rssi: -43
I (15404) wifi:pm start, type: 1
I (15469) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (16217) wifi:Set ps type: 1
I (253848) wifi:state: run -> init (ce00)
I (253850) wifi:pm stop, total sleep time: 184541140 us / 238443301 us
I (253851) wifi:new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1

Check with disconnected->reason, it'ss 8  (WIFI_REASON_ASSOC_LEAVE)

### Expected Behavior
Should not disconnect from AP randomly.

### Actual Behavior
Disconnect from AP with WIFI_REASON_ASSOC_LEAVE.

### Steps to reproduce

Just join an AP and wait... (it happens at random time so sometimes need to wait longer time)

AxelLin, 2020-09-28:
Below shows the log the device disconnect from AP several times.

I (4182) wifi:wifi driver task: 3ffc68dc, prio:23, stack:6656, core=0
I (4202) wifi:wifi firmware version: d3be909
I (4202) wifi:wifi certification version: v7.0
I (4202) wifi:config NVS flash: enabled
I (4203) wifi:config nano formating: disabled
I (4207) wifi:Init data frame dynamic rx buffer num: 32
I (4212) wifi:Init management frame dynamic rx buffer num: 32
I (4217) wifi:Init management short buffer num: 32
I (4222) wifi:Init dynamic tx buffer num: 32
I (4226) wifi:Init static rx buffer size: 1600
I (4230) wifi:Init static rx buffer num: 10
I (4234) wifi:Init dynamic rx buffer num: 32
I (4238) wifi_init: rx ba win: 6
I (4241) wifi_init: tcpip mbox: 32
I (4246) wifi_init: udp mbox: 6
I (4249) wifi_init: tcp mbox: 6
I (4253) wifi_init: tcp tx win: 5744
I (4257) wifi_init: tcp rx win: 5744
I (4262) wifi_init: tcp mss: 1440
I (4266) wifi_init: WiFi IRAM OP enabled
I (4270) wifi_init: WiFi RX IRAM OP enabled
I (13853) wifi:set country: cc=US schan=1 nchan=11 policy=0
I (13960) wifi:mode : sta (f0:08:d1:79:8c:28)
I (14690) wifi:new:<6,2>, old:<1,0>, ap:<255,255>, sta:<6,2>, prof:1
I (15329) wifi:state: init -> auth (b0)
I (15348) wifi:state: auth -> assoc (0)
I (15363) wifi:state: assoc -> run (10)
I (15373) wifi:connected with TEST-AP, aid = 1, channel 6, 40D, bssid = 18:a6:f7:ab:ca:66
I (15373) wifi:security: WPA2-PSK, phy: bgn, rssi: -39
I (15375) wifi:pm start, type: 1
I (15445) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (16206) wifi:Set ps type: 1
I (3578929) wifi:state: run -> init (ce00)
I (3579099) wifi:new:<6,2>, old:<6,0>, ap:<255,255>, sta:<6,2>, prof:1
I (3579103) wifi:state: init -> auth (b0)
I (3579128) wifi:state: auth -> assoc (0)
I (3579182) wifi:state: assoc -> run (10)
I (3579197) wifi:connected with TEST-AP, aid = 1, channel 6, 40D, bssid = 18:a6:f7:ab:ca:66
I (3579199) wifi:security: WPA2-PSK, phy: bgn, rssi: -31
I (3579202) wifi:pm start, type: 1
I (3579236) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3580207) wifi:Set ps type: 1
I (4700114) wifi:state: run -> init (ce00)
I (4700117) wifi:pm stop, total sleep time: 1054343312 us / 1120904987 us
I (4700117) wifi:new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
I (4700266) wifi:new:<6,2>, old:<6,0>, ap:<255,255>, sta:<6,2>, prof:1
I (4700268) wifi:state: init -> auth (b0)
I (4700288) wifi:state: auth -> assoc (0)
I (4700304) wifi:state: assoc -> run (10)
I (4700312) wifi:connected with TEST-AP, aid = 1, channel 6, 40D, bssid = 18:a6:f7:ab:ca:66
I (4700313) wifi:security: WPA2-PSK, phy: bgn, rssi: -38
I (4700315) wifi:pm start, type: 1
I (4700391) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4700392) wifi:state: run -> init (ce00)
I (4700393) wifi:pm stop, total sleep time: 0 us / 75236 us
I (4700395) wifi:new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
I (4700547) wifi:new:<6,2>, old:<6,0>, ap:<255,255>, sta:<6,2>, prof:1
I (4700549) wifi:state: init -> auth (b0)
I (4700552) wifi:state: auth -> assoc (0)
I (4700563) wifi:state: assoc -> run (10)
I (4700572) wifi:connected with TEST-AP, aid = 1, channel 6, 40D, bssid = 18:a6:f7:ab:ca:66
I (4700573) wifi:security: WPA2-PSK, phy: bgn, rssi: -37
I (4700575) wifi:pm start, type: 1
I (4700598) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4701207) wifi:Set ps type: 1
I (8496342) wifi:state: run -> init (ce00)
I (8496345) wifi:pm stop, total sleep time: 3567685118 us / 3795766526 us
I (8496346) wifi:new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
[... 272 lines cut ...]
I (62219708) wifi:Set ps type: 1
I (62230943) wifi:state: run -> init (ce00)
I (62230946) wifi:pm stop, total sleep time: 9631190 us / 11864185 us
I (62230946) wifi:new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
I (62231096) wifi:new:<6,2>, old:<6,0>, ap:<255,255>, sta:<6,2>, prof:1
I (62231098) wifi:state: init -> auth (b0)
I (62231101) wifi:state: auth -> assoc (0)
I (62231112) wifi:state: assoc -> run (10)
I (62231121) wifi:connected with TEST-AP, aid = 1, channel 6, 40D, bssid = 18:a6:f7:ab:ca:66
I (62231121) wifi:security: WPA2-PSK, phy: bgn, rssi: -37
I (62231124) wifi:pm start, type: 1
I (62231147) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (62231707) wifi:Set ps type: 1
I (62308341) wifi:state: run -> init (ce00)
I (62308343) wifi:pm stop, total sleep time: 71955625 us / 77216118 us
I (62308344) wifi:new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
I (62308493) wifi:new:<6,2>, old:<6,0>, ap:<255,255>, sta:<6,2>, prof:1
I (62308495) wifi:state: init -> auth (b0)
I (62308504) wifi:state: auth -> assoc (0)
I (62308519) wifi:state: assoc -> run (10)
I (62308548) wifi:connected with TEST-AP, aid = 1, channel 6, 40D, bssid = 18:a6:f7:ab:ca:66
I (62308548) wifi:security: WPA2-PSK, phy: bgn, rssi: -38
I (62308551) wifi:pm start, type: 1
I (62308647) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (62309209) wifi:Set ps type: 1

AxelLin, 2020-09-30:
My previous log uses WIFI_PS_MIN_MODEM.
Also has similar disconnect issue with WIFI_PS_NONE. (see below log)

I (1733) wifi:set country: cc=US schan=1 nchan=11 policy=0
I (1826) wifi:mode : sta (f0:08:d1:79:8c:60)
I (1950) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (2483) wifi:state: init -> auth (b0)
I (2488) wifi:state: auth -> assoc (0)
I (2499) wifi:state: assoc -> run (10)
I (2509) wifi:connected with TEST-AP, aid = 1, channel 1, 40U, bssid = 18:a6:f7:ab:ca:66
I (2510) wifi:security: WPA2-PSK, phy: bgn, rssi: -36
I (2511) wifi:pm start, type: 1
I (2527) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3463) wifi:Set ps type: 0
I (1803977) wifi:state: run -> init (ce00)
I (1803978) wifi:pm stop, total sleep time: 578233 us / 1801463917 us
I (1803979) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (1804127) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (1804129) wifi:state: init -> auth (b0)
I (1804155) wifi:state: auth -> assoc (0)
I (1804168) wifi:state: assoc -> run (10)
I (1804177) wifi:connected with TEST-AP, aid = 1, channel 1, 40U, bssid = 18:a6:f7:ab:ca:66
I (1804178) wifi:security: WPA2-PSK, phy: bgn, rssi: -39
I (1804180) wifi:pm start, type: 0
I (1804184) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1804967) wifi:Set ps type: 0
I (1837167) wifi:state: run -> init (ce00)
I (1837169) wifi:pm stop, total sleep time: 0 us / 32985686 us
I (1837169) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (1837305) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (1837307) wifi:state: init -> auth (b0)
I (1837312) wifi:state: auth -> assoc (0)
I (1837325) wifi:state: assoc -> run (10)
I (1837334) wifi:connected with TEST-AP, aid = 1, channel 1, 40U, bssid = 18:a6:f7:ab:ca:66
I (1837335) wifi:security: WPA2-PSK, phy: bgn, rssi: -40
I (1837337) wifi:pm start, type: 0
I (1837383) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1837966) wifi:Set ps type: 0

DonSimonus, 2020-09-30:
### I have the similar issue on a different IDF

## Environment
- Development Kit:      None
- Module or chip used: ESP32-WROVER-E
- IDF version: v4.0.1-412-g820621687
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Windows|
- Environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:        External 3.3V

## Problem Description
In the environment of our customer, the ESP disconnects continuously from AP with the WIFI_REASON_ASSOC_LEAVE reason. There is no problem with mine or any other tested environments yet. Our customer have many APs with the same SSIDs active in the system. 
Another problem is that the ESP does not find all visible networks (the strongest RSSI with the same SSID will not be found).

## Steps to reproduce
Hard to say. Maybe there is a problem with multiple APs with the same SSIDs which should be connected.

## Debug Logs

```
... WiFi is connected... and disconnects...

I (410048) wifi:state: run -> init (ce00)
D (410049) wifi:connect status 5 -> 6
D (410050) wifi:stop beacon/connect timer, send diassoc(8)
D (410052) wifi:sta leave
I (410053) wifi:pm stop, total sleep time: 17275222 us / 28990895 us

D (410065) wifi:stop CSA timer
D (410065) wifi:remove c0:ea:e4:e5:a6:8c from rc list
I (410067) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
D (410078) wifi:filter: set rx policy=8
D (410079) wifi:Send disconnect event, reason=8, AP number=0
D (410092) wlan: WIFI_EVENT_STA_DISCONNECTED(reason=8)
D (410094) wifi:Start wifi connect
D (410095) wifi:connect status 6 -> 0
D (410095) wifi:connect chan=0
D (410106) wifi:first chan=1
D (410107) wifi:connect status 0 -> 1
D (410108) wifi:filter: set rx policy=3
D (410110) wifi:clear scan ap list
D (410110) wifi:start scan: type=0x50f, priority=2, cb=0x4017b814, arg=0x0, ss_state=0x1, time=408334066, index=0
0x4017b814: cnx_start_handoff_cb at ??:?

D (410123) wifi:perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
I (410136) wlan: Sending WLAN_CONNECTING(error=ESP_OK)
D (410140) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (410142) wifi:profile match: ss_state=0x7
D (410144) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (410156) wifi:profile match: ss_state=0x7
D (410375) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (410377) wifi:find first mathched ssid, scan done
D (410378) wifi:filter: set rx policy=4
D (410379) wifi:first chan=1
D (410390) wifi:handoff_cb: status=0
D (410391) wifi:ap found, mac=c0:ea:e4:e5:a6:8c
D (410392) wifi:new_bss=0x3ffca568, cur_bss=0x0, new_chan=<1,0>, cur_chan=1
D (410404) wifi:filter: set rx policy=5
I (410405) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
D (410406) wifi:connect_op: status=0, auth=5, cipher=3
D (410418) wifi:auth mode is not none
D (410419) wifi:connect_bss: auth=1, reconnect=0
I (410420) wifi:state: init -> auth (b0)
D (410422) wifi:start 1s AUTH timer
D (410434) wifi:clear scan ap list
[... 165 lines cut ...]
I (491020) wifi:state: assoc -> run (10)
D (491022) wifi:start 10s connect timer for 4 way handshake
I (491057) wifi:connected with AP-WLAN-Partner, aid = 1, channel 1, BW20, bssid = c0:ea:e4:e5:8b:32
I (491059) wifi:security: WPA2-PSK, phy: bgn, rssi: -86
D (491060) wifi:remove all except c0:ea:e4:e5:8b:32 from rc list
D (491072) wifi:clear blacklist
D (491076) wifi:filter: set rx policy=7
I (491077) wifi:pm start, type: 1

D (491078) wifi:Send sta connected event
D (491090) wifi:connect status 1 -> 5
D (491091) wifi:obss scan is disabled
D (491091) wifi:start obss scan: obss scan is stopped
D (491092) wlan: WIFI_EVENT_STA_CONNECTED()
D (491168) wifi:obss scan is disabled
D (491169) wifi:start obss scan: obss scan is stopped
I (491629) wifi:AP's beacon interval = 1024000 us, DTIM period = 1
D (491632) wifi:set max rate: from <rate=130, phy=3, sig=0> to <rate=144, phy=3 sig=0>
D (491635) wifi:sig_b=0, sig_g=0, sig_n=0, max_b=22, max_g=108, max_n=144
D (491646) wifi:update trc
D (496368) wlan: IP_EVENT_STA_GOT_IP(172.17.100.92)
I (496370) wlan: Sending WLAN_CONNECTED_INET(error=ESP_OK)
I (496371) wlan: Sending WLAN_TIME_SYNC(error=ESP_OK)
I (496383) tcpip_adapter: sta ip: 172.17.100.92, mask: 255.255.255.0, gw: 172.17.100.1
```

AxelLin, 2020-10-04:
My test environment only has one AP named TEST-AP, It's not related to multiple APs with the same SSIDs in my case.

liuzfesp, 2020-10-12:
Hi @AxelLin @DonSimonus, the WiFi is disconnected because the AP has reset its TSF timer, in other words, the AP may has problem about its TSF timer.

The disconnect reason code (0xce00) is: `WIFI_REASON_AP_TSF_RESET`, it indicates that the TSF value in AP's beacon has been reset. For example, the TSF value in last beacon of AP is 0x123456789 (last_tsf_value), the TSF value in new beacon of AP is 0x12345678 (new_tsf_value), if new_tsf_value is less than last_tsf_value, ESP32 STA will disconnect the WiFi and raise reason code `WIFI_REASON_AP_TSF_RESET`.

AxelLin, 2020-10-12:
> Hi @AxelLin @DonSimonus, the WiFi is disconnected because the AP has reset its TSF timer, in other words, the AP may has problem about its TSF timer.
> 
> The disconnect reason code (0xce00) is: `WIFI_REASON_AP_TSF_RESET`, it indicates that the TSF value in AP's beacon has been reset. 

Hi @liuzfesp ,

I don't get it.
No idea which part of the log shows 0xce00 as you mentioned?
In my original report, I mentioned disconnected->reason, it's 8 (WIFI_REASON_ASSOC_LEAVE).

zhangyanjiaoesp, 2020-10-12:
> Hi @liuzfesp ,
>
> I don't get it.
> No idea which part of the log shows 0xce00 as you mentioned?
> In my original report, I mentioned disconnected->reason, it's 8 (WIFI_REASON_ASSOC_LEAVE).

It's `I (253848) wifi:state: run -> init (ce00)`

AxelLin, 2020-10-12:
> > Hi @liuzfesp ,
> > I don't get it.
> > No idea which part of the log shows 0xce00 as you mentioned?
> > In my original report, I mentioned disconnected->reason, it's 8 (WIFI_REASON_ASSOC_LEAVE).
> 
> Hi @AxelLin, It's `I (253848) wifi:state: run -> init (ce00)`. Can you provide a capture of your issues?

Hi @zhangyanjiaoesp 
When I got WIFI_EVENT_STA_DISCONNECTED, I use below code to get disconnected reason.
wifi_event_sta_disconnected_t *disconnected = (wifi_event_sta_disconnected_t *) event_data;
ESP_LOGI(TAG,"Disconnect reason: %d", disconnected->reason);

So why I got 8 WIFI_REASON_ASSOC_LEAVE rather than 206 WIFI_REASON_AP_TSF_RESET.

This is probably off-topic, but I hope disconnected->reason shows correct disconnect reason.

zhangyanjiaoesp, 2020-10-12:
> Hi @zhangyanjiaoesp
> When I got WIFI_EVENT_STA_DISCONNECTED, I use below code to get disconnected reason.
> wifi_event_sta_disconnected_t *disconnected = (wifi_event_sta_disconnected_t *) event_data; [...]

The reason code `WIFI_REASON_AP_TSF_RESET(206)` is defined by ourself, it's not the standard reason in the protocol, when we want to send deauth/disassoc packets, we need to add the `reason code` IE , so we change the reason to `WIFI_REASON_ASSOC_LEAVE(8)`, thus the receiver can parse it correctly.

[image]
[image]

gecko242, 2020-10-12:
Huh, this is all looking very familiar!
I am experiencing very similar problems. Previously, I have had no issues connecting to wifi, and staying connected. Unfortunately I cannot put an exact date to when it started, but recently I have been experiencing similar dropouts. 

Typically these occur 15s - 1 minute after connecting. 

Module or chip used: ESP32-WROOM-32U 16MB
IDF version: v4.3-dev-472-gcf056a7d0
Build System: idf.py
Operating System: Windows
Power Supply: Custom PCB

```
I (38292) wifi:state: run -> init (ce00)
I (38292) wifi:pm stop, total sleep time: 11798559 us / 14399660 us

I (38292) wifi:new:<11,0>, old:<11,0>, ap:<11,2>, sta:<11,0>, prof:1
I (06:36:23.724) SafeGuard.network: Disconnect Reason: 8
```

I have recently been changing some WiFI stuff, but nothing directly related to the connection configuration.

DonSimonus, 2020-10-12:
> Hi @AxelLin @DonSimonus, the WiFi is disconnected because the AP has reset its TSF timer, in other words, the AP may has problem about its TSF timer.
> 
> The disconnect reason code (0xce00) is: `WIFI_REASON_AP_TSF_RESET`, it indicates that the TSF value in AP's beacon has been reset. For example, the TSF value in last beacon of AP is 0x123456789 (last_tsf_value), the TSF value in new beacon of AP is 0x12345678 (new_tsf_value), if new_tsf_value is less than last_tsf_value, ESP32 STA will disconnect the WiFi and raise reason code `WIFI_REASON_AP_TSF_RESET`.
> 
> If the AP has TSF timer issue, I guess the issue should be stable and easy to be reproduced. Suggest to:
> 
> * Change to a different AP and do a test
> * Still use the same AP and capture the WiFi packets and we will double confirm the AP's TSF issue via packets

Hi @liuzfesp 
Unfortunately I can't take care of the problem this week because I'm on vacation.

Our customer has already tested it with a different AP (e.g. by a hotspot on a smartphone, AP at home etc.) and there it worked without problems.
It sounds like a problem with the APs at the customer's company environment.

Is this a known behavior of APs with this TSF reset, or what is the trigger for that?
How is that meant by "capture the WiFi packets"? Extend the debug log on the ESP or record the packets via a sniffer on the AP?

negativekelvin, 2020-10-12:
> we need to add the reason code IE , so we change the reason to WIFI_REASON_ASSOC_LEAVE(8), thus the receiver can parse it correctly.

@liuzfesp ok to send a valid reason code to the AP but why change it within the callback data?

zhangyanjiaoesp, 2020-10-13:
> > we need to add the reason code IE , so we change the reason to WIFI_REASON_ASSOC_LEAVE(8), thus the receiver can parse it correctly.
> 
> @zhangyanjiaoesp ok to send a valid reason code to the AP but why change it within the callback data?

Maybe we will optimize it in the future.

zhangyanjiaoesp, 2020-10-13:
> Hi @liuzfesp
> Unfortunately I can't take care of the problem this week because I'm on vacation.
>
> Our customer has already tested it with a different AP (e.g. by a hotspot on a smartphone, AP at home [...]
If user reset the router, the TSF timer will be reset.
> How is that meant by "capture the WiFi packets"? Extend the debug log on the ESP or record the packets via a sniffer on the AP?
You can use an software (Omnipeek or Wireshark) to capture the wireless packets.

DonSimonus, 2020-10-19:
> If user reset the router, the TSF timer will be reset.

The user do not reset the router(s) manually (I was there during the tests). Are there other influences that forces the AP to reset by itself (or just the TSF timer)?

> 
> > How is that meant by "capture the WiFi packets"? Extend the debug log on the ESP or record the packets via a sniffer on the AP?
> 
> You can use an software (Omnipeek or Wireshark) to capture the wireless packets.

So do you recommend sniffing (e.g. with Wireshark) directly on the router via LAN connection? Wouldn't it be a possibility to enable the IP debugging/logging (#define IP_DEBUG LWIP_DBG_OFN) on the ESP in order to visualize the packets?

negativekelvin, 2020-10-19:
@DonSimonus you need to sniff the raw packets directly off the air using a WiFi adapter that supports promiscuous.  There are a few projects that use esp32 as a sniffer.

negativekelvin, 2020-10-20:
https://github.com/espressif/esp32-wifi-lib/commit/435347a24cec805f81319d3ac6d2a2f17da57bd5

DonSimonus, 2020-10-23:
> @DonSimonus you need to sniff the raw packets directly off the air using a WiFi adapter that supports promiscuous. There are a few projects that use esp32 as a sniffer including https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/examples/wifi/simple_sniffer/README.md

@negativekelvin Thanks for your inputs. 
Do you know if it's a "normal" behaviour of WiFi clients to disconnect from APs if the TFS timer resets?
Are there other influences that forces the AP to reset by itself (or just the TSF timer)?

negativekelvin, 2020-10-23:
Tsf timer should not reset, it may be ok to disconnect if it actually reset to zero. I don't think we know whether it did or if an out of order timestamp was received due to mesh/repeater properties and the check was too strict.

AxelLin, 2020-10-26:
Hi @negativekelvin 

Is it ok to capture the log with wireshark?
If yes, what is the recommend filter setting for only showing TSF related packets?

DonSimonus, 2020-10-28:
> Hi @negativekelvin
> 
> Is it ok to capture the log with wireshark?
> If yes, what is the recommend filter setting for only showing TSF related packets?

I have the same question. Since in my case its a network in a company, there is a lot of traffic. Does it make sense to filter for the MAC address of the Espressif module or are there other "characteristics" of the regarding packages?

AxelLin, 2020-10-29:
> [espressif/esp32-wifi-lib@435347a](https://github.com/espressif/esp32-wifi-lib/commit/435347a24cec805f81319d3ac6d2a2f17da57bd5)

I still observed the same symptom with esp32-wifi-lib update.
So it's not clear to me if it's AP malfunction or something wong in ESP32 side.
Is it possible to show some debug info regarding received tsf timer value if enable verbose or debug log?

DonSimonus, 2020-10-29:
> [espressif/esp32-wifi-lib@435347a](https://github.com/espressif/esp32-wifi-lib/commit/435347a24cec805f81319d3ac6d2a2f17da57bd5)

Hi @negativekelvin 
Because I work on the 4.0 branch (v4.0.1-412-g820621687), I cannot try out the changes in my project without adapting the IDF version, which would result in further necessary changes in my project.
Is it therefore possible to implement the adaptation on the 4.0 branch as well?

zhangyanjiaoesp, 2020-10-29:
@DonSimonus 

The changes has been backported to v4.0, please wait for the latest release.

zhangyanjiaoesp, 2020-10-29:
> Hi @negativekelvin
> 
> Is it ok to capture the log with wireshark?
> If yes, what is the recommend filter setting for only showing TSF related packets?

You can capture wifi packets with wireshark, you can set the STA and AP's MAC address as the filter.

DonSimonus, 2020-10-29:
> @DonSimonus
> 
> The changes has been backported to v4.0, please wait for the latest release.

@zhangyanjiaoesp 
Many thanks. When do you think this will be integrated into the IDF and the release is available?

closed 2020-11-03 by espressif-bot

zhangyanjiaoesp, 2020-11-09:
@DonSimonus 

The v4.0.2 ( e5f754b) has been released, and the related change is db733d7321d35c21c6e0d055c939c0568b325cd9

mbratch, 2021-07-17:
I'm using ESP-IDF v4.3 and I get a wifi disconnect once or twice per day. I have 5 devices running the same code and they all get this disconnect at exactly the same time. They each take 40 seconds of retries and finally reconnect again.

My reason code is different than reported above. I'm seeing an "init (600)" which I assume is a reason code 6: WIFI_REASON_NOT_AUTHED. The devices all authenticate just fine. I'm not sure why this disconnect would occur.

AxelLin, 2021-07-18:
> My reason code is different than reported above. I'm seeing an "init (600)" which I assume is a reason code 6: WIFI_REASON_NOT_AUTHED according to the header file. The devices all authenticate just fine. I'm not sure why this disconnect would occur.

Hi @mbratch
Please create a new issue to make it easier to trace the issue.

mbratch, 2021-07-19:
> Hi @mbratch
> Please create a new issue to make it easier to trace the issue.

Thank you. I have submitted issue #7286.