espressif/esp-idf #5917: Https/wss not limiting max open sockets

baldhead69, 2020-09-26:
Hi,

I am implementing a websocket secure server on esp32 using the code below.

I am setting the max open sockets to 5, but when the number of open sockets already are 5, and i try to open a new socket,        the lib closes one socket that was open and open a new socket.

The lib limits to 5 open sockets, but closes one socket and opens another.

When i used unsecured websocket, the lib did not accepted to open new sockets. 

````
    httpd_ssl_config_t config = HTTPD_SSL_CONFIG_DEFAULT();
    config.httpd.max_open_sockets = 5;
```

negativekelvin, 2020-09-27:
https://github.com/espressif/esp-idf/blob/8bc19ba893e5544d571a753d82b44a84799b94b1/components/esp_http_server/include/esp_http_server.h#L162

https://github.com/espressif/esp-idf/blob/c9f29e0b59988779bf0792e2e954c3448fd37e3b/components/esp_https_server/include/esp_https_server.h#L100

baldhead69, 2020-09-27:
Thank's @negativekelvin.

Worked.

closed 2020-09-27 by baldhead69