espressif/esp-idf #5921: Passing void* arg to esp_now_register_recv_cb which can be used from C++ to pass this pointer

neel, 2020-09-28:
**Is your feature request related to a problem? Please describe.**

`esp_now_register_recv_cb` receives a callback which have have the following signature `(const uint8_t *mac_addr, const uint8_t *data, int data_len)`. With such a callback it is very difficult to use it from C++ classes. It forces global storage and static methods. I'd like to avoid static/global/singleton whenever possible

**Describe the solution you'd like**

I'd like to have the following signatures

    typedef void (*esp_now_recv_cb_t)(const uint8_t *mac_addr, const uint8_t *data, int data_len, void* arg);

The receive callback will have an additional `void*` argument

    esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t cb, void* arg = 0x0);

`esp_now_register_recv_cb` will take the argument and store that internally. This arg will be passed to the receive callback.

**Describe alternatives you've considered**

The current alternative is to have singleton classes 

**Additional context**

N/A

zhangyanjiaoesp, 2020-10-19:
hi @neel , I think we won't change the `esp_now_register_recv_cb()` API currently.

Alvin1Zhang, 2020-10-27:
Thanks for raising this feature request, feel free to reopen.

closed 2020-10-27 by Alvin1Zhang

bLind17, 2023-07-15:
I am facing the same issue and support this request!