espressif/esp-idf #5923: Nimble BLE Cached access crash

Francois-Belanger, 2020-09-28:
## Environment

- Development Kit:      [none] Custom board
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-1197-g8bc19ba89
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [macOS]
- Using an IDE?: [No]
- Power Supply:         [Battery]

## Problem Description

If I enable BLE Nimble and access the Flash a lot, I get this crash pretty quickly.

Without the BLE enabled, everything runs fine.  When Enable the BLE is only advertising for now, no connection.

## Code for BLE

Init:

`    ESP_ERROR_CHECK(esp_nimble_hci_and_controller_init());

    nimble_port_init();
    // Initialize the NimBLE host configuration
    ble_hs_cfg.sync_cb = ble_on_sync;
    ble_hs_cfg.reset_cb = ble_on_reset;
    ble_hs_cfg.gatts_register_cb = gatt_svr_register_cb;
    int rc = gatt_svr_init();
    assert(rc == 0);

    rc = ble_svc_gap_device_name_set(device_name.c_str());  // Set the default device name
    assert(rc == 0);

    ble_att_set_preferred_mtu(256);

    // Start the task
    nimble_port_freertos_init(ble_host_task);
`

The rest that is important:

`void BLETASK::ble_advertise(void) {
    struct ble_gap_adv_params adv_params;
    struct ble_hs_adv_fields fields;
    int rc;

    if (bleAdvertising) {
        return;
    }
    memset(&fields, 0, sizeof(fields));
    fields.flags = BLE_HS_ADV_F_DISC_GEN | BLE_HS_ADV_F_BREDR_UNSUP;
    fields.tx_pwr_lvl_is_present = 1;
    fields.tx_pwr_lvl = BLE_HS_ADV_TX_PWR_LVL_AUTO;

    fields.name = (uint8_t *)device_name.c_str();
    fields.name_len = device_name.size();
    fields.name_is_complete = 1;

    rc = ble_gap_adv_set_fields(&fields);
    if (rc != 0) {
        ESP_LOGE(TAG, "error setting advertisement data; rc=%d", rc);
        return;
    }

    /* Begin advertising */
    memset(&adv_params, 0, sizeof(adv_params));
    adv_params.conn_mode = BLE_GAP_CONN_MODE_UND;
    adv_params.disc_mode = BLE_GAP_DISC_MODE_GEN;
    rc = ble_gap_adv_start(ble_addr_type, NULL, BLE_HS_FOREVER, &adv_params, ble_gap_event, NULL);
[... 244 lines cut ...]
0x4008bf25: esp_flash_read at /esp-idf/components/spi_flash/esp_flash_api.c:576

0x40123621: esp_partition_read at /esp-idf/components/spi_flash/partition.c:347

0x400850dd: flash_read at /build/../main/esp_littlefs.c:51

0x4011eccd: lfs_bd_read at /build/../components/littlefs-master/lfs.c:84
 (inlined by) lfs_bd_read at /build/../components/littlefs-master/lfs.c:27

0x4011faf7: lfs_ctz_find at /build/../components/littlefs-master/lfs.c:2161

0x40120ca2: lfs_file_read at /build/../components/littlefs-master/lfs.c:2774

0x400ddb71: esp_lfs_file_read at /build/../main/esp_littlefs.c:257

0x400dcc66: AUDIOTASK::updateSeq(AUDIOSEQ&) at /build/../main/audio_Task.cpp:216

0x400dce7e: AUDIOTASK::tick() at /build/../main/audio_Task.cpp:291 (discriminator 2)

0x400dd041: AUDIOTASK::run(void*) at /build/../main/audio_Task.cpp:45

0x400db05a: Task::runTask(void*) at /build/../main/Task.cpp:60

```
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/5295568/sdkconfig.zip)

prasad-alatkar, 2020-09-29:
Hi @Francois-Belanger Can you please take a look at similar issues https://github.com/espressif/esp-idf/issues/1299 and https://github.com/espressif/arduino-esp32/issues/855 for solution? Please let me know if you still face any issue.

Francois-Belanger, 2020-09-29:
Hi @prasad-alatkar I don't use any interrupt in my code.  Does the Nimble use any on the client code ?  In the Nimble Example I didn't see anything with IRAM so I guess all callback are not in interrupt ?  Thanks

Francois-Belanger, 2020-09-29:
Hummm issue #5924 or #5834 ?  Looks similar.

Francois-Belanger, 2020-10-05:
Ok, if I disable  Software controls WiFi/Bluetooth coexistence everything looks stable.  @prasad-alatkar can you help please.  Looks like task priority ?

Thanks

prasad-alatkar, 2020-10-06:
@Francois-Belanger, Can you please try with disabling `SPI_MASTER_ISR_IN_IRAM` (`make menuconfig → Component config → Driver configurations → SPI configuration`) from menuconfig once ? I could see from the sdkconfig `CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL` is enabled in your case and following up few different threads related to the similar issues looks like disabling `SPI_MASTER_ISR_IN_IRAM` may help you here (Reference: https://www.esp32.com/viewtopic.php?t=7684 ).  

>if I disable Software controls WiFi/Bluetooth coexistence everything looks stable

I will look into it and update/explain you accordingly. Meanwhile please let me know if the above suggestion helps you overcome the issue.

Francois-Belanger, 2020-10-06:
Hi @prasad-alatkar I've tried disabling SPI_MASTER_ISR_IN_IRAM, no improvements, still crash with coex in software :-(  What is the penalty in speed to use the hardware coex ?

Thanks

prasad-alatkar, 2020-10-07:
As per my understanding, with hardware coex control BLE performance won't be affected at all, there is chance of little degradation in wifi performance but given that your BLE application is only advertising, I believe it should give enough window for wifi operations. You can refer detailed explanation on this topic here: [WIFI/BLE Simultaneously](https://esp32.com/viewtopic.php?t=6707&start=30) and here: [ESP32 FAQ](https://www.espressif.com/sites/default/files/documentation/ESP32_FAQs__EN.pdf) (Refer 5.3.2)

To narrow down the issue further, Can you please try to enable `CONFIG_NIMBLE_MEM_ALLOC_MODE_INTERNAL` (and disable `CONFIG_NIMBLE_MEM_ALLOC_MODE_EXTERNAL`) and see if you still observe the issue ? Please share the results.

Francois-Belanger, 2020-10-08:
Hi @prasad-alatkar putting Nimble Internal Memory and putting Coex Soft crash again. :-(

blueMoodBHD, 2024-09-09:
@Francois-Belanger, 
Sorry for the late reply. Does this issue still exist? Please update the IDF and try again.

closed 2024-09-23 by Alvin1Zhang