espressif/esp-idf #5924: Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)

SiyiCode, 2020-09-29:
I build a project,I referred wear_leveling example and gatt_server example; I write some data into a file per 5s,when my android device send data via BLE,then crashed.I check some similar issue and I set all interrupt handle have the IRAM_ATTR attribute，but crash again.the traceback show error at "fclose" or"fopen",how do I fix it?

## Debug Logs

Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x401274fc  PS      : 0x00060d34  A0      : 0x80044715  A1      : 0x3ffbe200
0x401274fc: chnl_map_check at ??:?

A2      : 0x00000000  A3      : 0x00000006  A4      : 0x00000001  A5      : 0x3ffb2fa6
A6      : 0x00002b66  A7      : 0x00000000  A8      : 0x800866ea  A9      : 0x3ffbe1e0
A10     : 0x3ffb2fa7  A11     : 0x0000000a  A12     : 0x00060121  A13     : 0x00000021
A14     : 0x00000050  A15     : 0x3ffbe970  SAR     : 0x00000016  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
Core 0 was running in ISR context:
EPC1    : 0x4008f483  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x401274fc
0x4008f483: portENTER_CRITICAL_NESTED at d:\apps\espressif\esp-idf\syzn\siyi_esp32\sy_c11_1.0.0\build/../../../../components/freertos/include/freertos/portmacro.h:330
 (inlined by) xPortInIsrContext at D:/Apps/espressif/esp-idf/components/freertos/port.c:338

0x401274fc: chnl_map_check at ??:?

ELF file SHA256: 8d090f09e683108d

Backtrace: 0x401274f9:0x3ffbe200 0x40044712:0x3ffbe220 0x400881c7:0x3ffbe250 0x40086ff5:0x3ffbe280 0x40086d89:0x3ffbe2a0 0x40088a8e:0x3ffbe2c0 0x400899b7:0x3ffbe2e0 0x40082d2d:0x3ffbe300 0x400951a7:0x3ffd2590 0x40095052:0x3ffd25b0 0x40095485:0x3ffd25f0 0x400957bf:0x3ffd2610 0x4008b0a8:0x3ffd2630 0x4012d141:0x3ffd2680 0x400dc652:0x3ffd26a0 0x400dd79d:0x3ffd26c0 0x400dc54a:0x3ffd2700 0x400d8d91:0x3ffd2720 0x400d8c45:0x3ffd2740 0x400d90ae:0x3ffd2760 0x400d9d7d:0x3ffd2780 0x400da5d1:0x3ffd27a0 0x400da5f9:0x3ffd27d0 0x400db7ff:0x3ffd2800 0x400d4c95:0x3ffd2820 0x4000bd4b:0x3ffd2840 0x400011c1:0x3ffd2860 0x40002055:0x3ffd2880 0x400020b5:0x3ffd28a0 0x40083396:0x3ffd28c0 0x400d6a37:0x3ffd28e0 0x4008f389:0x3ffd2900
0x401274f9: llm_update_duplicate_scan_exceptional_list at ??:?

0x400881c7: r_lld_pdu_rx_handler at ??:?

0x40086ff5: r_lld_evt_end at ??:?

0x40086d89: r_lld_evt_end_isr at ??:?

0x40088a8e: r_rwble_isr at ??:?

0x400899b7: r_rwbtdm_isr_wrapper at intc.c:?

0x40082d2d: _xt_lowint1 at D:/Apps/espressif/esp-idf/components/freertos/xtensa_vectors.S:1153

0x400951a7: spi_flash_ll_cmd_is_done at d:\apps\espressif\esp-idf\syzn\siyi_esp32\sy_c11_1.0.0\build/../../../../components/soc/esp32/include/hal/spi_flash_ll.h:71 (discriminator 1)
 (inlined by) spi_flash_hal_poll_cmd_done at D:/Apps/espressif/esp-idf/components/soc/src/hal/spi_flash_hal_iram.c:29 (discriminator 1)

0x40095052: spi_flash_hal_program_page at D:/Apps/espressif/esp-idf/components/soc/src/hal/spi_flash_hal_iram.c:128

0x40095485: spi_flash_chip_generic_page_program at D:/Apps/espressif/esp-idf/components/spi_flash/spi_flash_chip_generic.c:168

0x400957bf: spi_flash_chip_generic_write at D:/Apps/espressif/esp-idf/components/spi_flash/spi_flash_chip_generic.c:190

0x4008b0a8: esp_flash_write at D:/Apps/espressif/esp-idf/components/spi_flash/esp_flash_api.c:617

0x4012d141: esp_partition_write at D:/Apps/espressif/esp-idf/components/spi_flash/partition.c:389

0x400dc652: Partition::write(unsigned int, void const*, unsigned int) at D:/Apps/espressif/esp-idf/components/wear_levelling/Partition.cpp:49

0x400dd79d: WL_Flash::write(unsigned int, void const*, unsigned int) at D:/Apps/espressif/esp-idf/components/wear_levelling/WL_Flash.cpp:591

0x400dc54a: wl_write at D:/Apps/espressif/esp-idf/components/wear_levelling/wear_levelling.cpp:211

0x400d8d91: ff_wl_write at D:/Apps/espressif/esp-idf/components/fatfs/diskio/diskio_wl.c:63

0x400d8c45: ff_disk_write at D:/Apps/espressif/esp-idf/components/fatfs/diskio/diskio.c:74

0x400d90ae: sync_window at D:/Apps/espressif/esp-idf/components/fatfs/src/ff.c:1050

0x400d9d7d: sync_fs at D:/Apps/espressif/esp-idf/components/fatfs/src/ff.c:1102

0x400da5d1: f_sync at D:/Apps/espressif/esp-idf/components/fatfs/src/ff.c:4020

0x400da5f9: f_close at D:/Apps/espressif/esp-idf/components/fatfs/src/ff.c:4047

0x400db7ff: vfs_fat_close at D:/Apps/espressif/esp-idf/components/fatfs/vfs/vfs_fat.c:466
[... 83 lines cut ...]
I (630) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2
W (640) cpu_wakeup_src: Time spent in deep sleep: 647444ms
W (646) cpu_wakeup_src: wake up from system reset
I (680) ADC Task: eFuse Vref: NOT supported
I (690) Operation Task: restore mode:0, close level:0, open level:0
I (690) Operation Task: restore boot counts:0, motor time:0, machine time:0 overheat tick:0
I (695) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (704) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (714) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (734) BTDM_INIT: BT controller compile version [b1eb377]
I (735) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (824) phy: phy_version: 4370, 4e803b3, Aug 11 2020, 14:18:07, 0, 0
I (1091) C11_GATTS: REGISTER_APP_EVT, status 0, app_id 0

W (1093) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
W (1098) BT_BTM: BTM_BleWriteScanRsp, Partial data write into ADV
I (1104) C11_GATTS: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (1122) C11_GATTS: SERVICE_START_EVT, status 0, service_handle 40

I (1123) C11_GATTS: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

I (1127) C11_GATTS: the gatts demo char length = 5:SiYi

I (1133) C11_GATTS: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40

SiyiCode, 2020-09-29:
code referred in the log:
[image]
[image]

SiyiCode, 2020-09-29:
code referred in debug log"0x40083396: preserve_cfg_info_to_file at d:\apps\espressif\esp-idf\syzn\siyi_esp32\sy_c11_1.0.0\build/../main/src/app_operation_task.c:118"
`if(xPortInIsrContext() == 0 )
        {
            FILE *f = NULL;
            f = fopen(CFG_FILE_NAME_STR, "wb");
            if(f != NULL)
            {
                fwrite(&sy_c11_device.cfg_info.machine_info, 1, sizeof(cfg_info.machine_info), f);
                fclose(f);
            }
            else
            {
                ESP_LOGI(log_tag, "open file to write failed!");
            }
        }`

Alvin1Zhang, 2020-10-12:
Thanks for reporting, we will look into.

tomoyuki-nakabayashi, 2020-11-24:
We have a similar problem on ESP-IDF v3.3.4.

```
0x00000bac: ?? ??:0
0x40139eac: llm_update_duplicate_scan_exceptional_list at ??:?
0x40044712: ?? ??:0
0x4008b5c3: r_lld_pdu_rx_handler at ??:?
0x4008a3f5: r_lld_evt_end at ??:?
0x40089ca1: r_lld_evt_end_isr at ??:?
0x4008bd6e: r_rwble_isr at ??:?
0x4008cc8b: r_rwbtdm_isr_wrapper at intc.c:?
0x400857ad: _xt_lowint1 at /Users/nature/src/github.com/natureglobal/esp-idf/components/freertos/xtensa_vectors.S:1154
0x400862d9: spi_flash_op_block_func at /Users/nature/src/github.com/natureglobal/esp-idf/components/spi_flash/cache_utils.c:201
0x40082089: ipc_task at /Users/nature/src/github.com/natureglobal/esp-idf/components/esp32/ipc.c:62
```

On v3.3.3, we didn't see the problem.
FYI: We bisected between v3.3.3 and v3.3.4, and found the commit (https://github.com/espressif/esp-idf/commit/d674016d6483d10ee1253a7a505f3674ae0865ec) perhaps incurs the problem.

thanks,

tomoyuki-nakabayashi, 2020-11-24:
I found a commit which will solve this problem!
I hope this will be released as soon as possible.

espressif/esp32-bt-lib@5ed616a

phatpaul, 2021-03-01:
I am getting a similar crash when reading a file over BLE.  I'm using IDF 3.3.4 with GATTS reading/writing to FAT fs.  (my code is based on https://github.com/eagi223/esp-idf_Bluetooth_Multi-Service)

Does my crash look related to this bug?  How do I know when the fix is applied to the 3.3 branch?  

```
I (4028559) cgi_cfg_put: Uploading: /internalfs/html/cfg/confc.json~06
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x4017a3b0  PS      : 0x00060534  A0      : 0x80044715  A1      : 0x3ffc0040
0x4017a3b0: chnl_map_check at ??:?

A2      : 0x00000000  A3      : 0x00000006  A4      : 0x00000001  A5      : 0x3ffb2bca
A6      : 0x00000b72  A7      : 0x00000000  A8      : 0x8008af12  A9      : 0x3ffc0020
A10     : 0x3ffb2bcb  A11     : 0x0000000a  A12     : 0x400879d9  A13     : 0x00000021
0x400879d9: spi_flash_op_block_func at C:/msys32/home/pabbott/lumitec-dev-fw-plibox-esp32/Tools
/esp-idf/components/spi_flash/cache_utils.c:201

A14     : 0x00000064  A15     : 0x3ffc11a0  SAR     : 0x00000020  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
Core 0 was running in ISR context:
EPC1    : 0x400937b3  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4017a3b0
0x400937b3: vPortCPUReleaseMutexIntsDisabledExtram at C:/msys32/home/pabbott/lumitec-dev-fw-pli
box-esp32/Tools/esp-idf/components/freertos/tasks.c:3507

0x4017a3b0: chnl_map_check at ??:?

ELF file SHA256: d9bd4d647935ac51

Backtrace: 0x4017a3b0:0x3ffc0040 0x40044712:0x3ffc0060 0x4008ca23:0x3ffc0090 0x400485a5:0x3ffc0
0c0 0x4004868d:0x3ffc00e0 0x4008d14a:0x3ffc0100 0x4008e0eb:0x3ffc0120 0x40086d9d:0x3ffc0140 0x4
00879d6:0x3ffba190 0x40082133:0x3ffba1b0
0x4017a3b0: chnl_map_check at ??:?

0x4008ca23: r_lld_pdu_rx_handler at ??:?

0x4008d14a: r_rwble_isr at ??:?

0x4008e0eb: r_rwbtdm_isr_wrapper at intc.c:?

0x40086d9d: _xt_lowint1 at C:/msys32/home/pabbott/lumitec-dev-fw-plibox-esp32/Tools/esp-idf/com
ponents/freertos/xtensa_vectors.S:1154

0x400879d6: spi_flash_op_block_func at C:/msys32/home/pabbott/lumitec-dev-fw-plibox-esp32/Tools
/esp-idf/components/spi_flash/cache_utils.c:201

0x40082133: ipc_task at C:/msys32/home/pabbott/lumitec-dev-fw-plibox-esp32/Tools/esp-idf/compon
ents/esp32/ipc.c:62

Rebooting...
ets Jul 29 2019 12:21:46
```

Weijian-Espressif, 2022-08-30:
We have fixed the issue on v4.0 and later @lingfengPeng @tomoyuki-nakabayashi  @phatpaul @AxelLin 

Please update to the latest branch code.

AxelLin, 2022-08-30:
@Weijian-Espressif 
Why this issue status is still Open if it's already fixed?

Weijian-Espressif, 2022-08-30:
There has been no reply from other colleagues, so it cannot be closed

martin96richard, 2022-12-02:
This error is still present in version 4.3. It often happens during an OTA. It seems that the OTA is hogging the flash and the Bluetooth is struggling to access it :

`[INFO]    	44327	<WIFI HTTPS OTA>	HTTP Client OTA received   729088 bytes (41%)!
[INFO]    	44926	<WIFI HTTPS OTA>	HTTP Client OTA received   745472 bytes (42%)!
[INFO]    	45516	<WIFI HTTPS OTA>	HTTP Client OTA received   765952 bytes (43%)!
[INFO]    	46082	<WIFI HTTPS OTA>	HTTP Client OTA received   782336 bytes (44%)!
[INFO]    	46550	<WIFI HTTPS OTA>	HTTP Client OTA received   798720 bytes (45%)!
[INFO]    	47237	<WIFI HTTPS OTA>	HTTP Client OTA received   819200 bytes (46%)!
[INFO]    	47752	<WIFI HTTPS OTA>	HTTP Client OTA received   835584 bytes (47%)!
[INFO]    	48265	<WIFI HTTPS OTA>	HTTP Client OTA received   851968 bytes (48%)!
[INFO]    	48942	<WIFI HTTPS OTA>	HTTP Client OTA received   872448 bytes (49%)!
[INFO]    	49410	<WIFI HTTPS OTA>	HTTP Client OTA received   888832 bytes (50%)!
[WARNING] 	49809	<BQ769x2>	Command 0x76: checksum mismatch: read 0xe7, computed 0xdf
[INFO]    	49924	<WIFI HTTPS OTA>	HTTP Client OTA received   905216 bytes (51%)!
[INFO]    	50564	<WIFI HTTPS OTA>	HTTP Client OTA received   925696 bytes (52%)!
[INFO]    	51072	<WIFI HTTPS OTA>	HTTP Client OTA received   942080 bytes (53%)!
[INFO]    	51548	<WIFI HTTPS OTA>	HTTP Client OTA received   958464 bytes (54%)!
Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x40145000  PS      : 0x00060d34  A0      : 0x80046686  A1      : 0x3ffbe700  
0x40145000: r_platform_reset at ??:?

A2      : 0x00000000  A3      : 0x40145000  A4      : 0x00000000  A5      : 0x0000000c  
0x40145000: r_platform_reset at ??:?

A6      : 0x3ffb93cc  A7      : 0x3ffb8360  A8      : 0x80019fb8  A9      : 0x00001c68  
A10     : 0xf2f2f2f2  A11     : 0x00000000  A12     : 0x3ffc4a08  A13     : 0x00000022  
A14     : 0x00001400  A15     : 0x00000001  SAR     : 0x00000017  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x40144ffd:0x3ffbe700 0x40046683:0x3ffbe730 0x40047515:0x3ffbe750 0x40087731:0x3ffbe770 0x40087385:0x3ffbe790 0x40089aa6:0x3ffbe7b0 0x4008ad23:0x3ffbe7d0 0x40083cb9:0x3ffbe7f0 0x4008b148:0x3ffb7180 0x4008b04c:0x3ffb71a0
0x40144ffd: r_get_stack_usage at ??:?

0x40087731: r_lld_evt_end at ??:?

0x40087385: r_lld_evt_end_isr at ??:?

0x40089aa6: r_rwble_isr at ??:?

0x4008ad23: r_rwbtdm_isr_wrapper at intc.c:?

0x40083cb9: _xt_lowint1 at /home/RICHAR_M/gouach/bms-v3-code-dup/tools/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4008b148: spi_flash_op_block_func at /home/RICHAR_M/gouach/bms-v3-code-dup/tools/esp-idf/components/spi_flash/cache_utils.c:115 (discriminator 1)

0x4008b04c: ipc_task at /home/RICHAR_M/gouach/bms-v3-code-dup/tools/esp-idf/components/esp_ipc/ipc.c:62`

Weijian-Espressif, 2022-12-05:
Hi ALL, Please use the latest code of each release version for testing. We cannot directly backport the code for other versions. 
If you have problems with the latest release version, please create a new issue, we will deal with it as soon as possible

AxelLin, 2022-12-09:
> This error is still present in version 4.3. It often happens during an OTA. It seems that the OTA is hogging the flash and the Bluetooth is struggling to access it :

If you are using v4.3 branch, please check if latest v4.3.4 release still has the issue.
Or pull the latest v4.3 branch to verify. (Please include "git describe" output in your bug report)

closed 2023-08-10 by Alvin1Zhang