espressif/esp-idf #5925: Install IDF, python report "Can not perform a '--user' install. User site-packages are not visible in this virtualenv."

kylongmu, 2020-09-29:
This message is shows when I try to install IDF, or build a project. Maybe it's caused by new python update or Win10 update some security rules.

```
Some Python dependencies must be installed.  Check above message for details.
Can not perform a '--user' install. User site-packages are not visible in this virtualenv.
```

After a lot of test, I found it's caused by a python option: in this file
idf_tools.py
`line 1185 subprocess.check_call([sys.executable, '-m', 'pip', 'install', '--user', 'virtualenv']`

Remove --user option
Then redo install and export under IDF folder
build success!

igrr, 2020-09-29:
Hi @kylongmu, when you are running the install script, do you have a Python virtual environment active?

You can check by running `python -c 'import sys; print(sys.executable)'` — this should indicate the location where the interpreter is running from.

Since the IDF Tools installation creates a virtual environment itself, it can not be started from another virtual environment. This may be possible to support, to allow installing IDF packages into an existing environment... however I am not sure what is the exact use case.

kylongmu, 2020-09-30:
Hi @igrr Thanks for your advise, I'm just open a CMD to excute the install, without any other operations.
Follow your suggestion, I got this:
```
D:\Develop\Git_Project>python -c 'import sys; print(sys.executable)'
  File "<string>", line 1
    'import
          ^
SyntaxError: EOL while scanning string literal

```

igrr, 2020-10-02:
Could you please paste the complete output you get when running install.bat?

kylongmu, 2020-10-03:
@igrr I've already installed all the environments, to show the complete output again need to delete all the things, I'm on my project now,  I'm not want to redo it.

tom-borcin, 2021-03-26:
Hello @kylongmu, we are closing this issue. Check for active virtual environments has already been added.

If you think this issue should remain open, please let me know. Thank you for your contribution!

closed 2021-03-26 by tom-borcin

pequalsnp, 2022-05-11:
Hey @igrr and @tom-borcin I am running into the exact same problem.  I am running the using ESP_IDF in vscode.  Here is the output I get when trying to "Install ESP-IDF extension Python packages":
```
ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.
Command failed: "/opt/esp/python_env/idf4.3_py3.6_env/bin/python" -m pip install --user virtualenv
ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.
```

Here is the output of the command you suggested:
```
root@4517c04c6b10:/workspaces/bathtubgin/esp32/temp-controller# python -c 'import sys; print(sys.executable)'
/opt/esp/python_env/idf4.3_py3.6_env/bin/python
```

so that looks like the correct environment.

igrr, 2022-05-11:
@pequalsnp Could you please report this at https://github.com/espressif/vscode-esp-idf-extension/issues/? I think the problem is that `pip install --user virtualenv` should only be executed if the virtual environment isn't yet created. In your case the virtual environment is already created and activated, so it shouldn't be necessary to `pip install --user virtualenv`.

mpapec, 2022-05-27:
```
Command failed: "C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe" -m pip install --user virtualenv
ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.
```

The same issue on linux/Fedora and vscode. 
Is there a python expert who could solve this virtualenv issue?

It seems Espriff struggles for years to solve this particular problem
https://www.esp32.com/viewforum.php?f=40

xobs, 2023-04-18:
This seems to happen when performing an install for the first time, at least with the latest esp-idf:

[image]

xobs, 2023-04-18:
This happens when installing both from the command line and from vscode.