espressif/esp-idf #5926: Warning when use esptool.py to write a pre-encrypted bootloader to flash

AxelLin, 2020-09-29:
## Environment

- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.3-dev-1197-g8bc19ba893e5
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

When I use esptool.py to write a pre-encrypted bootloader, the esptool.py shows
"Warning: Image file at 0x1000 doesn't look like an image file, so not changing any flash settings.".
Such warning looks scary, I'm not sure if it's a bug or something I missed. 
(And it does not suggest how to fix/silence the warning)
but it looks like it does write the image to flash and my device can boot.

Note, it does not show the warning if I write a pre-encrypted  application firmware.

Below is the command to write pre-encrypted bootloader:
$ esptool.py --chip esp32 -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 bootloader-enc.bin
esptool.py v3.0-dev
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting......
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: f0:08:d1:79:8c:60
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Image file at 0x1000 doesn't look like an image file, so not changing any flash settings.
Compressed 23520 bytes to 23536...
Wrote 23520 bytes (23536 compressed) at 0x00001000 in 0.3 seconds (effective 684.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

### Expected Behavior
The warning message is not expected.

### Actual Behavior
Got below warning:
"Warning: Image file at 0x1000 doesn't look like an image file, so not changing any flash settings."

### Steps to reproduce
Just use esptool.py to flash a pre-encrypted bootloader image. (as the command mentioned above).

projectgus, 2020-10-05:
Hi @AxelLin ,

The reason for this warning is that esptool.py has the capability to update the header of a binary image at the time of flashing. The arguments that control this are:

```
--flash_mode dio --flash_freq 40m --flash_size 4MB
```

If the settings in a plaintext bootloader .bin file are different to this, they'll get updated as the binary is written.

In the case of an encrypted image, esptool.py can't recognise the .bin file is an image and also can't change anything in the binary file safely - so it prints a warning that it can't necessarily do everything which has been asked of it.

To not print the warning now, you can change the `write_flash` arguments as follows:

```
esptool.py --chip esp32 -b 921600 --before=default_reset --after=hard_reset write_flash ----flash_size keep 0x1000 bootloader-enc.bin
```

(Specifically: the --flash_mode & --flash_freq args have been dropped entirely, `--flash_size` currently defaults to detecting the image size so we have to tell it to "keep" the existing size instead.)

> Expected Behavior
>
> The warning message is not expected.

We have a plan to update the IDF build system so it no longer prompts to pass any of these arguments to `esptool.py write_flash`. Once this change is made, the warning won't be printed when following the default IDF flashing steps.

Will update this issue once that is the case.

closed 2022-11-25 by dobairoland

brightproject, 2023-07-04:
I use arduino-cli  for compile and load program into ESP32-wroom
My log in cli
```
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0011dfff...
Warning: Image file at 0x1000 is protected with a hash checksum, so not changing the flash frequency setting. Use the --flash_frequency=keep option instead of --flash_frequency=80m in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 17488 bytes to 12168...
```