espressif/esp-idf #5927: Missing patch in long term branch

swdevelops, 2020-09-29:
There is a patch in closed issue IDFGH-781 which has been applied to 4.x branches, will be the same solution a part of the long term branch 3.3.x?

Thanks

mahavirj, 2020-10-06:
@swdevelops If you are talking about fix in commit-id https://github.com/espressif/esp-idf/commit/ab6ad844b152594d7df7d4e23df3ec8397acc176#diff-e5ab27131c9215b651df791f52f73d91, then its not applicable for v3.x branch. This issue appeared because of lwIP upgrade to v2.1.0 release, v3.x still runs on older version of lwIP v2.0.3.

Are you observing any specific issue with v3.3.x release? If yes, can you please help to share detailed information per github issue template?

mahavirj, 2020-10-14:
@swdevelops Closing this issue, please feel free to reopen in case your query is not answered or you have any specific issue on v3.3.x release (would recommend new issue in that case).

closed 2020-10-14 by mahavirj