espressif/esp-idf #5931: Esp32 Reboot when GSM Module is connecting to network

dhirajp15, 2020-10-01:
Hi ,

I am facing the exact same issue. Using GSM module and esp32 together , when GSM module tries to connect to Network the ESP32 reboots with following logs:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6336
ho 0 tail 12 room 4
load:0x40078000,len:11308
load:0x40080400,len:6680
entry 0x40080760

**Note:** This voltage fluctuation is observed at EN pin , while the VCC is does not show any fluctuation.( I assume it is RF noise) .
Also when I keep GSM antenna away from PCB , it does not reset.

[image]

Using two **47uF**  resistors at EN pin to suppress the noise : 

[image]

Please suggest if there is any better way to fix this issue

_Originally posted by @dhirajp15 in https://github.com/espressif/arduino-esp32/issues/3520#issuecomment-701896350_

nuclearcat, 2020-10-02:
IMO, it is particular PCB layout issue. I have tested ESP32 with SIM800 and SIM7000 and they work perfectly.

gecko242, 2020-10-19:
As @nuclearcat said, this sounds like a layout issue. I have an ESP32 in close proximity with a SIM800 with no problems at all.

david-cermak, 2020-10-22:
As said, this seems like a PCB/layout issue. In the posted schematic snippet, there's a net `ESP...` going somewhere, connected to the `EN` pin. Is it a reset button? (path leading though the entire board, where is it connected to...?)
I'm just guessing here, but the pulses look almost like a "GSM buzz" and the reason why the  `Vcc` net is clean might simply be the voltage regulator on the line..?

Also I don't see any problem adjusting your R/C values as you did to 47uF/1k.

closed 2020-12-22 by Alvin1Zhang