espressif/esp-idf #5932: Connect to WPA2-Enterprise Access Point Problem

CKe-rde, 2020-10-01:
- Board: ESP32-WROOM/ESP32 DevKitV1
- IDF-Version: v4 release

**Problem Description**
Connection to WPA2-Enterprise Access Point is not possible.

This applies to all three EAP methods (EAP-TLS, EAP-PEAP, EAP-TTLS)

**Wifi environment**
Cisco WPA2 Enterprise Access Point with a Freeradius server.

**ESP32 Log Output**
 The ESP32 prints the following error repeatedly:

> D (4024) wpa: TLS: using phase1 config options
D (4024) wpa: TLSv1: Failed to initialize verify hash
I (4034) wpa: SSL: Failed to initialize new TLS connection
I (4034) wpa: EAP-TLS: Failed to initialize SSL.
E (4044) wpa: Method private structure allocated failure

The Freeradius server prints the following message 50 times before ending the authenticication process:

> (157) eap: Peer sent packet with method EAP NAK (3)
(157) eap: Peer NAK'd our request for TLS (13) with a request for TLS (13), skipping...
(157) eap: WARNING: !!! We requested to use an EAP type as normal.
(157) eap: WARNING: !!! The supplicant rejected that, and requested to use the same EAP type.
(157) eap: WARNING: !!!     i.e. the supplicant said 'I don't like X, please use X instead.
(157) eap: WARNING: !!! The supplicant software is broken and does not work properly.
(157) eap: WARNING: !!! Please upgrade it to software that works.

**Expected behaviour**
ESP32 connects to Access Point.

**Actual behavoir**
Connection cannot be established using the WPA2 Enterprise Example Code.
Timeout using the AT-Command Firmware v2.1 Release

Using the WPA2 Enterprise credentials or the certifictates created with Freeradius for EAP-TLS on several smartphones works without a problem. So the Freeradius configuration, the Wifi environment or the certificates should not be the causing the error.

sagb2015, 2020-10-07:
Hi @CKe-rde,  this could be interop issue. Can you help to share the certificates used and exact commit ID?

CKe-rde, 2020-10-07:
> 
> 
> Hi @CKe-rde, this could be interop issue. Can you help to share the certificates used and exact commit ID?

The commit ID of the esp-at Firmware is: a8f0b01652197c4a56d9805aee9e3ae61db3847c
The certificates are in the attached zip archive.

The certificates are with sha256 with RSA Encryption. Could that be a problem? 
I've read in an other issue from last month that sha256 isn't supported.

[ESP32_Cert.zip](https://github.com/espressif/esp-idf/files/5340757/ESP32_Cert.zip)

sagb2015, 2020-10-08:
Hi @CKe-rde, I think you are referring to https://github.com/espressif/esp-idf/issues/5805. That issue is fixed on master. Can you help to check? We will also check on our side. If required, we will backport the fix onto v4.0.

CKe-rde, 2020-10-08:
@sagb2015 yes that is the [issue](https://github.com/espressif/esp-idf/issues/5805) I was referring to. There is now mentioned that the fix is not yet included in the master branch. I can test the WPA2 autnetication with the master when the fix is included.

sagb2015, 2020-10-08:
@CKe-rde  Yes, the commit should be available soon since it is internally merged. You will have to wait until then.

closed 2020-10-08 by sagb2015

sagb2015, 2020-10-20:
@CKe-rde  In case you missed  this is now fixed on master.

CKe-rde, 2020-10-20:
@sagb2015  Thank you for the remainder. It works with the examples/wifi/wpa2_enterprise.

Will this also work with [AT Command Firmware](https://github.com/espressif/esp-at)?
I've tried testing it with the v2.1 for the ESP32 but unfortunately I encountered a Problem during the build proccess (which is mentionend in this [#issue](https://github.com/espressif/esp-at/issues/475)) and working on resolving that.