espressif/esp-idf #5934: I have the same problem occurred once with esp-idf v4.0.1, but couldn't duplicate it. Need help on this.

chewhs00, 2020-10-02:
I have the same problem occurred once with esp-idf v4.0.1, but couldn't duplicate it. Need help on this.
[crash08232020.txt](https://github.com/espressif/esp-idf/files/5153915/crash08232020.txt)

_Originally posted by @chewhs00 in https://github.com/espressif/esp-idf/issues/4449#issuecomment-684187985_

igrr, 2020-10-05:
@chewhs00 Could you please try pulling the latest release/v4.0 branch (to include the toolchain update in https://github.com/espressif/esp-idf/commit/6093407d7876a9ea43f2a23ac9e8d22e2e1a60d0) and re-installing the tools (install.sh or install.bat depending on your platform)? Given the backtrace, the issue may be related to the PSRAM cache issue, the workaround for which has been corrected in the new version of the toolchain.

Alvin1Zhang, 2020-10-12:
@chewhs00 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

chewhs00, 2020-10-13:
@igrr Sorry for the late reply. I am about to migrate to v4.1 release for this project and another new project. Does this release include the fix that you mentioned?

igrr, 2020-10-13:
v4.1 release tag does not, but if you use the latest release/v4.1 branch, it will include the toolchain update.

chewhs00, 2020-10-21:
@Alvin1Zhang , currently I upgraded to release/v4.1 branch, and I will keep an eye to see if this indeed solves the problem. For now, you can close the case. Thank you and @igrr for the help.

closed 2020-10-21 by chewhs00