espressif/esp-idf #5936: Cant connect to router but connecting to mobile hotspot works

Viriatto, 2020-10-02:
> ### Hardware:
> **Router:**                                                   Hitron CVE30360
> **Board:**							DOIT Devkit V1
> **Core Installation version:**			1.0.4
> **IDE name:**						Arduino IDE
> **Flash Frequency:**					80Mhz
> **PSRAM enabled:**                                     (Not sure)
> **Upload Speed:**					921600
> **Computer OS:**                                         Windows 10

### Description:
When in station mode, can connect normally to mobile hotspot without any errors or problems, but connecting to home router displays a ton of AUTH_EXPIRE, ASSOC_EXPIRE, NO_AP_FOUND and 4WAY_HANDSHAKE_TIMEOUT errors. Sometimes after dozens of these it ends up connecting for a couple of minutes/seconds before disconnecting again.

### What I have tried:

- Adding `Wifi.disconnect()` at the beggining of `setup()` - [Comment](https://github.com/espressif/arduino-esp32/issues/2144#issuecomment-450394742)
- Tried different firmware versions (10.4, 1.03 and 1.02)
- Changed router mode from 802.11 g/n to 802.11 b/g/n
- Rebooted router 
- Added these lines of code on setup as suggested in [this comment](https://github.com/espressif/arduino-esp32/issues/2144#issuecomment-545303384)
```
ESP_ERROR_CHECK(nvs_flash_erase());
nvs_flash_init();
ESP.restart();
``` 

### I have "trimmed" some of the code for easier readibility, if you think I should post the full code let me know
```cpp

#include <WiFi.h>
#include "nvs_flash.h"

static void WiFiEvent(WiFiEvent_t event, WiFiEventInfo_t info)
{
  if (event == SYSTEM_EVENT_STA_DISCONNECTED) {
     WiFi.reconnect();
  }
}

void connectWifi() {
  const char* WIFI_SSID = "SSID";
  const char* WIFI_PASSWORD = "PASSWORD";

  IPAddress GATEWAY(192, 168, 1, 1);  
  IPAddress STATIC_IP(192, 168, 1, 80);  
  IPAddress SUBNET(255, 255, 255, 0); 
  
  const char* CONSOLE_CONNECTED = "CONNECTED TO ";
  const char* CONSOLE_IP_ADDRESS = "IP ADDRESS: ";

  WiFi.config(STATIC_IP, GATEWAY, SUBNET);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.println("");

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  
  Serial.println("");
  Serial.print(CONSOLE_CONNECTED);
  Serial.println(WIFI_SSID);
  Serial.print(CONSOLE_IP_ADDRESS);
  Serial.println(WiFi.localIP());
}

void setup(void) {
  Serial.begin(115200);
  Serial.println();

  ESP_ERROR_CHECK(nvs_flash_erase());
  nvs_flash_init();
  
  connectWifi();
  WiFi.onEvent(WiFiEvent);
}

void loop(void) {}
```

### Debug Messages when connecting to home router:
```
...[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
......[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
.[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
.
CONNECTED TO NOS4920
IP ADDRESS: 192.168.1.80
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 4 - ASSOC_EXPIRE
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND

```
### Debug Messages when connecting to mobile hotspot:
```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START

....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
.
CONNECTED TO HUAWEI
IP ADDRESS: 192.168.1.80

```

Thank you very much for your help

HarveyRong-Esp, 2020-10-21:
Hi @Viriatto,  

Have you tried changing the router?

Can you provide a wireless packet capture corresponding to the unconnected wifi log?

The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets:
https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

Viriatto, 2020-10-21:
Hey @HarveyRong-Esp,

Connecting to mobile hotspot works just fine. Just doesn't seem to consintently connect to this specific home router. As for the wireshark setup, I'm not sure how to do it, I tried filtering for router and the esp MAC address only while attempting to connect but it didn't record anything.

Thank you

HarveyRong-Esp, 2020-10-23:
Have you tried using non-ESP devices to connect to the router? 

Generally speaking, we need to analyze connection problems through data packets, it is difficult to analyze without data packets

HarveyRong-Esp, 2020-11-17:
Hi @Viriatto， Can you provide the model of the router in question? We will buy and try to copy locally.

Viriatto, 2020-11-17:
Hi @HarveyRong-Esp,

I'm not sure how I missed your other comment and I apologize for that. Yes, other devices can connect to this router normally.

As for the router model it is a Hitron CVE30360. 

I appreciate the effort

HarveyRong-Esp, 2020-12-01:
Hi @Viriatto, I’m sorry, we have not been able to reproduce the issue, but the analysis of the issue can only be done through Log and wireless data packets, you can do not need to perform any filtering when capturing packets. But it should be noted that the primary and secondary channels must be consistent with the AP.

HarveyRong-Esp, 2020-12-29:
@Viriatto， Is there any update?
Can this issue be reproduced 100%? Can recover when a issue occurs? If it can be recovered, it may be caused by sta not receiving beacon, auth or assoc when the environment is bad. This situation is normal.

Viriatto, 2020-12-31:
Hi @HarveyRong-Esp sorry for not responding. I decided to change the way I was doing the project and I am now using the microcontroller as an AP instead, and it is working as expected. I did however try to run it again as station and faced the same issues as previously commented. I am certain it is something with my router that isn't working properly because Ive been facing the same issues while connecting different devices to the network recently. I will get a new router next month and will get back to you

HarveyRong-Esp, 2021-01-18:
Hi @Viriatto， Are there any updates?

Viriatto, 2021-01-18:
Works as expected with the new router.

closed 2021-01-18 by Viriatto