espressif/esp-idf #5937: dead code in esp_efuse_mac_get_default

phoddie, 2020-10-02:
The following lines in `esp_efuse_mac_get_default` are dead code:

https://github.com/espressif/esp-idf/blob/c77c4ccf6c43ab09fd89e7c907bf5cf2a3499e3b/components/esp_common/src/mac_addr.c#L99-L102

Regardless of the value of `mac_low`, the function returns `ESP_OK`

negativekelvin, 2020-10-02:
What do you mean there is a range of 50 values

phoddie, 2020-10-02:
Yes, there is a range of values.  But, whether the value of `mac_low` is inside or outside of that range, the function returns `ESP_OK`. It is clear if you look at what follows the dead code.

https://github.com/espressif/esp-idf/blob/c77c4ccf6c43ab09fd89e7c907bf5cf2a3499e3b/components/esp_common/src/mac_addr.c#L98-L110

negativekelvin, 2020-10-03:
I see, the else needs to be removed so that it aborts

phoddie, 2020-10-03:
Yes, something like that. I'm not entirely sure I understand the intent of the code, so I don't dare suggest a specific change. Thank you for your patience to understand the bug report.

closed 2020-11-03 by espressif-bot