espressif/esp-idf #5938: wifi heap consumption

chegewara, 2020-10-03:
I dont know if its bug, but for sure it is issue. My case:
- esp32-S2,
- arduino as component, esp32s2 branch + IDF master,
- wifi softAP example by copy/paste with wifi default settings in menuconfig, https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/softAP/main/softap_example_main.c
- free heap before adding wifi code is over 180kB,
- free heap after adding wifi code is less than 1kB, 180kB consumed by wifi?
- when i am trying to connect to AP i am getting this error:
`W (106278) wifi:alloc eb len=216 type=2 fail, heap:140`
stack failed due to allocation error

I could make my code to work by using SPIRAM for wifi and setup minimum amount of wifi buffers, but the question is about overall wifi heap usage and why default values are so high?

Thanks

zhangyanjiaoesp, 2020-10-26:
hi @chegewara ,

I have tested with the softAP example on esp32-s2, the heap size consumption is 194708-151116=43592.

closed 2020-11-12 by chegewara

jdoubleu, 2021-08-10:
I can confirm, the WiFi logic roughly using `~43KB`. When you're also using the `nvs_flash_init()`, it eats up an additional `~5KB` of memory.

This might help you: There's an option in the menuconfig under `Component config > ESP32-specific > SPI RAM config > [ ] Try to allocate memories of WiFi and LWIP in SPIRAM firstly. If failed, allocate internal memory`.