espressif/esp-idf #5939: Random disconnect without handle-able event

ZweiEuro, 2020-10-04:
## Environment

- Development Kit:     ESP32 Devkit V1 -Doit (I think, it's soldered i can't look at it anymore) (here)[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/modules-and-boards.html#esp32-wroom-32]
- Kit version -
- Module or chip used:  ESP32-WROOM-32

- IDF version (run ``git describe --tags`` to find it): framework-espidf 3.40001.200521 (4.0.1)  (PIO)
- Build System:         platformio
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0

- Operating System:   Linux

- Using an IDE?:  Vscode platformio
- Power Supply:         USB

## Problem Description
I wrote a simple program that pings my server every second (simple http get request).
My esp connects to wifi (mobile hotspot)  just fine, but occasionally disconnects with reason for the wifi handler (making it impossible to react or start reconnecting when appropriate. Also handling the event gracefully is really hard.

My workaround is to restart the whole esp when the http connection fails, because I know that if the wifi has internet it has to time out.
I always get this just before the disconnect.
```
 (46748) wifi:bcn_timout,ap_probe_send_start
I (55868) wifi:bcn_timout,ap_probe_send_start
I (64978) wifi:bcn_timout,ap_probe_send_start
I (79538) wifi:bcn_timout,ap_probe_send_start
I (88938) wifi:bcn_timout,ap_probe_send_start
```

But no "handler event" on the wifi handler.

### Expected Behavior

No random disconnects, or rather an event that makes it handle-able in a less gruesome way than making the system ping every second.

### Actual Behavior

Seems that esphttp_client does something weird with the wifi

```
␛[0;32mI (71) boot: Chip Revision: 1␛[0m
␛[0;32mI (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0␛[0m
␛[0;32mI (39) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader␛[0m
␛[0;32mI (39) boot: compile time 15:58:27␛[0m
␛[0;32mI (40) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (45) boot: SPI Speed      : 40MHz␛[0m
␛[0;32mI (49) boot: SPI Mode       : DIO␛[0m
␛[0;32mI (53) boot: SPI Flash Size : 4MB␛[0m
␛[0;32mI (58) boot: Partition Table:␛[0m
␛[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000␛[0m
␛[0;32mI (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000␛[0m
␛[0;32mI (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (91) boot:  3 factory          factory app      00 00 00010000 001c2000␛[0m
␛[0;32mI (98) boot:  4 ota_0            OTA app          00 10 001e0000 001c2000␛[0m
␛[0;32mI (106) boot:  5 storage          Unknown data     01 82 003a2000 0001e000␛[0m
␛[0;32mI (113) boot: End of partition table␛[0m
␛[0;32mI (118) boot: Defaulting to factory image␛[0m
␛[0;32mI (122) boot_comm: chip revision: 1, min. application chip revision: 0␛[0m
␛[0;32mI (129) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x39b78 (236408) map␛[0m
␛[0;32mI (223) esp_image: segment 1: paddr=0x00049ba0 vaddr=0x3ffbdb60 size=0x03f1c ( 16156) load␛[0m
␛[0;32mI (230) esp_image: segment 2: paddr=0x0004dac4 vaddr=0x40080000 size=0x00400 (  1024) load␛[0m
␛[0;32mI (230) esp_image: segment 3: paddr=0x0004decc vaddr=0x40080400 size=0x02144 (  8516) load␛[0m
␛[0;32mI (242) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xfebcc (1043404) map␛[0m
␛[0;32mI (621) esp_image: segment 5: paddr=0x0014ebec vaddr=0x40082544 size=0x1a1ac (106924) load␛[0m
␛[0;32mI (683) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (683) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (684) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (687) cpu_start: Application information:␛[0m
␛[0;32mI (692) cpu_start: Project name:     master_gate␛[0m
[... 185 lines cut ...]
␛[0;32mI (115062) mg_net_services: HTTP_EVENT_DISCONNECTED␛[0m
I (115072) wifi:state: run -> init (0)
I (115072) wifi:pm stop, total sleep time: 87492309 us / 108977867 us

I (115082) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
␛[0;32mI (115092) wifi station: disconnected, repeat 0␛[0m

```

And then i call disconnect and connect.

### Steps to reproduce
For me this happens with any code that uses wifi. This happened to me before with: [issue](https://github.com/espressif/esp-idf/issues/5674)

### Code to reproduce this issue

make the http handler repeat each second, and you got code that does that

## Debug Logs

## Other items if possible

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] coredump (This provides stacks of tasks.)

ZweiEuro, 2020-10-04:
After changing how i allocate the data from the response (  i needed to process it a bit further so i needed a buffer).
from char buffer[evt->data_len] to a static char* buffer which i alloc and free with malloc.

Could someone confirm this ? how does this influence a bcn timeout ?

ZweiEuro, 2020-11-04:
I FOUND THE REASON! IT ONLY TOOK 6 MONTHS !
So, what I did wrong had nothing to do with the net, it was just the cause.
For my system it started a task that pinged each hour, that task accessed the local data (spiffs)
for some surely _very smart_ reason (jesus that's bad programming) I remounted and mounted the FS everytime i read or wrote from that file.
I am assuming the system at some point just has no mount points left or just isn't built for that kind of pressure, which I totally understand.

Weirdly enough the beacon timeout happens as a result, but i cannot for the life of me tell you _why_ exactly that gets killed. 
Most likely something with the filesystem gets butchered.

closed 2020-11-04 by ZweiEuro

gadget-man, 2020-11-04:
Would you be able to post a before/after code snippet so that I can see what you changed - it may be a good indicator for other seeing similar problems?

ZweiEuro, 2020-11-04:
not really? 
Well every time my coded accessed the spiffs it executed this:
```
esp_err_t ret = esp_vfs_spiffs_register(&spiffs_conf);
      if (ret != ESP_OK)
      {
        if (ret == ESP_FAIL)
        {
          if (debug_local_storage)
            ESP_LOGE(TAG, "Failed to mount or format filesystem");
        }
        else if (ret == ESP_ERR_NOT_FOUND)
        {
          if (debug_local_storage)
            ESP_LOGE(TAG, "Failed to find SPIFFS partition");
        }
        else
        {
          if (debug_local_storage)
            ESP_LOGE(TAG, "Failed to initialize SPIFFS (%s)", esp_err_to_name(ret));
        }
        return mg_err_writing_blob;
      }
```

and after getting everything it needed it executed this:
`esp_vfs_spiffs_unregister(spiffs_conf.partition_label);`

Which is just , hella bad for a microcontroller system.

So i removed the call and made the mount once at bootup and then never unmounted it since i need it anyways.

that fixed my problem.

gadget-man, 2020-11-05:
OK thanks. I really wish there was some way to get better insights from the core code as to what is causing the `bcn_timeout`  error to occur, rather than just guessing and finding the issues by process of elimination.

ZweiEuro, 2020-11-05:
You don't need to tell me haha
noone could help me or point me to any direction, I just had to reverse engineer where the error was coming from
almost 6 months it drove me insane

geowal19, 2021-10-02:
I would like to add some clarity for this issue. My case of this issue was caused by a memory leak. Instead of resulting in a crash, the WiFi stopped working with the following error:

`I (151938) wifi:bcn_timout,ap_probe_send_start`

Use `esp_get_free_heap_size()` to see if the memory is being used up and track your memory usage.