espressif/esp-idf #5940: Update OpenOCD on release-4.2 branch

zx64, 2020-10-04:
Please consider cherry picking commit f857ab85c3c2bf509bf2e2fc6f59c73a0f2ca536 onto the `release-4.2` branch.

I'm using that branch for my ESP32-S2 Kaluga devkit and the July version of OpenOCD has a few new config files that makes `idf openocd` work with that hardware.

Thanks!

igrr, 2020-10-05:
Thanks for the reminder @zx64, will do.

zx64, 2020-10-22:
Appears to have been fixed by 7d8d2b982e114904bce724d555ff11b7e436a68e, thanks!
(Worth pointing out that for some reason this issue didn't get notified and/or closed automatically by whatever process you have for internally triaging and managing issues etc.)

closed 2020-10-22 by zx64

Alvin1Zhang, 2020-10-23:
Thanks for the updates.