espressif/esp-idf #5941: Can't build bootloader using idf.py

skywa04885, 2020-10-05:
Now I really cannot deal with this issue, have to make something working in a short amount of time, and it literally just does not work.

When executing idf.py i get this error:

`xtensa-esp32-elf-gcc: error: Drive/Projects/ESP/hello_world/build/bootloader/bootloader.map: No such file or directory
ninja: build stopped: subcommand failed.
[3/360] Linking CXX static library esp-idf/e`

I just installed IDF and followed the installation guide exactly. Please help !

igrr, 2020-10-05:
Hi @skywa04885, could you please fill in the issue template you were offered when creating the issue? Information such as: operating system, ESP-IDF version, compiler version, full output from the build process can all help to find the issue.

From the brief excerpt you have shared, I suspect that your project folder has spaces in its path. This isn't supported by the ESP-IDF build system, yet. Please try to copy the hello_world project into a different directory, and do `idf.py fullclean`, and then build again.

If you are interested in support for spaces in project path, please visit https://github.com/espressif/esp-idf/issues/5576 and click "Subscribe" button to receive notifications.

Alvin1Zhang, 2020-10-12:
@skywa04885 Thanks for reporting, would you please help share more details of the issue per the issue template? Thanks.

closed 2021-01-12 by Alvin1Zhang