espressif/esp-idf #5943: aws iot sdk v4.0 integration in esp-idf v3.3 and esp-idf v4.1

chaabanemalki, 2020-10-05:
Hello,

AWS released few days ago the aws iot 4.0 and there were a branch with aws 4.0 beta in the esp-idf now it is removed.

Can you please share your timeline about the integration of aws IoT 4.0 into esp-idf 4.1 and v3.3

Thank you,
Chaabane

chaabanemalki, 2020-10-14:
Hello,

any feedback regarding my request ?

shahpiyushv, 2020-10-14:
@chaabanemalki , it is not really clear what exactly the requirement is. Are you asking about the latest [202009.00](https://github.com/aws/aws-iot-device-sdk-embedded-C/releases/tag/202009.00) release of [aws-iot-device-sdk-embedded-C](https://github.com/aws/aws-iot-device-sdk-embedded-C)? As per my knowledge, there was never any aws 4.0 beta branch either in esp-idf or esp-aws-iot.

Alvin1Zhang, 2020-10-20:
@chaabanemalki Thanks for reporting, would you please help share if any updates? Thanks.

chaabanemalki, 2020-10-20:
Hello, Thank you for the feedback.

in the readme here https://github.com/espressif/aws-iot-device-sdk-embedded-C

it says 

> We have released version 4.0.0 beta 1 of this SDK on the v4_beta branch and encourage everyone to give it a try.

but when you click on the link of the branch `v4_beta`  it says not found. not sure if it was merged or deleted ?
So my initial question is: Now that the official aws iot v4.0 is out. Does espressif plan to integrated ?

shahpiyushv, 2020-10-20:
@chaabanemalki , the repository you mentioned is just a (somewhat stale) fork of [this AWS repo](https://github.com/aws/aws-iot-device-sdk-embedded-C). The `v4_beta` branch has now been deprecated and renamed to `v4_beta_deprecated` in the original repo as you can check [here](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/v4_beta_deprecated). So, we do not plan to have any port for that. However, we are evaluating the porting efforts for Release [202009.00](https://github.com/aws/aws-iot-device-sdk-embedded-C/releases/tag/202009.00) and will provide an update on that soon.