espressif/esp-idf #5944: esp_tls_conn_read error: panic'ed

kzoltaan, 2020-10-05:
ESP32-WROOM-32  / external 5V|
ESP-IDF:  v4.0-327-gfe67bedee-dirty 
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Linux
 
I create a tcp client and I run the receiver with xTaskCreate.
 
```
xReturned = xTaskCreate(&tcp_client_receiver, "recv task", 4096, NULL, tskIDLE_PRIORITY, NULL);

static void tcp_client_receiver(void *pvParameter)
{
	
	char buf[512];
	int len;
	ESP_AT_OTA_DEBUG("START READ\r\n");
	
	len = sizeof(data_buffer) - 1;
    bzero(data_buffer, sizeof(data_buffer));
	 
	do {
			if ((len = esp_transport_read(ssl, (char *)data_buffer, 512, 60000))<=0)   ////// <----- tcp_client.c:103
			{
				tcp_client_close();
				break;
			}
			 
			ESP_LOGD(TAG, "%d bytes read", len);
			
			sprintf(buf, "+TCPDATA:%d,%s\r\n", len, (char *)data_buffer);
			esp_at_port_write_data((uint8_t *)&buf, strlen( buf ));
		 
			
    } while(Connected);
	
	ESP_AT_OTA_DEBUG("STOP READ\r\n");
	vTaskDelete(NULL);	 
}

```

When I close the peer, I ESP reboot.
```
E (88398) esp-tls: read error :-80:
E (88404) TRANS_SSL: esp_tls_conn_read error, errno=Connection reset by peer
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c2ff  PS      : 0x00060f30  A0      : 0x80108c99  A1      : 0x3ffe95a0
A2      : 0x00000000  A3      : 0x3ffb54e0  A4      : 0x0000000c  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000050  A8      : 0x00000000  A9      : 0x3ffe9520
A10     : 0x00000058  A11     : 0x3ffdf498  A12     : 0x3ffe9590  A13     : 0x0000000c
A14     : 0x00000000  A15     : 0x00000005  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: df6c6b3bcb67cb299801608abe89eba151821d49f7f34fe644fd6980775f4487

Backtrace: 0x4000c2fc:0x3ffe95a0 0x40108c96:0x3ffe95b0 0x401088e5:0x3ffe95d0 0x401df029:0x3ffe95f0 0x400fd652:0x3ffe9610 0x400999ed:0x3ffe9830
0x40108c96: esp_transport_set_errors at /esp32/esp-at/esp-idf/components/tcp_transport/transport.c:306

0x401088e5: ssl_read at /esp32/esp-at/esp-idf/components/tcp_transport/transport_ssl.c:137 (discriminator 2)

0x401df029: esp_transport_read at /esp32/esp-at/esp-idf/components/tcp_transport/transport.c:182

0x400fd652: tcp_client_receiver at /esp32/esp-at/main/tcp_client.c:103

0x400999ed: vPortTaskWrapper at /esp32/esp-at/esp-idf/components/freertos/port.c:143

```
How Can I handle the TCP problems without reboot?
Thanks

david-cermak, 2020-10-06:
Hi @kzoltaan 

Could you please share the code when the tcp_transport is initialized? My guess is that you init only the ssl-transport, not using transport lists. We typically initialize transports as
* init the list with `esp_transport_list_init()` first
* init the transport with `esp_transport_***_init()`
* add the transport to the list with `esp_transport_list_add()`

But please share your init code, this should work even if the transports get created separately.

Alvin1Zhang, 2020-10-12:
@kzoltaan Thanks for reporting, would you please help share if any updates or more details for this issue? Thanks.

kzoltaan, 2020-10-13:
@david-cermak  u r right. it is working with this init.

`transport_list = esp_transport_list_init();
esp_transport_handle_t tcp = esp_transport_tcp_init();

esp_transport_set_default_port(tcp, server_port);
esp_transport_list_add(transport_list, tcp, "tcp");

ssl = esp_transport_ssl_init();
	
if (esp_transport_connect(ssl, server_ip, server_port, 2000) < 0) {
        ESP_LOGE(TAG, "Connection failed, sock < 0");
        return false;
} 
ESP_LOGD(TAG, "CONNECTED");
Connected = true;
xReturned = xTaskCreate(&tcp_client_receiver, "recv task", 4096, NULL, tskIDLE_PRIORITY, NULL);
`

closed 2020-10-13 by kzoltaan